mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #545: Horizontal Ghosting in bottom half of 64x64? Can PIN_E be tuned?

JingleheimerSE, 2023-12-18:
@mrfaptastic wondering if you have a suggestion for eliminate the ghosting I am seeing in the bottom half of this Waveshare 64x64 panel. I assume its an issue with the PIN_E and the extra timing needed but I cant seem to find the nob, if it exists, to tune it.

I'm only really seeing ghosting on vertical edges like the checkerboard or text. In those cases I get flickering of near by pixels.
[image]
[image]

Horizontal lines are much cleaner except for the last pixel:
[image]

Wiring wise, I'm using the Adafruit MatrixPortal S3 directly connected to the HUB75 header so I don't have any flexibility in my pin choice. I've played with the power supply and added a capacitor but didn't see any changes. Using this same setup on a no-name Aliexpress panel, I get nice clean lines but it uses the ICN2038S driver:
[image]

The panel appears to have a SM16208SJ driver with a SM5166PS multiplexor. Given that, I've got my driver in the code set to `HUB75_I2S_CFG::SHIFTREG` 
[image]

I've been through the Common Issues page a couple of times and have tweaked the following, which helped clean up the upper half:
```
  mxconfig.latch_blanking = 4;
  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_8M;
```

I'm currently using the PIO example as a test bed but my primary use case is actually displaying a clock and static text using https://github.com/TillFleisch/ESPHome-HUB75-MatrixDisplayWrapper.

Here's my tweaked:
[image]

mrfaptastic, 2023-12-18:
Try increasing the latch blanking amount.

Looks like interference to me, perhaps these LED Matrix PCBs have issues at high speeds.

Try reducing the colour depth of the library to see if that helps as well.

ergindemir, 2024-03-04:
Did you toggle the clkphase?
Have you tried Adafruit protomatter library?
If the issue persists with protomatter, it's panel otherwise library issue.