mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #551: ESP32 S3 Hardware and LCD connections

phodina, 2023-12-28:
Hi,

I'm designing a controller board for a [backpack](https://gitlab.com/phodina/badboard) I recently bought which had let's say poor and proprietary implementation.

Normally I'd use STM32F4 with the FSMC interface to drive the RGB LED panel. However, as I'd like to connect to the controller over wireless (not sure what's better WiFi/Bluetooth) I have switch to ESP32.

I used mostly ESP32 C3 to play with the RISC-V core.

However, I never used the ESP32 and I could not find in the datasheet in the pinout nor LCD section anything about the pins that are available.

I checked the default pins and set accordingly the connections in the schematics.
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/src/platforms/esp32s3/esp32s3-default-pins.hpp
[image]

Also the suggested link for the hardware seems to be no longer valid.
[image]

So is this the best setup?

Also is it possible to change pins for `G2` and `B2` as they are connected to the RTC crystal that I'd like to use for keeping precise time?

mrfaptastic, 2023-12-28:
C series chips don't have the LCD module. Won't ever work. Unless you use PxMatrux library. Please read README what chips are supported.

phodina, 2023-12-28:
> C series chips don't have the LCD module. Won't ever work. Unless you use PxMatrux library. Please read README what chips are supported.

Yes, that's why the selected MCU is **ESP32 S3**. That section is clear :)

> Various people have created PCBs for which one can simply connect an ESP32 to a PCB, and then the PCB to the HUB75 connector, such as:

 >   Brian Lough's [ESP32 I2S Matrix Shield](http://blough.ie/i2smat/)
 
 This link is no longer available.
 
What I miss is the description of the pins. In the ESP32 datasheet the pinout does not list which pins are used for LCD.
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf
[image]

Lastly can these pins be remapped?

phodina, 2023-12-28:
I've been able to find the mapping for SPI RAM on the ESP32 S3 chip
https://api.riot-os.org/group__cpu__esp32__esp32s3.html
[image]

Though there's the speed limitation
> For the ESP32-S3 only, you can use SPIRAM/PSRAM to drive the HUB75 DMA buffer when using an ESP32-S3 with OCTAL SPI-RAM (PSTRAM) (i.e. ESP32 S3 N8R8 variant). However, due to bandwidth limitations, the maximum output frequency is limited to approx. 13Mhz, which will limit the real-world number of panels that can be chained without flicker. Please do not use PSRAM as the DMA buffer if using QUAD SPI (Q-SPI), as it's too slow.

phodina, 2023-12-28:
In the ESP-IDF source code there's the definition for the pins but still does not document the LCD peripheral

https://github.com/espressif/esp-idf/blob/ee3a46dbc3aa8575db42af6afbe37acbc2280bbe/examples/peripherals/lcd/rgb_panel/main/rgb_lcd_example_main.c#L33

toine512, 2023-12-28:
Chapter 6 of the ESP32-S3 TRM.
All pins not directly affected to special interfaces (USB, JTAG, SPI flash, SPI RAM) are usable. That means all pins left available to the GPIO matrix.

Default GPIO mapping used by the library is here : https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/src/platforms/esp32s3/esp32s3-default-pins.hpp
If you need E you may use GPIO39 for example.

mrfaptastic, 2023-12-28:
Default library pins for S3 are defined here:

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/src%2Fplatforms%2Fesp32s3%2Fesp32s3-default-pins.hpp

phodina, 2023-12-29:
> ESP32-S3 TRM
> All pins not directly affected to special interfaces (USB, JTAG, SPI flash, SPI RAM) are usable. That means all pins left available to the GPIO matrix (not in "direct connection" mode).

Thanks. That's exactly what I was looking for.

phodina, 2023-12-29:
> Default library pins for S3 are defined here:
> 
> https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/src%2Fplatforms%2Fesp32s3%2Fesp32s3-default-pins.hpp

Thanks. For the definition of the pins but the there but the question was more broad as what other pins can be used :)  as I'd like some peripherals to be used and it would conflict with the default pins.