mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #553: 1/10 scan

muthu920, 2023-12-30:
Is possible add custom scan to this library, our board is 1/10scan 80x40 led per panel, [ESP32-HUB75-MatrixPanel-DMA] works but 9th and 10th row is not printed for each scan because its printing 1/8s, is possible to convert to 1/10scan
[image]

board707, 2023-12-30:
Do you have several such panels? Try another one

muthu920, 2023-12-30:
no i have only one panel, does this library support 1/10scan ?

muthu920, 2023-12-30:
beacuse only 9th and 10th row missing but 8row display perfect

muthu920, 2023-12-30:
driver ic RUC7258D

mrfaptastic, 2023-12-30:
I think it's a defective panel. 

Can you try with another library like PxMatrix?

muthu920, 2023-12-30:
ok i will check and update

> I think it's a defective panel.
> 
> Can you try with another library like PxMatrix?

Ok i will check and update

muthu920, 2023-12-30:
[image]

Using VirtualMatrix example

[image]

muthu920, 2023-12-30:
if i use virtualDisp->fillScreen(virtualDisp->color565(100, 0, 0));

The all pixel show perfect 

[image]

muthu920, 2023-12-30:
when i set a pixel at   9th and 10th row but it show at 10th and 11th row 

[image]

[image]

muthu920, 2023-12-30:
i think need to modify this library to 1/10Scan, any idea? Right now library scanning is 1/8

muthu920, 2023-12-30:
> I think it's a defective panel.
> 
> Can you try with another library like PxMatrix?

does this library support 1/10scan

board707, 2023-12-30:
What's the second chip aside 7258
 on your panel?

If you have a Stm32 bluepill or Rp2040 board, you can test the panel with DMD_Stm32 library.

muthu920, 2023-12-30:
> What's the second chip aside 7258 on your panel?
> 
> If you have a Stm32 bluepill or Rp2040 board, you can test the panel with DMD_Stm32 library.

DP245D second chip

muthu920, 2023-12-30:
I set to 

HUB75_I2S_CFG mxconfig(
        PANEL_RES_X * 2,              // DO NOT CHANGE THIS
        PANEL_RES_Y / 2,              // DO NOT CHANGE THIS
        PANEL_CHAIN           
    );

virtualDisp->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);
and changed 

[image]

[image]

now i get 
[image]

[image]

board707, 2023-12-30:
D245 is not a driver, should be yet another  one

muthu920, 2023-12-30:
> DP245d is not a driver, should be yet another one

DP5125D

board707, 2023-12-30:
Thank you. The good news that it is a standard driver, it should be supported by the library.

muthu920, 2023-12-30:
> Thank you. The good news that it is a standard driver, it should be supported by the library.

I tried E pin gnd and Valid GPIO too, but same result

ok, i will buy one new board and check and update.

muthu920, 2023-12-30:
> Thank you. The good news that it is a standard driver, it should be supported by the library.
> 
> It seems to me that the reason is a broken panel or bad connections. Check your board and panel for unsoldered pins.

In my board it mentioned P4-1921-80x40-10S-S2.0

muthu920, 2024-01-01:
Tested with new panel but same result.

Panel not mistake, we test with other driver which comes with board it working perfect.

The problem with our library is it sacn 8 rows perfect but not scanning 9th and 10th row, does we need not modify virtiual to real pixel mapping for 1/10 scan because 1/8 scan working perfect but not 1/10 scan.

mrfaptastic, 2024-01-01:
The library only natively supports 1/16 scan panels as outlined in the README, however, with some pixel-remapping  (I.e. Virtual Matrix class) we have been able to incorporate support for other random panels.

You'll need to change the virtual matrix class to add the necessary support for this panel.

muthu920, 2024-01-02:
Ok thank you will update

closed 2024-01-09 by mrfaptastic