mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #558: Unable to compile for ESP-S3

utiq, 2024-01-08:
I'm using PlatformIO and I'm unable to compile, I get this error:

```bash
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/screen.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0xc): undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/screen.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x10): undefined reference to `Bus_Parallel16::config(Bus_Parallel16::config_t const&)'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/screen.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x14): undefined reference to `Bus_Parallel16::init()'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/screen.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x18): undefined reference to `Bus_Parallel16::dma_transfer_start()'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/screen.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA17allocateDMAmemoryEv+0x50): undefined reference to `Bus_Parallel16::enable_double_dma_desc()'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/screen.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA17allocateDMAmemoryEv+0x54): undefined reference to `Bus_Parallel16::allocate_dma_desc_memory(unsigned int)'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/screen.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o): in function `MatrixPanel_I2S_DMA::configureDMA(HUB75_I2S_CFG const&)':
/Users/me/ESP32-HUB75-MatrixPanel-DMA/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:213: undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/me/ESP32-HUB75-MatrixPanel-DMA/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:217: undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'
```

utiq, 2024-01-09:
I tried with both, I downloaded the zip file from the release page v3.0.9 and it didn't work
I also tried cloning master and it didn't work

mrfaptastic, 2024-01-09:
I cannot replicate this issue and I have PlatformIO with the ESP32 Platform Installed. Clean install your PlatformIO ESP32.

[image]

utiq, 2024-01-09:
Nothing, I nuked .platformio and reinstalled, same error.
Does any of these have anything to do?
- I'm on a Mac M3
- I'm trying to build for espidf

```
[env]
platform = espressif32
framework = espidf, arduino
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
```

In general, there is any workaround that I could try?

utiq, 2024-01-09:
Oh it finaled compiled, I needed to add **lib_deps**
```
[env]
platform = espressif32
framework = espidf, arduino
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
lib_deps =
    ESP32 HUB75 LED MATRIX PANEL DMA Display
```