mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #561: Double sided screen

metingul, 2024-01-09:
Hello,
I used P4, 80X40 pixel modules and it works great. I want to make double sided screen, of course I can resize scrren sizes and duplicate content but the screen size is 160X80, so I dont want to workload to ESP, may be if there any solution for double screen/duplicating.

thank you.

mrfaptastic, 2024-01-09:
Sorry.  It's not clear to me what you asking.

metingul, 2024-01-09:
The resolution of the screen I made is 160X80 pixels. If I make it double-sided, I can use it by making the screen resolution 160X160, but I don’t think the ESP will support a 160X160 resolution. Both sides of the screen will display the exact same content, maybe there is a simpler way to display the same content on both sides, I pulled a parallel cable but it didn’t work.

if no way, I have to use two each ESP32.

mrfaptastic, 2024-01-09:
The easiest way is to write a sketch for that resolution and see if it runs.  You don't need to have any panels connected - look at the error log to see if it works / has enough memory.

board707, 2024-01-09:
>  Both sides of the screen will display the exact same content, maybe there is a simpler way to display the same content on both sides,

Make two 160x80 screens and connect it in parallel to all pins