mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #566: Control 3 led panel parallel ?

phongans, 2024-01-10:
I am looking to control 3 LED panels independently. How would displaying a video in a 192x128 format work with an ESP32?

board707, 2024-01-10:
Please clarify your request. Why do you need to control panels "in parallel" ? 
Normally led panels connects in series, you don't need a parallel connection to display the video on 3 panels

phongans, 2024-01-10:
I don't know, so I want to connect 3 boards so that I can run a complete vertical video with 3xPanel using an ESP32.
[image]

board707, 2024-01-10:
So you don't need to control panels independently.
Connect 3 panels in series and you can use it as a whole big screen.

phongans, 2024-01-10:
How to show one video with 3 chain?

board707, 2024-01-10:
No, I don't
Wait for @mrfaptastic , he is maintainer of the library

mrfaptastic, 2024-01-10:
I've gone out of my way to document how this is done.  Read the PDF.