mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #568: I try test show video vertical with 3 chain led panel

phongans, 2024-01-10:
I have tried using and installing it as per the instructions, but there might have been a mistake somewhere.

[image]

board707, 2024-01-10:
looks like you specified a wrong panel dimensions

phongans, 2024-01-10:
This is config dimensions

#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>

// 2) Set pinout
#define PIN_E 32
#define PIN_A 22
#define PIN_B 21
#define PIN_C 2

// 3) Set configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 4                     // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1                     // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS *NUM_COLS // total number of panels chained one to another

board707, 2024-01-10:
Panels on the photo are definitely not a 64x64.

>  just connect with 2 panel

If you connect 2 panels, why did you set 4 panels in the code?

phongans, 2024-01-10:
[image]

This is show when set ROWS 2 & COLS 1

board707, 2024-01-10:
You need a D pin for these panels

phongans, 2024-01-10:
D pin? How to config D pin, when i set default with GPIO 17 

```
#define D_PIN 17
```

board707, 2024-01-10:
I wrote because in your pin definitions above no D pin

phongans, 2024-01-10:
> I wrote because in your pin definitions above there is no D pin

[image]

That okay, so how can show video @@

board707, 2024-01-10:
looks correct
Any thanks for me? :)

Sorry, I have no experience with video output

phongans, 2024-01-10:
Thanks so much @board707

closed 2024-01-10 by phongans