mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #571: how to support high frame rate or high refresh rate LED screen?

houstontang, 2024-01-16:
Sir, I got some LED screen based on CFD435A(column scan) and WM5166PC (row scan), it's high frame rate screen, how can I change the refresh rate in the library?  thanks!

Houston

board707, 2024-01-16:
According to[ the info](https://en.chipfountain.com/soul5/52abeee4-9d31-c688-a7fa-43afaa2515a3.html), the driver is not compatible with the current library.

Lukaswnd, 2024-03-19:
Besides that, the library continuously puts out the frames. So you have to increase the i2sspeed, to maximise the DMA speed. Other than that, you have to lower the color depth. Both can be changed in the config

`
HUB75_I2S_CFG mxconfig(
      PANEL_RES_X,   // module width
      PANEL_RES_Y,   // module height
      PANEL_CHAIN    // Chain length
    );
    
mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;
mxconfig.setPixelColorDepthBits(6)
    
dma_display = new MatrixPanel_I2S_DMA(mxconfig);
    `