mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #577: Virtual frame or virtual canvas

K0rkunc, 2024-01-29:
Hello, instead of writing each message to the screen one by one, I want to create one or more virtual screens and be able to switch between them on the physical screen at desired intervals. Is there a way to achieve this?

Lukaswnd, 2024-03-19:
The library has no support for that. You could write a canvas class, like the one below I use. The Idea is to create a Buffer, which you can write your text to and later get the data pointer and use it as an image to draw the whole thing to the matrix.
Pro: - you don't need to interpret the font each time you make it.
Con:  - still need to put each Pixel to the matrix
         - takes up a lot of memory. You basicly have to use external PSRAM to keep multiple and/or large canvas in memory
         
Make sure to use your GFX implementation and replace the datatypes like color_t to uint16_t etc..

class MemoryDisplay : public GFX{

        color_t* m_data = nullptr;

    public:

        MemoryDisplay(dimen_t width, dimen_t height, bool external);
        
        void writePixel(coord_t x, coord_t y, color_t color) override;

        void writeFastHLine(coord_t x, coord_t y, dimen_t w, color_t color) override;
        void fillScreen(color_t color) override;

        inline color_t* getData() const { return m_data; }

};

mrcodetastic, 2024-08-24:
I have a library for this now: https://github.com/mrcodetastic/GFX_Lite

closed 2024-08-24 by mrcodetastic