mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #579: display->print lettering is dropping pixels

prabbit237, 2024-02-02:
I saw something strange in the neighborhood but the Ghostbusters were busy so I'm reporting it here.
========================================
When using the print function on an ESP32 and a HUB75 64x64, some of the pixels are dropping out. This can be seen in the video at:

https://www.youtube.com/watch?v=Z6VJ1vh9FLc

The ESP32 is plugged in via a PCB:
[image]

In the video, watch the row of A's as they drop down the screen. The blue markers are solely there to show 32 lines below the top of the white time text (the white text will drop down in the video as well so the markers will also move.)

When the green row of letters A gets to the 32 row below the white time display, they start dropping out some of the columns (watch the left side of the second letter A. It also happens on # 7, 9 and 10 but the second one is easiest to spot.) When they get to around row 42 or so, they come back on.  Note that the green rectangle never drops out, showing the panel itself seems to be OK.

The dropouts have nothing to do with the fact that it's scrolling (I first saw them when just printing the time and lettering as stationary characters and was trying different things to test the issue.) They have nothing to do with the green bar (added just to make sure it was only letters affected) or the blue lines (added solely to verify that they were dropping at 32 rows below the top of the white text.) And also the green bar is intentionally out of step with the row of A's so that it shows the two aren't related as to the drops.

It ONLY drops the green pixels so if I make the A row all white, the pixels with the issue will be purple (where the red and blue still show.)

Further experiments show that if I set the time row to the following colors and leave the A row green, I get the following results.

255,0,0     OK
0,255,0     OK
0,0,255     OK

255,255,0   drops
255,0,255   drops (but not quite as much)
0,255,255   drops
255,255,255 drops

This first code is what's in the video:

````
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define R1_PIN 26
#define G1_PIN 27
#define B1_PIN 25
#define R2_PIN 12
#define G2_PIN 13
#define B2_PIN 14
#define A_PIN 23
#define B_PIN 19 // Changed from library default
#define C_PIN 5
#define D_PIN 17
#define E_PIN 32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16
#define LDR 35

/*--------------------- MATRIX PANEL CONFIG -------------------------*/
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1  // Total number of panels chained one to another

// Another way of creating config structure
// Custom pin mapping for all pins
HUB75_I2S_CFG::i2s_pins _pins = {
    R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
HUB75_I2S_CFG mxconfig(
    64,                    // width
    64,                    // height
    1,                     // chain length
    _pins,                 // pin mapping
    HUB75_I2S_CFG::FM6126A // driver chip
);

MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup()
{
  /************** SERIAL **************/
  Serial.begin(115200);
  TickType_t xLastWakeTime;
  const TickType_t xFrequency = 150;
  uint8_t i=0;
  uint8_t j=0;
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(60); // 0-255

  for (;;)
  {
    dma_display->clearScreen();
    dma_display->setTextColor(dma_display->color565(255,255,255));
    dma_display->setTextSize(2);
    dma_display->setCursor(0, j);
    dma_display->print("12");
    dma_display->setCursor(28, j);
    dma_display->print("00");
    dma_display->setCursor(52, j);
    dma_display->setTextSize(2,1);
    dma_display->print("A");
    dma_display->setTextSize(1);
    dma_display->setTextColor(dma_display->color565(0,255,0));
    dma_display->setCursor(0, (i%47));
    dma_display->print("AAAAAAAAAA");
    dma_display->fillRect(0,(i+8)%56,64,8,dma_display->color565(0,255,0));
    dma_display->drawFastHLine(0,(32+j)%64,4,dma_display->color565(0,0,255));
    dma_display->drawFastHLine(60,(32+j)%64,4,dma_display->color565(0,0,255));
    i++;
    vTaskDelayUntil(&xLastWakeTime, xFrequency);
    if ((i%64)==0)
        j+=4;
  }
}

void loop()
{
}
````

Simpler code without all of the scrolling, etc. First the image. Note that two of the pixels are fainter than the rest but not dropped completely. They are solid but just much dimmer.
[image]

````
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define R1_PIN 26
#define G1_PIN 27
#define B1_PIN 25
#define R2_PIN 12
#define G2_PIN 13
#define B2_PIN 14
#define A_PIN 23
#define B_PIN 19 // Changed from library default
#define C_PIN 5
#define D_PIN 17
#define E_PIN 32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16
#define LDR 35

/*--------------------- MATRIX PANEL CONFIG -------------------------*/
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1  // Total number of panels chained one to another

// Another way of creating config structure
// Custom pin mapping for all pins
HUB75_I2S_CFG::i2s_pins _pins = {
    R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
HUB75_I2S_CFG mxconfig(
    64,                    // width
    64,                    // height
    1,                     // chain length
    _pins,                 // pin mapping
    HUB75_I2S_CFG::FM6126A // driver chip
);

MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup()
{
    /************** SERIAL **************/
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
    dma_display->begin();
    dma_display->setBrightness8(60); // 0-255
    dma_display->setTextColor(dma_display->color565(255, 255, 255));
    dma_display->setTextSize(2);
    dma_display->setCursor(0, 0);
    dma_display->print("12");
    dma_display->setCursor(28, 0);
    dma_display->print("0");
    dma_display->setCursor(52, 0);
    dma_display->setTextSize(2, 1);
    dma_display->print("A");
    dma_display->setTextSize(1);
    dma_display->setTextColor(dma_display->color565(0, 255, 0));
    dma_display->setCursor(0, 27);
    dma_display->print("AAAAAAAAAA");
}

void loop()
{
}
````

And then here's code with the top row in different colors. In this one, the very bottom-left pixel of the 9th A is actually flickering a bit (it looks like it's fully on in the picture.) I also added two yellow spots where pixels are missing here. If I make the whole upper row 255,0,255, then those two pixels come on but they flicker (the one on the second A is almost full brightness but on the last A, it's just barely visible.)
[image]
````
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define R1_PIN 26
#define G1_PIN 27
#define B1_PIN 25
#define R2_PIN 12
#define G2_PIN 13
#define B2_PIN 14
#define A_PIN 23
#define B_PIN 19 // Changed from library default
#define C_PIN 5
#define D_PIN 17
#define E_PIN 32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16
#define LDR 35

/*--------------------- MATRIX PANEL CONFIG -------------------------*/
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1  // Total number of panels chained one to another

// Another way of creating config structure
// Custom pin mapping for all pins
HUB75_I2S_CFG::i2s_pins _pins = {
    R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
HUB75_I2S_CFG mxconfig(
    64,                    // width
    64,                    // height
    1,                     // chain length
    _pins,                 // pin mapping
    HUB75_I2S_CFG::FM6126A // driver chip
);

MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup()
{
    /************** SERIAL **************/
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
    dma_display->begin();
    dma_display->setBrightness8(60); // 0-255
    dma_display->setTextColor(dma_display->color565(255, 255, 255));
    dma_display->setTextSize(2);
    dma_display->setCursor(0, 0);
    dma_display->print("12");
    dma_display->setCursor(28, 0);
    dma_display->setTextColor(dma_display->color565(0, 255, 255));
    dma_display->print("0");
    dma_display->setTextColor(dma_display->color565(255, 0, 255));
    dma_display->print("0");
    dma_display->setCursor(52, 0);
    dma_display->setTextSize(2, 1);
    dma_display->setTextColor(dma_display->color565(255, 255, 0));
    dma_display->print("A");
    dma_display->setTextSize(1);
    dma_display->setTextColor(dma_display->color565(0, 255, 0));
    dma_display->setCursor(0, 27);
    dma_display->print("AAAAAAAAAA");
}

void loop()
{
}
````

mrfaptastic, 2024-02-02:
mxconfig.clkphase   ?

prabbit237, 2024-02-05:
I tried it with both mxconfig.clkphase=false and with mxconfig.clkphase=true and no difference.

mrcodetastic, 2024-08-24:
I'm closing this issue as it's probably a defective panel.

closed 2024-08-24 by mrcodetastic