mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #582: Arduino: soc/i2s_periph.h Error When Building Any Example

whiteshavedchocolate, 2024-02-04:
### Discussed in https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/discussions/581

<div type='discussions-op-text'>

<sup>Originally posted by **whiteshavedchocolate** February  3, 2024</sup>
I've been using the Arduino SmartMatrix library for a few years, successfully driving different LED modules. I wanted to try out the ESP32-HUB75-MatrixPanel-DMA library, as it has support for the ESP32-S3.

I feel like a moron, but after installing this library in Arduino on my Windows 11 laptop, I cannot successfully build any of the examples. I get this error:
...ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/platforms/esp32/esp32_i2s_parallel_dma.hpp:56:54: fatal error: soc/i2s_periph.h: No such file or directory

I did a search in the "src" folder, and I'm unable to find i2s_periph.h. I'm guessing I'm missing something really basic, but I'm stumped. Anybody know what's up? Thank you.</div>

mrfaptastic, 2024-02-04:
What version of the esp32 board library and and age device did you select in arduino?

Providing a screenshot might be easier

mrfaptastic, 2024-02-04:
It should just compile as is.

What version of the esp32 board library is installed and what target device did you select in arduino?

Providing a screenshot might be easier

whiteshavedchocolate, 2024-02-04:
Ahh, I've been limited to using board version 1.0.4 to be able to work with
the SmartMatrix library. I'm using the WROVER-E (16MB) module.

[image: image.png]
[image: image.png]

Does your library require the ESP32 board library to be 2.0 or higher?

Michael

whiteshavedchocolate, 2024-02-04:
OK, I just updated the ESP32 board library to version 2.0.14, and it
compiled fine! Thank you! I figured it was something elementary!
[image: image.png]

closed 2024-03-03 by mrfaptastic