mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #584: Help displays  clock with results like this the display vibrates

muslihrzl, 2024-02-06:
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/157888752/672de30f-a83c-45d8-a980-f133fba178cc

`
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#include "Font/Digit12x26.h"

class OneEightScanPanel : public VirtualMatrixPanel
  {
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:
    
    VirtualCoords getCoords(int16_t &x, int16_t &y);
  };
  
inline VirtualCoords OneEightScanPanel::getCoords(int16_t &x, int16_t &y) {
    VirtualMatrixPanel::getCoords(x, y); // call to base to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
    }

      coords.x = x + ((y / panelResY) * (virtualResX)) ;
      coords.y = y % panelResY;

  if ( (y & 8) == 0) { 
      coords.x += ((coords.x / 32)+1)*32; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
    }
  else {
      coords.x += (coords.x / 32)*32; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
    }
       coords.y = (y >> 4)*8 + (y & 0b00000111);   
      return coords;
    }

   
  // Panel configuration
  #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
  #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
  
  #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
  #define PANEL_CHAIN 1 //NUM_ROWS*NUM_COLS    // total number of panels chained one to another
    
  // placeholder for the matrix object
  MatrixPanel_I2S_DMA *dma_display    = nullptr;
  OneEightScanPanel   *virtualDisp    = nullptr;

  #include <Wire.h>
  #include <DS3231.h>

  // Object Declarations
  RTClib          RTC;
  DS3231          Clock;
  DateTime        now;
  float           floatnow   = 0;
  uint8_t         daynow     = 0;
  
  /******************************************************************************
   * Setup!
   ******************************************************************************/
  void setup() {

    HUB75_I2S_CFG mxconfig( PANEL_RES_X*2, PANEL_RES_Y/2, PANEL_CHAIN );
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
    dma_display->begin();
    virtualDisp = new OneEightScanPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);
   
    Wire.begin(32,33);
 
  } 
  
  
  void loop() {
  
    dma_display->setBrightness8(10);
    dma_display->clearScreen();
    
    bigClock ();

    updateTime();
    

  } 

void updateTime()
  { now = RTC.now();
    floatnow = (float)now.hour() + (float)now.minute() / 60 + (float)now.second() / 3600;
    daynow   = Clock.getDoW();    // load day Number
  }

void bigClock ()
  {
    static uint16_t    y;
    static uint16_t    s; // 0=in, 1=out              
    static uint32_t   lsRn;
    uint32_t          Tmr = millis();

        virtualDisp->setFont(&Digit12x26);
        virtualDisp->setTextColor(virtualDisp->color444(0,255,0));
        virtualDisp->setCursor(2, 25);
     if (now.hour()<10) virtualDisp->print(0, DEC); // hours
        virtualDisp->print(now.hour(), DEC);
        
        virtualDisp->setFont(&Digit12x26);
        virtualDisp->setTextColor(virtualDisp->color444(0,255,0));
        virtualDisp->setCursor(36, 25);
     if (now.minute()<10) virtualDisp->print(0, DEC); // minute
        virtualDisp->print(now.minute(), DEC);

       if  (now.second() % 2 == 1) // blink dot 
          {
              virtualDisp->drawLine(31, 7, 32, 7, virtualDisp->color444(0,255,0)); 
              virtualDisp->drawLine(30, 8, 33, 8, virtualDisp->color444(0,255,0));
              virtualDisp->drawLine(30, 9, 33, 9, virtualDisp->color444(0,255,0));  
              virtualDisp->drawLine(31, 10, 32, 10, virtualDisp->color444(0,255,0)); 
              
              virtualDisp->drawLine(31, 21, 32, 21, virtualDisp->color444(0,255,0)); 
              virtualDisp->drawLine(30, 22, 33, 22, virtualDisp->color444(0,255,0));
              virtualDisp->drawLine(30, 23, 33, 23, virtualDisp->color444(0,255,0));  
              virtualDisp->drawLine(31, 24, 32, 24, virtualDisp->color444(0,255,0));
           }
           virtualDisp->drawRect(0,0, virtualDisp->width()-0, virtualDisp->height()-0, virtualDisp->color565(255,255,255));
        } 
`

muslihrzl, 2024-02-06:
For code like this

`
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#include "Font/Digit12x26.h"

class OneEightScanPanel : public VirtualMatrixPanel
  {
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:
    
    VirtualCoords getCoords(int16_t &x, int16_t &y);
  };
  
inline VirtualCoords OneEightScanPanel::getCoords(int16_t &x, int16_t &y) {
    VirtualMatrixPanel::getCoords(x, y); // call to base to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
    }

      coords.x = x + ((y / panelResY) * (virtualResX)) ;
      coords.y = y % panelResY;

  if ( (y & 8) == 0) { 
      coords.x += ((coords.x / 32)+1)*32; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
    }
  else {
      coords.x += (coords.x / 32)*32; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
    }
       coords.y = (y >> 4)*8 + (y & 0b00000111);   
      return coords;
    }

   
  // Panel configuration
  #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
  #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
  
  #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
  #define PANEL_CHAIN 1 //NUM_ROWS*NUM_COLS    // total number of panels chained one to another
    
  // placeholder for the matrix object
  MatrixPanel_I2S_DMA *dma_display    = nullptr;
  OneEightScanPanel   *virtualDisp    = nullptr;

  #include <Wire.h>
  #include <DS3231.h>

  // Object Declarations
  RTClib          RTC;
  DS3231          Clock;
  DateTime        now;
  float           floatnow   = 0;
  uint8_t         daynow     = 0;
  
  /******************************************************************************
   * Setup!
   ******************************************************************************/
  void setup() {

    HUB75_I2S_CFG mxconfig( PANEL_RES_X*2, PANEL_RES_Y/2, PANEL_CHAIN );
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
    dma_display->begin();
    virtualDisp = new OneEightScanPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);
   
    Wire.begin(32,33);
 
  } 
  
  
  void loop() {
  
    dma_display->setBrightness8(10);
    dma_display->clearScreen();
    
    bigClock ();

    updateTime();
    

  } 

void updateTime()
  { now = RTC.now();
    floatnow = (float)now.hour() + (float)now.minute() / 60 + (float)now.second() / 3600;
    daynow   = Clock.getDoW();    // load day Number
  }

void bigClock ()
  {
    static uint16_t    y;
    static uint16_t    s; // 0=in, 1=out              
    static uint32_t   lsRn;
    uint32_t          Tmr = millis();

        virtualDisp->setFont(&Digit12x26);
        virtualDisp->setTextColor(virtualDisp->color444(0,255,0));
        virtualDisp->setCursor(2, 25);
     if (now.hour()<10) virtualDisp->print(0, DEC); // hours
        virtualDisp->print(now.hour(), DEC);
        
        virtualDisp->setFont(&Digit12x26);
        virtualDisp->setTextColor(virtualDisp->color444(0,255,0));//green
        virtualDisp->setCursor(36, 25);
     if (now.minute()<10) virtualDisp->print(0, DEC); // minute
        virtualDisp->print(now.minute(), DEC);

       if  (now.second() % 2 == 1) // blink dot 
          {
              virtualDisp->drawLine(31, 7, 32, 7, virtualDisp->color444(0,255,0)); 
              virtualDisp->drawLine(30, 8, 33, 8, virtualDisp->color444(0,255,0));
              virtualDisp->drawLine(30, 9, 33, 9, virtualDisp->color444(0,255,0));  
              virtualDisp->drawLine(31, 10, 32, 10, virtualDisp->color444(0,255,0)); 
              
              virtualDisp->drawLine(31, 21, 32, 21, virtualDisp->color444(0,255,0)); 
              virtualDisp->drawLine(30, 22, 33, 22, virtualDisp->color444(0,255,0));
              virtualDisp->drawLine(30, 23, 33, 23, virtualDisp->color444(0,255,0));  
              virtualDisp->drawLine(31, 24, 32, 24, virtualDisp->color444(0,255,0));
           }
           virtualDisp->drawRect(0,0, virtualDisp->width()-0, virtualDisp->height()-0, virtualDisp->color565(255,255,255));
        } 
`

bug2k19, 2024-02-06:
... I'm new to these things and no expert, but is it because you are clearing the screen and re-drawing every loop rather than updating the screen only if something has changed?

board707, 2024-02-06:
Completely agree with @bug2k19 
@muslihrzl, refresh entire screen only when the minutes change. And redraw the blinking dots without erasing the entire screen - to light up the dots, draw them with the color, and to erase them, draw the dot with the background color.
And also do this only when seconds changed, and not every pass of the loop

muslihrizal, 2024-02-06:
@bug2k19 , @board707  Yes, thanks for the suggestion, I will try it