mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #585: DrawIcon with VirtualMatrixPanel (2X2 structure) is not available (class VirtualMatrixPanel' has no member named drawIcon)

Abdulrahman-owayj, 2024-02-11:
hi, Im using 4 panels with (2x2) structure, it works great with (fillRect, print , etc), but I think there is no drawicon property for the virtualDisplay right ? i get ( class VirtualMatrixPanel' has no member named drawIcon )

is there a solution for it?

and I appreciate your effort for this library, You did a very fantastic job!

Abdulrahman-owayj, 2024-02-14:
Well, i tried your commented code and it works great, im not sure why its not released.

the only problem is that when i use drawIcon more than one time the DRAM is filled and i get an error, i need to find a solution to release resources after every call

thank you for your effort

Lukaswnd, 2024-03-19:
The VirtualMatrixPanel itselfe is not ment to have any GFX functionallity. You are suppost, to use Adafruit_GFX, GFX, or any other GFX implementation, you have to combine yourselfe.

mrcodetastic, 2024-08-24:
I will be removing drawIcon from the library at some point.

closed 2024-08-24 by mrcodetastic