mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #591: Problem with  32x32 RGB

LeonidDnipro, 2024-02-21:
I have a 32x32 RGB matrix model "P6(3535)32x32-8S-V1.6" HUB75 interface, made on 24 ICN5020B chips, one 74N138 decoder whose output goes to 16 chips 4953 (2 pieces of P-Channel MOSFET in one package).
It was not possible to draw a test line diagonally or pixel to pixel fill the matrix. By searching, I realized that the matrix is addressed as 64x16.
In order to fill the matrix point by point, I wrote a program that selects coordinates according to the picture.
In the picture, arrows show the electrical connection of the ICN5020B chips blocks (UB1, UR1, UG1, marking of chip on the matrix board)
[image]
The configuration I used is below:
`
// Configure for your panel(s) as appropriate!
#define PIN_E 32
#define PANEL_WIDTH 64 
#define PANEL_HEIGHT 16          // Panel height of 64 will required PIN_E to be defined.

#ifdef VIRTUAL_PANE
 #define PANELS_NUMBER 1         // Number of chained panels, if just a single panel, obviously set to 1
#else
 #define PANELS_NUMBER 1  //2         // Number of chained panels, if just a single panel, obviously set to 1
#endif

#define PANE_WIDTH PANEL_WIDTH * PANELS_NUMBER
#define PANE_HEIGHT PANEL_HEIGHT
#define NUM_LEDS PANE_WIDTH*PANE_HEIGHT

MatrixPanel_I2S_DMA *dma_display = nullptr;
uint8_t color1 = 0;  

void setup(){
  Serial.begin(115200);
//  Serial.println("Starting pattern test...");
  HUB75_I2S_CFG::i2s_pins _pins={R1, G1, B1, R2, G2, B2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK};
  HUB75_I2S_CFG mxconfig(
    PANE_WIDTH,
    PANE_HEIGHT,
    PANELS_NUMBER,
    _pins
  );

  mxconfig.latch_blanking=1;
  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
    // OK, now we can create our matrix object
  mxconfig.i2sspeed =HUB75_I2S_CFG::HZ_10M;
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display ->setBrightness8(100);`

 How do I configure it so that I can fully use the library?

Lukaswnd, 2024-03-19:
You first want to get the pixel mapping correct.
Have you checked out the Four_Scan_Panlen example?

LeonidDnipro, 2024-03-22:
I managed to sequentially fill the led matrix horizontally and then vertically. But for this I used the remap function for X and Y.
The video shows sequential filling. In the video, the last episode is filled pixel by pixel without using a remap.
I provide the code below. Physically, the matrix is ​​32 by 32 pixels, but logically in the code I used 64 by 16. As I drew in the table in one row, four chip ICN5020B  with 16 outputs each are connected in series, which gives these 64 pixels. 

Link my video: https://youtu.be/yloUcVTsIZk

`#include <Arduino.h>

#ifdef IDF_BUILD
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <driver/gpio.h>
#include "sdkconfig.h"
#endif

#ifdef VIRTUAL_PANE
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#else
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#endif

/*  Default library pin configuration for the reference
  you can redefine only ones you need later on object creation*/
#define R1 11  
#define G1 10
#define B1 9  
#define R2 13  
#define G2 12
#define B2 14  
#define CH_A 6
#define CH_B 7
#define CH_C 5
#define CH_D 8
#define CH_E -1 // assign to any available pin if using panels with 1/32 scan
#define CLK 16
#define LAT 4
#define OE 15

// Configure for your panel(s) as appropriate!
#define PIN_E 32
#define PANEL_WIDTH 64 
#define PANEL_HEIGHT 16 
#define PANELS_NUMBER 1

#define PANE_WIDTH PANEL_WIDTH * PANELS_NUMBER
#define PANE_HEIGHT PANEL_HEIGHT
//#define NUM_LEDS PANE_WIDTH*PANE_HEIGHT

MatrixPanel_I2S_DMA *dma_display = nullptr;
uint16_t color;
int x; int y; int bl; int i,j;

int remap_x(int x1){
	
//  0 pixel  <= x < 32 pixel	
 if(x1<16 ){
   if(y<8  || (y>=16 && y<24) )  x1=x1;
   if((y>=8 && y<16) || (y>=24 && y<32)){
       x1=x1+16;
       }
  return x1;       
}
  if(x1>=16 &&  x1<32){
    if(y<8  || (y>=16 && y<24) )  x1=x1+0x10;
    if((y>=8 && y<16) || (y>=24 && y<32) )  x1=x1+0x20;
  return x1;  
  }

//  32 pixel  <=x < 63 pixel
 if(x1<48 ){
   if(y<8  || (y>=16 && y<24) )  
       x1=x1 + 32;
   if((y>=8 && y<16) || (y>=24 && y<32)){
       x1=x1+0x30;
       }
  return x1;       
}
  if(x1>=48 &&  x1<64){
    if(y<8  || (y>=16 && y<24) )
       x1=x1+0x30;
    if((y>=8 && y<16) || (y>=24 && y<32) )  
       x1=x1+0x40;
  return x1;  
  }
return x1;

}
// for one row panel led
int remap_y(int yr){
  if(yr<16) 
     yr=yr & 0x7;
  if(yr>=16 && yr<32) 
     yr=(yr & 0x7)+8;
return yr;
}

//_______________________________________________
void setup(){

  Serial.begin(115200);
  Serial.println("Starting test");

  HUB75_I2S_CFG::i2s_pins _pins={R1, G1, B1, R2, G2, B2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK};
  HUB75_I2S_CFG mxconfig(
    PANE_WIDTH,
    PANE_HEIGHT,
    PANELS_NUMBER,
    _pins
  );
  //mxconfig.driver = HUB75_I2S_CFG::ICN2038S;   // for panels using FM6126A chips
  mxconfig.latch_blanking=1;
  mxconfig.clkphase = false; 
  mxconfig.i2sspeed =HUB75_I2S_CFG::HZ_10M;

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display ->setBrightness8(100);
}

//========================================================================================
void loop(){

dma_display->clearScreen();
 color = dma_display->color565(30, 65, 65); 
    for ( y = 0; y < 32; y++) {
        for ( x = 0; x < 32; x++) { //32
            dma_display-> drawPixel(remap_x(x), remap_y(y), color); 
          //  Serial.print("F");Serial.print(x);   Serial.print("-");  Serial.print(y);   Serial.print("= "); 
          //  Serial.print("T");Serial.print(remap_x(x));   Serial.print(":");  Serial.print(remap_y(y));   Serial.print("|"); 
            delay(15);
        }
    }
delay(500);

//draw pixel to pixel vertical
dma_display->clearScreen();
 color = dma_display->color565(0, 70, 100); 
    for ( x = 0; x < 32; x++) {
        for ( y = 0; y < 32; y++) {
            dma_display-> drawPixel(remap_x(x), remap_y(y), color); 
        //    Serial.print("F");Serial.print(x);   Serial.print("-");  Serial.print(y);   Serial.print("= "); 
        //    Serial.print("T");Serial.print(remap_x(x));   Serial.print(":");  Serial.print(remap_y(y));   Serial.print("|"); 
            delay(15);
        }
    }

Serial.println("draw line "); 
delay(2000);

dma_display->clearScreen();
 color = dma_display->color565(30, 25, 85); 
    for ( y = 0; y < 16; y++) {
        for ( x = 0; x < 64; x++) { //32
            dma_display-> drawPixel(x, y, color); 
            delay(20);
        }
    }

  delay(2000);
  dma_display->clearScreen();
  Serial.println("\n====");
}`