mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #595: Color issue - Red is bright, green is dim, blue is nonexistant

jrlambs, 2024-02-28:
I have a weird issue.  Title kinda describes it all.  In my program I'm loading I cycle through red, green, blue then show some text in white.  The Red shows up nice and bright, the green shows up but maybe like 1/10th as bright as the red.  and the blue as far as I can tell doesn't show up at all.  When the "White" text shows up it ofc is red because the green/blue components of it aren't offsetting the colors.  

I've checked and re-checked wiring/pins about 30 times.  I've tried a couple of different esp32 boards and still come back to the same issue.  Any thoughts on anything I can troubleshoot?

mrfaptastic, 2024-02-28:
What chips are at the back of the board?

jrlambs, 2024-02-28:
I see a few chips:

One labeled:
ICN2037BP
D10266NB

One labeled
RUC7258D
0AE09836

One Labeled:
MW245B
BZCBOTAB12

board707, 2024-02-28:
ICN2037BP - driver,  RUC7258D - multiplexer
Nothing unusual or incompatible.

jrlambs, 2024-02-28:
Hmm I've tried 3 panels and all behave the same.    I'm using this board:
https://github.com/hallard/WeMos-Matrix-Shield-DMA
with this esp controller (I've tried 2 different ones esp boards):
https://www.aliexpress.us/item/3256801435529393.html?spm=a2g0o.order_list.order_list_main.87.711c1802wDC3eN&gatewayAdapt=glo2usa

nothing else hooked up to the board than the esp controller and the female headers on the back.  Can you think of any ports I have mapped wrong that might cause this issue?  

This is my config:

```
#define R1_PIN  25
#define B1_PIN  27
#define R2_PIN  14
#define B2_PIN  13
#define A_PIN   23
#define C_PIN   5
#define CLK_PIN 16
#define OE_PIN  15
#define G1_PIN  26
#define G2_PIN  12
#define B_PIN   19
#define D_PIN   17
#define E_PIN   -1
#define LAT_PIN 4
#define PANEL_RES_X 32
#define PANEL_RES_Y 32
#define PANEL_CHAIN 1

HUB75_I2S_CFG mxconfig(
  PANEL_RES_X,   // module width
  PANEL_RES_Y,   // module height
  PANEL_CHAIN    // Chain length
);

mxconfig.gpio.a = A_PIN;
mxconfig.gpio.b = B_PIN;
mxconfig.gpio.c = C_PIN;
mxconfig.gpio.d = D_PIN;
//mxconfig.gpio.e = E_PIN;
mxconfig.gpio.r1 = R1_PIN;
mxconfig.gpio.g1 = G1_PIN;
mxconfig.gpio.b1 = B1_PIN;
mxconfig.gpio.r2 = R2_PIN;
mxconfig.gpio.g2 = G2_PIN;
mxconfig.gpio.b2 = B2_PIN;
mxconfig.gpio.clk = CLK_PIN;
mxconfig.gpio.oe = OE_PIN;
mxconfig.gpio.lat = LAT_PIN;
  
mxconfig.clkphase = false;
mxconfig.driver = HUB75_I2S_CFG::FM6126A;
delay(10);
// Display Setup
dma_display = new MatrixPanel_I2S_DMA(mxconfig);

```

jrlambs, 2024-02-28:
Is there maybe a different driver I should be using?  If I comment out the driver line basically every color shows as red... not even light green anymore :(

jrlambs, 2024-02-29:
Ah I'm an idiot.  My power supply was dead.. so it was just using whatever power it was getting from the esp board.  Plugged it into a good power supply and I have all my colors!

closed 2024-02-29 by jrlambs