mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #601: S-P4-2020-A3 origin pixel issue

Parthenergycloud, 2024-03-06:
Hello,
I have used P4-2020_A3 RGB Panel on my side.
I order this panel from Here,
[https://evelta.com/rgb-full-color-led-matrix-panel/](url)

I have connected 4 panels in  a zigzag manner with esp32 HUB75 shield.
*****************************************************
Chain of 4x 64x32 panels for this example:
+---------+---------+
|    2          |    1        |
|                | (ESP32) |
+---------+---------+
|    4          |    3        |
|                |              |
+---------+---------+
Here is the image of HUB75 with esp32.
[image]

I used  ESP32-HUB75-MatrixPanel-DMA library for connecting 4 panels and i used Chainedpanels example form library.
In that i found some error in displaying.

→ Error i facing for display something in Chain of 4x 64x32 panels
> First line of the display is shifted last line you can see in below image.

[image]

[image]

→ In this image you can see that my first line is shifted in last line and i setup panels according to numbers.

→ Second thing is panal 4 first line is shifted in panel 1 last line and panel 2 first line shifted to panel 3 first line.

"I want solution of this problem of shifting line"

Here, is the example code which i used.
`#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
// #include "FreeSerif9pt7b.h"

#define PANEL_RES_X 64
#define PANEL_RES_Y 32
#define NUM_ROWS 2
#define NUM_COLS 2
#define PANEL_CHAIN NUM_ROWS * NUM_COLS

#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN_ZZ

MatrixPanel_I2S_DMA *dma_display = nullptr;
VirtualMatrixPanel *virtualDisp = nullptr;

int scrollPos = 0; // Initial position for text scrolling

void setup() {
HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,
    PANEL_RES_Y,
    PANEL_CHAIN
);

dma_display = new MatrixPanel_I2S_DMA(mxconfig);
dma_display->setBrightness8(255);
dma_display->begin();

virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

virtualDisp->setPhysicalPanelScanRate(NORMAL_TWO_SCAN);
// virtualDisp->setFont(&FreeSerif9pt7b);
virtualDisp->setTextColor(virtualDisp->color565(0, 0, 255));
virtualDisp->setTextSize(2);
}

void loop() {
// Clear previous text
virtualDisp->fillScreen(virtualDisp->color565(0, 0, 0));

// Print text at the current scroll position
virtualDisp->setCursor(scrollPos, 0);
virtualDisp->print("HELLO");

// Increment scroll position
scrollPos--;

// Reset scroll position if text has scrolled completely off the display
if (scrollPos < -(virtualDisp->width())) {
    scrollPos = virtualDisp->width();
}

// virtualDisp->swapBuffers(false);
virtualDisp->setTextWrap(false);
    // Delay to control scrolling speed
    delay(20); // Adjust as needed
}`

Here is video for running above code in the panel.

> In video you can see that when my hello is reach to first line i moves to last line of 2nd ROW display.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/assets/110959437/9dc73fe5-3174-473f-aad8-2717a7c54963

Anyone has a solution for this problem? @mrfaptastic 
Please help me to solve this problem.

mrfaptastic, 2024-03-06:
Clkphase?

mrfaptastic, 2024-03-06:
Perhaps similar to this post:

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/545#issuecomment-1977823058

Parthenergycloud, 2024-03-06:
> Clkphase?
i did not set any clkphase now so it is required?.

closed 2024-03-06 by Parthenergycloud