mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #604: Battery Vs. Panel

wamk9, 2024-03-13:
Hi Everyone!

I have a question about the power of that panels, can i use a step up or something with a battery 18650? If yes, have a module example to adquire to this situation?

Thanks!

mrfaptastic, 2024-03-15:
I have often used a usb power pack to drive these panels.  

High capacity powerpack is required and one that can output 2A.

Lukaswnd, 2024-03-18:
I do not recommend to only use a boost converty, this has no discharge protection an will destroy your batteries. There is a common 4 cell module, which is sold on everyones favourite china supplyer. It has 5V 3A output.
I have had the experience that the enery consumptionof the panels on the sellers is not that bad. But they assume full white on each pixel with maximum brightess. Which you never do.
From my experience, they use around 50% of the enery, when setting the brightness to 80% and not using white, as color, but rather mixed colored, with only 2 of the 3 colors. And around 25% with 20% brightness. So you can do the mathyourselfe.

An one more tipp. If the powersupply is too weak, and the esp crahes due to low energy provided (Brownout detector). It might happen, that It crashes over and over again, because the esp keeps its pinstate, which leads to relighting of the same leds, that made it crash. So make sure to disconnect the panels from the power supply