mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #605: Help me with the 32x32-8s panel

plmokn1299, 2024-03-15:
Hello, I am a beginner.
I'm running a translator. Please understand if the grammar is weird.
I have a 32x32-8s panel, and I have an adafruit Matrix Portal m4 board. I thought it was a 16s panel and used a few examples, but it didn't work properly, so I looked it up and it was an 8s panel from China. I was really happy to find this library, but this one doesn't work as well. How do I use the Matrix Portal m4 board?

mrfaptastic, 2024-03-15:
Impossible to help without photos or videos. 

Create a sketch that lights up each pixel one by one starting from the top left pixel.

plmokn1299, 2024-03-18:
I selected the matrix total m4 board in the arduino IDE and redefined the pin.
When you include a library, the following error appears.
[image]

Lukaswnd, 2024-03-19:
Make sure to update your Libraries and the arduino_esp32 base lib.

plmokn1299, 2024-03-25:
Even if you update it, the result is the same.

Lukaswnd, 2024-03-26:
Which Arduino-ESP32 Version are you running? Do the 'basic' example scetches work?
Try to add the line 
#include <esp_err.h>
at the top of your .ino file. And check if the error is exactly the same.

plmokn1299, 2024-03-28:
The arduino ESP32 version is 2.0.13, but I installed the Adafruit SAMD Boards and selected matrix portal m4. The board manager URL used https://adafruit.github.io/arduino-board-index/package_adafruit_index.json .
The default example works fine and the same problem appears when you add #include <esp_err.h>.
Pin values were #defined to use the Matrix Portal4 board.

Lukaswnd, 2024-04-02:
One thing I realised just now:
The Matrix Portal M4 has an ESP32 (classic) just as wifi controller.
The Matrix Portal S3 has an ESP32 S3 as the main an only chip.

So on the M4 you are not supposed to program the esp. You are actually porgramming the ATSAMD51J19 Cortex. You can't use the onboard esp to controll the Matrix. Therefore this library is not suited for the Matrix Portal M4 and you have to use the adafruit implementation.

plmokn1299, 2024-04-08:
Thank you.
I was doing what I couldn't.
Even if I use the adafruit protomatter library, I can't use it because it's only for 16scan.
Please teach me the way.