mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #608: P3 (resolution 104 * 52) panel display position incorrect

weny1212, 2024-03-29:
I used a P3 panel with a size of 320 * 160 and a resolution of 104 * 52. I tested it using Example 1 SimpleTestShapes and found that the horizontal position of both graphics and text was incorrect. The left side was offset by about 8 pixels to the right. That's why, thank you! The parameter configuration is as follows:
#define PANEL_RES_X 104   //64     
#define PANEL_RES_Y 52      //32     
#define PANEL_CHAIN 1     
[image]
[image]
[image]
[image]

weny1212, 2024-03-29:
The chip is ICND2046AP

board707, 2024-03-29:
Oh, it is a very weird (or interesting) panel :) I have worked with a similar one.
Each icdn2046 driver has a 16 outputs. Each panel row has a 7 x 16bit drivers, which gives you 112 pixels.
Since the panel is only 104 pixels wide, the 8 pins in the middle of the driver chain 
are not connected to anything. 
When loading data, you have to insert zeros into each row.

weny1212, 2024-03-29:
> Oh, it is a very weird (or interesting) panel :) I have worked with a similar one. Each icdn2046 driver has a 16 outputs. Each panel row has a 7 x 16bit drivers, which gives you 112 outputs. Since the panel is only 104 pixels wide, the 8 pins in the middle of the driver chain are not connected to anything, which leads to the loss of some pixels in the picture. To prevent this, you have to insert zeros into each row when loading data.

Thank you very much. As I am not very familiar with the “ESP32-HUB75 Matrix Panel DMA” library, can you tell me where to fix this issue?

board707, 2024-03-29:
> As I am not very familiar with the “ESP32-HUB75 Matrix Panel DMA” library,

Unfortunately, me too. So we should wait the author, @mrfaptastic, answer the question

weny1212, 2024-03-29:
Hope the author can provide assistance， @mrfaptastic，thanks

mrfaptastic, 2024-03-29:
Doesn't seem like a library issue to me. Given the display works well and is outputting pixels, wouldn't you just make a wrapper in your sketch to accommodate for this weird "non visible" pixels?

Set the display resolution which includes the non existent pixels per whatever @board707 was saying, then adjust for this on your sketch, if that makes sense.

board707, 2024-03-29:
> Doesn't seem like a library issue to me.

Sure, it's definitely not the library's fault.

weny1212, 2024-03-29:
> Doesn't seem like a library issue to me. Given the display works well and is outputting pixels, wouldn't you just make a wrapper in your sketch to accommodate for this weird "non visible" pixels?
> 
> Set the display resolution which includes the non existent pixels per whatever @board707 was saying, then adjust for this on your sketch, if that makes sense.
> 
> Aka. set your panel resolution in the sketch to 114, not 104, and then figure out which pixels are missing and adjust for this in your pixel mapping routine.

Yes, there is no problem with the library itself. How to match the resolution of these anomalies? I don't know where to modify the adaptation in the library now. Can you give me some guidance? thanks

jkandasa, 2024-08-16:
@mrcodetastic thanks for the great library !!

@mrcodetastic @board707 thanks for the detailed information!! I have a similar display and your information helps me to fix this.

@weny1212 I have created custom class to fix this as follows,
```cpp
class Custom_MatrixPanel_I2S_DMA : public MatrixPanel_I2S_DMA {
 public:
  Custom_MatrixPanel_I2S_DMA(const HUB75_I2S_CFG &opts) : MatrixPanel_I2S_DMA(opts) {}

  inline void drawPixel(int16_t x, int16_t y, uint16_t color) {
    if (_rotation == 0 || _rotation == 2) {
      if (x >= _deadPixelOffset) {
        x += 8;
      }
    } else {
      if (y >= _deadPixelOffset) {
        y += 8;
      }
    }

    MatrixPanel_I2S_DMA::drawPixel(x, y, color);
  }

  inline void drawPixelRGB888(int16_t x, int16_t y, uint8_t r, uint8_t g, uint8_t b) {
    if (_rotation == 0 || _rotation == 2) {
      if (x >= _deadPixelOffset) {
        x += 8;
      }
    } else {
      if (y >= _deadPixelOffset) {
        y += 8;
      }
    }

    MatrixPanel_I2S_DMA::drawPixelRGB888(x, y, r, g, b);
  }

  // dead pixels offset can be in different place based on the display rotation
  // update it, if user updates the display rotation
  inline void setRotation(uint8_t r) {
    MatrixPanel_I2S_DMA::setRotation(r);
    rotation = (r & 3);
    switch (rotation) {
      case 0:
        _deadPixelOffset = 56;
        _rotation = 0;
        break;
      case 2:
        _deadPixelOffset = 48;
        _rotation = 2;
        break;
      case 1:
        _deadPixelOffset = 48;
        _rotation = 1;
        break;
      case 3:
        _deadPixelOffset = 56;
        _rotation = 3;
        break;
    }
  }

 private:
  int8_t _deadPixelOffset = 56;
  uint8_t _rotation = 0;
};
```

and on your actual code use `Custom_MatrixPanel_I2S_DMA` in the place of `MatrixPanel_I2S_DMA`,
```cpp
// do not use 104, (112 = 104+8)
#define PANEL_RES_X 112  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 52   // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1    // Total number of panels chained one to another

Custom_MatrixPanel_I2S_DMA *dma_display = nullptr;

// in setup()
  HUB75_I2S_CFG mxconfig(
      PANEL_RES_X,  // module width
      PANEL_RES_Y,            // module height
      PANEL_CHAIN            // Chain length
  );

  dma_display = new Custom_MatrixPanel_I2S_DMA(mxconfig);
```

board707, 2024-08-16:
@jkandasa 
My congratulations:)
I see only small drawback in the solution - it seems that is for a single panel only.

jkandasa, 2024-08-16:
@board707 thanks for the review.
Currently I have only single panel. hence I have tested only with single panel
you are right, we have to rewrite the pixel calculation, if we have multiple panel chained.

mrcodetastic, 2024-08-16:
Thank you for your kind words @jkandasa.

Well done with the solution you developed!