mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #609: P5 screens in chains, 3.0.9 and 3.0.10 horizontal misalignment

zvonko32, 2024-03-29:
When I installed version 3.0.10 I noticed a horizontal shift when the screens are connected in chains (2x2), the same is with version 3.0.9.,
on 3.0.8 everything displays correctly.

p.s. great library, big thanks to the author.

[image]

zvonko32, 2024-03-29:
Now I see that this problem could be Issue "2: Vertical output isn't aligned", I will test tomorow.

closed 2024-04-18 by zvonko32

mrcodetastic, 2024-04-18:
Yeah. I changed the default as having it the other way caused issues with more than 50% of the users of this library.

mrcodetastic, 2024-04-18:
Please feel free to raise a pull request for where you see this error.