mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #619: 3 RGB Channels

Chowdish, 2024-04-16:
I've recently acquired some matrix displays. and the library is kinda working. 

However, my display has r3, g3, b3. And the driver is not showing these correctly. just dead leds.

I know this display is not officially supported. but i thought i might give it a shot anyways. 

The chips on the display is: 

RUC7258D

CFD435AQP9

Some images:

[image]
[image]
[image]

mrcodetastic, 2024-04-17:
The 'CFD435AQP9' is a 'PWM' chipset that means these panels operate differently and not compatible/related to this library. 

There are many variations of these types of panels and unlike the panels this library drives which are essentially just a long chain of 'dumb' shift registers, the chips on this panel are basically a framebuffer that take pixel data and then computes the PWM output frequency.

I was given a bunch of similar eWaste panels and spent weeks trying to reverse engineer how it worked with varied success. If you have the datasheet and code then

https://github.com/mrcodetastic/ESP32S3-MBI5153-DMG1075-DMA

Somebody else wrote a custom library based on this libarary for yet another chipset:

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/discussions/324

CFD435AQP9 sounds new, so you'll need to write your own code after reading the datasheet.

mrcodetastic, 2024-04-17:
I would be interested to see video of how it operates with this library as I would have expected garbled output only. Only by luck I would have thought it would show anything at all.

Chowdish, 2024-04-17:
> I would be interested to see video of how it operates with this library as I would have expected garbled output only. Only by luck I would have thought it would show anything at all.

I’ll try this library later, and send some videos of it. 

And thanks for the help btw, i appreciate it.

mrcodetastic, 2024-04-17:
If your in Europe or the UK and have a lot of these panels then feel free to shoot me one and if i get bored id love to reverse engineer

Chowdish, 2024-04-17:
I'm in Denmark, so i might.

How would you use the library you mentioned above?

mrcodetastic, 2024-04-17:
I would have to hack the library I've written for another pwm chip and try make it work for this one. 

I'm in the UK.

mrcodetastic, 2024-04-18:
Option #2 - you invite me over for a weekend in Denmark and we code the solution up and then get on the 🍺's.

Don't worry, I'll get my own accommodation.

Chowdish, 2024-04-20:
Im having issues using the library you mentioned, is it correctly assumed i
have to use platformio?. Do you have discord, then we could discuss how I
might get this working?

Chowdish, 2024-04-20:
Also here is the video i promised 👍

lør. 20. apr. 2024 kl. 09.56 skrev Kristoffer Joensen ***@***.***

board707, 2024-04-22:
> However, my display has r3, g3, b3.

Hi
What is panel's dimensions in pixels?

Chowdish, 2024-04-22:
> > However, my display has r3, g3, b3.
> 
> Hi What is panel's dimensions in pixels?

I believe its 98x110

board707, 2024-04-22:
> its 98x110

How it can be? If it has a three RGB channels - the height maust be a factor of three, isn't it?

Chowdish, 2024-04-22:
> > its 98x110
> 
> How it can be? If it has a three RGB channels - the height must be a factor of three, isn't it?

I counted wrong, its 96x110

board707, 2024-04-22:
> its 96x110

Which of digits a height and width here?
Could you please provide a clear photo of the entire panel from the rear?

Chowdish, 2024-04-22:
> > its 96x110
> 
> Which of digits a height and width here? Could you please provide a clear photo of the entire panel from the rear?

[image]

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/86896364/5fb7f7ff-716e-435c-98f9-795f24bf753f

board707, 2024-04-22:
Thank you
Unfortunately, I can't see the video.
On the photo it looks like 96 width X 110 height panel.
Because there is only one connection socket, it seems that the panel doesn't support chaining.

board707, 2024-04-22:
The main problem is the driver. As mentioned @mrcodetastic  above:

>The 'CFD435AQP9' is a 'PWM' chipset that means these panels operate differently and not compatible/related to this library.

In addition with 3 RGB channels and, assumedly, without ability to chaining,  it seems to much efforts to make this panel works with the library.

mrcodetastic, 2024-04-22:
This conversation is repeating itself. I already noted it's a SPWM panel which has nothing to do with this library. 

Please refer to a discussion topic on S-PWM panels.

You're going to read the data sheet on the SPWM chip and reverse engineer through trial and error. I tried to create code for some other garbage SPWM panels I was given. Somebody else wrote another library. Search across GitHub to see if somebody hasn't already written code for this chip.

mrcodetastic, 2024-04-22:
This conversation is repeating itself. I  noted it's a SPWM panel which has nothing to do with this library. 

Please refer to a discussion topic on S-PWM panels.

You're going to read the data sheet on the SPWM chip and reverse engineer through trial and error. I tried to create code for some other garbage SPWM panels I was given, and it was a right pain in the ass.

Search across GitHub to see if somebody hasn't already written code for this chip.

Chowdish, 2024-04-22:
Cant find anything with this exact chip. 

But i can get some output on the display, so something in this library is working, scrambled, but working none the less.

So you guys think its a lost cause? I dont have the knowledge to reverse engineer this, im just an electrician. 

Ive got 15 of these panels though 🥲

board707, 2024-04-22:
If you have 15 panels, how do you plan to use them if it not support chaining?

Chowdish, 2024-04-22:
I have no idea, but something tells me they do.

https://www.fabuluxled.com/thinpad-series_p20.html

Im pretty sure this is the product it came from, it uses “Thinpad” in the name. And it looks similar, so maybe.

mrcodetastic, 2024-04-22:
> Cant find anything with this exact chip. 
> 
> But i can get some output on the display, so something in this library is working, scrambled, but working none the less.

It's because the library is sending binary to the chips on this panel, but it's completely in the wrong format so it's garbage. I'm amazed it's actually showing anything at all. 

> So you guys think its a lost cause? I dont have the knowledge to reverse engineer this, im just an electrician. 

Pretty much. I can't find a dataset online about this chip so it'll be difficult to write any code for it. 

> Ive got 15 of these panels though 🥲

I'll send you some dkk and you post one my way? Absolutely no guarantees I will have time or be able to figure out how it works, but if I get bored will try out of curiosity.