mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #629: Font change

zch9288, 2024-05-09:
Hello author
 I am from China, the following text is from the machine translation。
When using this library in China, you need to display Chinese characters，
Now in order to change the font, I need to change the file，《ESP32-VirtualMatrixPanel-I2S-DMA.h》，#include <Fonts/XXXXX.h>andthis->display->setFont(&XXXXX),It's very inconvenient.
The Library currently relies on the Adafruit GFX Library，At the same time, there is a U8g2 for Adafruit GFX, we can call U8G2 font。
Can you add this function to facilitate the use of any font。

 Thank you

board707, 2024-05-09:
> Now in order to change the font, I need to change the file，《ESP32-VirtualMatrixPanel-I2S-DMA.h》

You don't need to change the library. 
As far I can see in the library code, you can add
```
#include <Fonts/XXXXX.h>
```
to your sketch and call 
```
display->setFont(&XXXXX),
```
directly from the code.

zch9288, 2024-05-09:
Thank you for your answer, this is really useful, but these work well for Western text repositories because they are relatively small,But for Chinese, it's not enough, like in U8G2, <const uint8_t u8g2_font_wqy16_t_gb2312[308472] U8G2_FONT_SECTION("u8g2_font_wqy16_t_gb2312") >.SO,Therefore, instead of using the ***.H file, the ***.C file is used, so the method just now cannot be used

board707, 2024-05-09:
In your first comment you wrote that you have to add #include <Font.h> to the library and nothing about including the .c file.
If you can do it with library - you could do it to the skecth as well.
Or provide the minimal code example how do you use U8G2 fonts with this library.

zch9288, 2024-05-10:
No C file is added, which is my problem, because the current library, if you want to change the font, only add H file
When using 

 WS2812's library files previously.

#include <Adafruit_GFX.h>
#include <Adafruit_NeoMatrix.h>
#include <Adafruit_NeoPixel.h>
#include <U8g2_for_Adafruit_GFX.h>
U8G2_FOR_ADAFRUIT_GFX u8gf;

void setup() {
  Serial.begin(115200);
  EEPROM.begin(512);
  matrix.begin();
  _**u8gf.begin(matrix);**_  //This is increasing
  matrix.setTextWrap(false);
  u8gf.setFont(u8g2_font_wqy15_t_gb2312);
}

When I added this statement u8gf.begin(matrix),I can just use it

    u8gf.setFontMode(1);               //1使用u8g2透明模式(0是默认的)
    u8gf.setFontDirection(2);          //2从左到右(0是默认值)2:14,1 0:0,14
    u8gf.setForegroundColor(colourC);  //字体颜色
    u8gf.setCursor(14, 1);             //正常起始点14,1;向右移动一半,7,1
    u8gf.print(ch2); 

Now when I use this method 

#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#include <U8g2_for_Adafruit_GFX.h>
U8G2_FOR_ADAFRUIT_GFX u8gf;

void setup() {
  if (not dma_display->begin())
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
  virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);
  u8gf.begin(virtualDisp);
}

report an error：D:\我的资料库\Documents\Arduino\ChainedPanelsESP32S\ChainedPanelsESP32S.ino: In function 'void setup()':
D:\我的资料库\Documents\Arduino\ChainedPanelsESP32S\ChainedPanelsESP32S.ino:201:25: error: no matching function for call to 'U8G2_FOR_ADAFRUIT_GFX::begin(VirtualMatrixPanel*&)'
In file included from D:\我的资料库\Documents\Arduino\ChainedPanelsESP32S\ChainedPanelsESP32S.ino:18:
d:\�ҵ����Ͽ�\Documents\Arduino\libraries\U8g2_for_Adafruit_GFX\src/U8g2_for_Adafruit_GFX.h:152:10: note: candidate: 'void U8G2_FOR_ADAFRUIT_GFX::begin(Adafruit_GFX&)'
     void begin(Adafruit_GFX &gfx) { u8g2.gfx = &gfx; }
          ^~~~~
d:\�ҵ����Ͽ�\Documents\Arduino\libraries\U8g2_for_Adafruit_GFX\src/U8g2_for_Adafruit_GFX.h:152:10: note:   no known conversion for argument 1 from 'VirtualMatrixPanel*' to 'Adafruit_GFX&'

board707, 2024-05-10:
> U8G2_FOR_ADAFRUIT_GFX u8gf;

As far as I see, your problem is not with the font. The library `U8G2` is a separate product, which has nothing to do with current library. You can't use `U8G2` with VirtualMatrixPanel .

zch9288, 2024-05-15:
#include <U8g2_for_Adafruit_GFX.h>
U8G2_FOR_ADAFRUIT_GFX u8g2Fonts;

  u8g2Fonts.begin(*virtualDisp);
  u8g2Fonts.setFontMode(1);
  u8g2Fonts.setFontDirection(0);
  u8g2Fonts.setFont(u8g2_font_wqy16_t_gb2312); 
  u8g2Fonts.setForegroundColor(TFT_RED);
  u8g2Fonts.drawUTF8(80, 61, "中文");  

I tried it, and that's it