mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #631: Artifacts when playing a GIF file on a uniform background

Olejan, 2024-05-18:
I converted a [Kinetic video](https://www.youtube.com/watch?v=-9WS0Nrg57Y) from YouTube into a gif file with my screen size 128 * 64 (I used this [service](https://ezgif.com/video-to-gif)). I display this gif file on the screen as shown in the [AnimatedGIFPanel_SPIFFS](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/AnimatedGIFPanel_SPIFFS) example. The video plays, but leaves a lot of artifacts. At first I thought the video was too dynamic and reduced the video playback speed. This did not give any result. Reducing fps didn't help either. At the same time, if you shoot a real video with a uniform background on camera and move, for example, your hand or black stick in it, then there will be no such effect. The video will play back without artifacts. As I understand it, these artifacts are caused by the fact that the gif file contains such a parameter as transparency.

I use four 64*32 modules connected according to the CHAIN_TOP_RIGHT_DOWN scheme, so I use VirtualMatrixPanel. If I don't use VirtualMatrixPanel, using only one panel out of four, then the video also plays with artifacts. I set the panel brightness very low - 20 units.

How to fix it?

I use ESP32-HUB75-MatrixPanel-DMA library version 3.0.10
My OS is Windows 10 64 bit

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/5707511/612c1d13-a93b-48d4-b342-7faa7f8d945f

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/5707511/18b03317-d7f9-4ecf-8c11-417d3eb9eca6

Lukaswnd, 2024-05-22:
It seems like all new pixels of the gif are drawn, but not all of the old ones removes. This might be caused by a double buffer, which you do not have enabled, right?

Can you confirm transparency is disabled i? Simply add a print statement in the 
`if (pDraw->ucHasTransparency) // if transparency used`
part of GIF_Draw.

If you see the print statement in your logs, try a differnt converter or try to disabe transparency in the converter

Olejan, 2024-05-22:
> This might be caused by a double buffer, which you do not have enabled, right?

I don't use double buffer. If I use ```mxconfig.double_buff = true;``` in the ```display_init``` function, then the image in the panel disappears completely. I set the parameter into ```false``` - the image appears.

> Can you confirm transparency is disabled i? Simply add a print statement in the
> `if (pDraw->ucHasTransparency) // if transparency used`
> part of GIF_Draw.

Yes, my gif image use transparency

Olejan, 2024-05-22:
I changed service of video to gif conversion, and the artifacts disappeared. But the file size has doubled. I am currently using [this service](https://www.img2go.com/convert-video-to-gif).

Lukaswnd, 2024-05-27:
I've been throught this my selfe last year, but I don't remeber exactily, but I think there is a bug in the transparent implementation of the AnimatedGIF library, thus some converters work, other don't and disabeling transparency in the convter helps. And I am very sure this is not a problem with is library, rather with AnimetedGIF

Olejan, 2024-05-27:
@Lukaswnd Apparently so

OzerOzdemir, 2024-10-08:
Probable same issue as https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/685
I have verified that the gif above (Kinetik gif) has layers with different sizes and therefore is not handled correctly.

thegamerx1, 2025-07-31:
> I don't use double buffer. If I use `mxconfig.double_buff = true;` in the `display_init` function, then the image in the panel disappears completely. I set the parameter into `false` - the image [...]

This is probably because AnimatedGif library does not run flipDMABuffer(), so it's writing to a buffer that is never displayed.
Running flipDMABuffer() inside the while loop fixes it but causes glitches because it's writing parts of the GIF to 2 different buffers. Copying the displayed buffer to the second one would most likely fix it.