mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #633: Aurora demo won't compile

fscii, 2024-05-20:
Arduino IDE: 2.3.2
ESP-HUB75 library version:  3.0.10
Board manager esp32 v2.0.16

Error:

In file included from c:\Users\fscii\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                 from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\AuroraDemo.ino:43:
c:\Users\fscii\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\Patterns.h:59,
                 from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\AuroraDemo.ino:52:
C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\PatternInvaders.h: In member function 'virtual unsigned int PatternInvadersMedium::drawFrame()':
C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\PatternInvaders.h:95:98: error: no matching function for call to 'MatrixPanel_I2S_DMA::fillRect(int, int, int, int, CRGB&)'
           dma_display->fillRect(x + (i * 2), y + (j * 2), x + (i * 2 + 1), y + (j * 2 + 1), color);
                                                                                                  ^
In file included from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\AuroraDemo.ino:1:
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:546:16: note: candidate: 'virtual void MatrixPanel_I2S_DMA::fillRect(int16_t, int16_t, int16_t, int16_t, uint16_t)'
   virtual void fillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint16_t color)
                ^~~~~~~~
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:546:16: note:   no known conversion for argument 5 from 'CRGB' to 'uint16_t' {aka 'short unsigned int'}
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:556:23: note: candidate: 'virtual void MatrixPanel_I2S_DMA::fillRect(int16_t, int16_t, int16_t, int16_t, uint8_t, uint8_t, uint8_t)'
   virtual inline void fillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint8_t r, uint8_t g, uint8_t b)
                       ^~~~~~~~
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:556:23: note:   candidate expects 7 arguments, 5 provided
In file included from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\Patterns.h:59,
                 from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\AuroraDemo.ino:52:
C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\PatternInvaders.h:98:104: error: no matching function for call to 'MatrixPanel_I2S_DMA::fillRect(int, int, int, int, CRGB&)'
             dma_display->fillRect(x + (8 - i * 2), y + (j * 2), x + (9 - i * 2), y + (j * 2 + 1), color);
                                                                                                        ^
In file included from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\AuroraDemo.ino:1:
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:546:16: note: candidate: 'virtual void MatrixPanel_I2S_DMA::fillRect(int16_t, int16_t, int16_t, int16_t, uint16_t)'
   virtual void fillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint16_t color)
                ^~~~~~~~
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:546:16: note:   no known conversion for argument 5 from 'CRGB' to 'uint16_t' {aka 'short unsigned int'}
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:556:23: note: candidate: 'virtual void MatrixPanel_I2S_DMA::fillRect(int16_t, int16_t, int16_t, int16_t, uint8_t, uint8_t, uint8_t)'
   virtual inline void fillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint8_t r, uint8_t g, uint8_t b)
                       ^~~~~~~~
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:556:23: note:   candidate expects 7 arguments, 5 provided
In file included from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\Patterns.h:59,
                 from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\AuroraDemo.ino:52:
C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\PatternInvaders.h: In member function 'virtual unsigned int PatternInvadersLarge::drawFrame()':
C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\PatternInvaders.h:141:82: error: no matching function for call to 'MatrixPanel_I2S_DMA::fillRect(int, int, int, int, CRGB&)'
           dma_display->fillRect(1 + x * 6, 1 + y * 6, 5 + x * 6, 5 + y * 6, color);
                                                                                  ^
In file included from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\AuroraDemo.ino:1:
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:546:16: note: candidate: 'virtual void MatrixPanel_I2S_DMA::fillRect(int16_t, int16_t, int16_t, int16_t, uint16_t)'
   virtual void fillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint16_t color)
                ^~~~~~~~
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:546:16: note:   no known conversion for argument 5 from 'CRGB' to 'uint16_t' {aka 'short unsigned int'}
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:556:23: note: candidate: 'virtual void MatrixPanel_I2S_DMA::fillRect(int16_t, int16_t, int16_t, int16_t, uint8_t, uint8_t, uint8_t)'
   virtual inline void fillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint8_t r, uint8_t g, uint8_t b)
                       ^~~~~~~~
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:556:23: note:   candidate expects 7 arguments, 5 provided
In file included from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\Patterns.h:59,
                 from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\AuroraDemo.ino:52:
C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\PatternInvaders.h:144:96: error: no matching function for call to 'MatrixPanel_I2S_DMA::fillRect(int, int, int, int, CRGB&)'
             dma_display->fillRect(1 + (4 - x) * 6, 1 + y * 6, 5 + (4 - x) * 6, 5 + y * 6, color);
                                                                                                ^
In file included from C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024419-24552-1pp2lv6.pndt\AuroraDemo\AuroraDemo.ino:1:
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:546:16: note: candidate: 'virtual void MatrixPanel_I2S_DMA::fillRect(int16_t, int16_t, int16_t, int16_t, uint16_t)'
   virtual void fillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint16_t color)
                ^~~~~~~~
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:546:16: note:   no known conversion for argument 5 from 'CRGB' to 'uint16_t' {aka 'short unsigned int'}
[... cut ...] t16_t, int16_t, int16_t, uint8_t, uint8_t, uint8_t)'
   virtual inline void fillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint8_t r, uint8_t g, uint8_t b)
                       ^~~~~~~~
c:\Users\fscii\Documents\Arduino\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:556:23: note:   candidate expects 7 arguments, 5 provided

exit status 1

Compilation error: no matching function for call to 'MatrixPanel_I2S_DMA::fillRect(int, int, int, int, CRGB&)'

Lukaswnd, 2024-05-22:
Can you try to add the GFX ROOT?

In the very first line of AuroraDemo.ino add
#define USE_GFX_ROOT

and install the a root gfx fork, for example this one
https://github.com/mrcodetastic/GFX_Root

fscii, 2024-05-27:
Added: #include <GFX.h>

#define USE_GFX_ROOT

Same error:

'PatternInvaders.h: In member function 'virtual unsigned int PatternInvadersMedium::drawFrame()':
C:\Users\fscii\AppData\Local\Temp\.arduinoIDE-unsaved2024427-17304-ws680p.3kegi\AuroraDemo\PatternInvaders.h:95:98: error: no matching function for call to 'MatrixPanel_I2S_DMA::fillRect(int, int, int, int, CRGB&)'
           dma_display->fillRect(x + (i * 2), y + (j * 2), x + (i * 2 + 1), y + (j * 2 + 1), color);'

A litany of errors follow

davemaster, 2024-06-14:
the same... is there a solution? How that example run on ESP32... any advice. Thanks in advance

6BQ5-2, 2024-06-17:
The simplest solution is to comment out the lines with "invaders" in the sketch and also do the same for Patterns.h

The Invaders effect was not working too good. Not big deal to exclude it.

davemaster, 2024-06-17:
I solved.

Try this:
1. In Patterns, reduce the patters, use only one or two; comment those who are comment as "fail" or "this doesn't works" or etc.
2. Set PATTERN_COUNT= 2 (2, the number of patterns you left)
3. One by one, add the patterns, don't forget to put the same number of patterns, in PATTERN_:COUNT
4. Error about the IDE doesn't find a file YOU SEEN is there... put the folder of the project in the ROOT... 
Best Regards