mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #634: Outdoor panel with 16207S driver low brightness

jwee1369, 2024-05-22:
Hi,

I have one 64x64 panel with 16207S, for the most part the library works and i tried with the setup below:

in the first image -  when the panel is filled with blue, everything is connected and working as expected. 

In the second image - However, one time i accidentally unplug my ESP32 MCU from the power source, i noticed, the 4 lines (in blue) but they happened to be brighter (this is the brightness i am trying to achieve) - works good under sunlight

Any explanation or solution to achieve the (right) brightness with four lines?

Some things I have tried:
1. decrease increase brightness - works as expected
2. increase refresh_rate to 100 - no effect
3. define PIXEL_COLOR_DEPTH_BITS 12

Appreciate your help and thanks

[image]

[image]

`
  HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};

  HUB75_I2S_CFG mxconfig(
              PANEL_RES_X*2,              
              PANEL_RES_Y/2,             
              NUM_ROWS * NUM_COLS,          
              _pins,            // Uncomment to enable custom pins
              HUB75_I2S_CFG::ICN2038S
  );
  
  mxconfig.i2sspeed = mxconfig.HZ_20M; /
  mxconfig.min_refresh_rate = 10;
  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  if( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
  

  dma_display->clearScreen();
  delay(500);

  virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);
  virtualDisp->setPhysicalPanelScanRate(FOUR_SCAN_64PX_HIGH);

 
  dma_display->setBrightness8(255);    // range is 0-255, 0 - 0%, 255 - 100%
  virtualDisp->drawLine(0, 0, dma_display->width()-1, 0, myBLUE);
`

board707, 2024-05-22:
> Any explanation or solution to achieve the (right) brightness with four lines?

Unfortunately, there is no way you can achieve this brightness.
Your matrix has 1/16 scan, it means that it divided by 16 parts, switched one after another, so every pixel lit only 1/16 part of total time.
When you switch on the matrix without mcu, some random pixels lights statically and its brightness 16 times more than in normal mode.

jwee1369, 2024-05-22:
sorry I misunderstood, I assume setting the brightness sets the PWM signal pulse to 100% duty cycle. 

I'm just trying to understand what is limiting the led brightness since it can technically go brighter - in the second image when the 16 scans were all output to the same row of leds 16 times. In normal operation when one row only get one scan, it is rather dim. 

Is this limited by the driver current output for each of the scan, since 16 scans towards 1 row is brighter

board707, 2024-05-22:
Sorry, I don't know English well and I can't explain it more clearly
Re-read my first answer

jwee1369, 2024-05-22:
No problem, thank you for your help 😊

board707, 2024-05-22:
> what is limiting the led brightness 

The fact that every led is lit only 1/16 part of total time. You can't change this, Try to understand.

jwee1369, 2024-05-23:
I'm still trying to wrap my head around this, but reading other libraries, there are ways around it. Any chance we could implement the same solution for this library?

[https://community.pixelmatix.com/t/how-to-maximise-led-brightness/1206/2]()
[https://github.com/hzeller/rpi-rgb-led-matrix/issues/395]()

board707, 2024-05-23:
> Any chance we could implement the same solution for this library?

What the solution? The link to the pixelmatrix forum recommends you get an 8s panels instead of 16s - it is not an "increase the brightness" of the same panel.
The second link is about the tweaking the parameters of hzeller's library - it hardly has something to do with the currect library.

Sorry by saying that - but the fact that the total brightness of 16s panels is only 1/16 of the nominal brightness of its leds - it is not a bug or incorrect hardware settings, it is by design and how the HUB75 panels works.
But you can still trying to look at "hacks" if you want.

mrcodetastic, 2024-07-23:
Outdoor panels are 1/8  or 1/4 scan for this reason. They're brighter as the "on" time for each row isn't divided as much.

closed 2024-08-24 by mrcodetastic