mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #635: P8 32x16

mjmokhtar, 2024-05-27:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/6c517300-fde4-405a-8790-3f2110867778

i have problem using p8 32x16, coordinate in matrix in not position 

`    else if (panel_scan_rate == FOUR_SCAN_16PX_HIGH)
    {
        if ((virt_y & 8) == 0)
        {
            coords.x += (panelResX >> 2) * (((coords.x & 0xFFF0) >> 4) + 1); // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (panelResX >> 2) * (((coords.x & 0xFFF0) >> 4)); // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        if (virt_y < 32)
            coords.y = (virt_y >> 4) * 8 + (virt_y & 0b00000111);
        else
        {
            coords.y = ((virt_y - 32) >> 4) * 8 + (virt_y & 0b00000111);
            coords.x += 256;
        }
    }
`    

I think this code from virtual matrix have issue for me
I don't know how fix it    

`#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

// Panel configuration
#define PANEL_RES_X 32 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 16 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

#define SERPENT true
#define TOPDOWN false

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
VirtualMatrixPanel *FourScanPanel = nullptr;

void setup() {
    delay(250);

    Serial.begin(115200);
    Serial.println(""); Serial.println(""); Serial.println("");
    Serial.println("*****************************************************");
    Serial.println("*         1/8 Scan Panel Demonstration              *");
    Serial.println("*****************************************************");

    HUB75_I2S_CFG mxconfig(
        PANEL_RES_X/2, // DO NOT CHANGE THIS
        PANEL_RES_Y/2, // DO NOT CHANGE THIS
        NUM_ROWS * NUM_COLS // DO NOT CHANGE THIS
    );

    mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

    //mxconfig.driver = HUB75_I2S_CFG::FM6126A; // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

    // Create our matrix object
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);

    // Adjust default brightness to about 75%
    dma_display->setBrightness8(96); // range is 0-255, 0 - 0%, 255 - 100%

    // Allocate memory and start DMA display
    if (!dma_display->begin())
        Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

    dma_display->clearScreen();
    delay(500);

    // Create FourScanPanel object based on our newly created dma_display object
    FourScanPanel = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);

    // THE IMPORTANT BIT BELOW!
    FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);
}

void loop() {
    for (int y = 0; y < PANEL_RES_Y; y++) { // Loop through each row
        for (int x = 0; x < PANEL_RES_X; x++) { // Loop through each column
            FourScanPanel->drawPixel(x, y, FourScanPanel->color565(255, 255, 255)); // Light up one pixel
            Serial.print("Lighting pixel at (");
            Serial.print(x);
            Serial.print(", ");
            Serial.print(y);
            Serial.println(")");
            delay(100); // Delay to see the pixel lighting up
            FourScanPanel->drawPixel(x, y, FourScanPanel->color565(0, 0, 0)); // Turn off the pixel
        }
    }
}`

hope this problem have solution

board707, 2024-05-27:
> i have problem using p8 32x16
` if (panel_scan_rate == FOUR_SCAN_16PX_HIGH)`

Your panel is not a FOUR_SCAN_PANEL so you haven't use a VirtualMatrix class. The matrix 32x16 s8 is a normal matrix with scan a half of height, so it should work without of coordinate transformations.
If you experienced an issues, check your connections first.

mjmokhtar, 2024-05-29:
[image]

i have try to code normal matrix

`#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

//************************************************************ Display Config Declaration

#define PANEL_RES_X 32  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 16  // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1   // Total number of panels chained one to another

MatrixPanel_I2S_DMA* dma_display = nullptr;

uint16_t myBLACK;
uint16_t myWHITE;
uint16_t myRED;
uint16_t myGREEN;
uint16_t myBLUE;

void setup() {
  Serial.begin(115200);
  
  // Configurations for Display
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,  // module width
    PANEL_RES_Y,  // module height
    PANEL_CHAIN   // Chain length
  );

  // mxconfig.double_buff = true;
  // mxconfig.gpio.e = 32;
  mxconfig.clkphase = false;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(255);  //0-255
  
  // Define colors
  myBLACK = dma_display->color565(0, 0, 0);
  myWHITE = dma_display->color565(255, 255, 255);
  myRED = dma_display->color565(255, 0, 0);
  myGREEN = dma_display->color565(0, 255, 0);
  myBLUE = dma_display->color565(0, 0, 255);

  // Print dot at each coordinate
  for (int y = 0; y < PANEL_RES_Y; y++) {
    for (int x = 0; x < PANEL_RES_X * PANEL_CHAIN; x++) {
      dma_display->drawPixel(x, y, myRED);
      // Print coordinates
      Serial.print("(");
      Serial.print(x);
      Serial.print(",");
      Serial.print(y);
      Serial.print(") ");
      delay(100); // Delay 500ms before printing the next dot
    }
    Serial.println(); // Move to the next line after each row
  }

  // Display text "Hello"
  displayText();
}

void loop() {
  // Nothing here, just printing dots in setup()
}

void displayText() {
  dma_display->fillScreen(myBLACK);  // Clear the screen
  dma_display->setCursor(0, 0);      // Set the cursor at the top-left corner
  dma_display->setTextColor(myWHITE); // Set the text color to white
  dma_display->print("Hello");       // Print "Hello" on the display
}
`

but result is not print a text 

can you tell me what i should to do ??

board707, 2024-05-29:
Please show a clear photo of the panel rear side. Try to read the names of all chips on the matrix and show them in the message.

board707, 2024-05-29:
Do you know what is the scan factor of your panel? The labels on the panel tells nothing about scan.

Let' assume, that the panel is 1/4 scan.
Return to your code in the first message. Do not edit the VirtualPanel class in the library.
Please replace the last line of your `setup()` :
```
FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);
```
to this:
```
FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
```
Also replace your  `loop()` function to the code below and show the video:
```
void loop() {

 for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 0, 0));
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```

mjmokhtar, 2024-05-29:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/7f78116a-64d8-4680-9052-4de4cb94e6b6
i don't know what scan factor for this panel, maybe you right about to assume its panel have 1/4s 
here is result of this image what I should do for try print text

mjmokhtar, 2024-05-29:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/40f33525-e4ae-4af7-bd11-4d563ade1e69

board707, 2024-05-29:
Looks better
Please replace the `getCoords()` function in my last code to this:
```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

uint8_t pxbase =8;   // pixel base
 if ((coords.y & 4) == 0)
        {
            coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

  coords.y = (coords.y >> 3) * 4 + (coords.y & 0b00000011);
  return coords;
}
```
and test again.

mjmokhtar, 2024-05-29:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/aa69a714-38f0-4901-8bc1-fc54a88219df
 
here is video

board707, 2024-05-29:
Next iteration, I hope the last:
```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

uint8_t pxbase =8;   // pixel base
 if ((coords.y & 4) == 0)
       {
            coords.x = (coords.x / pxbase)*2*pxbase   + 7 - (coords.x & 0x7); // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
           coords.x += ((coords.x / pxbase) + 1) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

  coords.y = (coords.y >> 3) * 4 + (coords.y & 0b00000011);
  return coords;
}
```

mjmokhtar, 2024-05-29:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/7019cd97-4186-4958-b424-bd0604c0330f

here is video 
you doing great 

can you tell me how to print hello word ?

board707, 2024-05-29:
My congratulations!
Your panel setup process turned out to be almost the same as in my tutorial in #622 discussion.

> how to print hello word ?

You have to keep the matrix initialization as in the last code, and can do anything what you want in `loop()`
All the library examples should works fine with this code.

mjmokhtar, 2024-05-29:
okay thank you so much

board707, 2024-05-29:
To @mrcodetastic 
This issue is another confirmation that the getCoords() code for `FOUR_SCAN_16PX_HIGH` panels (lines 406-424 of `ESP32-VirtualMatrixPanel-I2S-DMA.h` file) is incorrect. 
I already wrote about this in the discussion #622

mrcodetastic, 2024-05-30:
Is there a pull request I can merge?

mjmokhtar, 2024-05-30:
a few question
if we have p8 40x20 
where do I need to change the coordinate matrix

mjmokhtar, 2024-05-30:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/108962994/4fa61d52-afa3-418a-ab7c-76805262d48f

here is video when i used 40x20 
can you help me ?

board707, 2024-05-30:
@mrcodetastic 
> Is there a pull request I can merge?

I'll try to prepare it in few days

board707, 2024-05-30:
@mjmokhtar 
> here is video when i used 40x20
> can you help me ?

Every panel has its own coordinate matrix. The only way to figure it out I know - testing panel as we did it yesterday. 
Please start from showing the rear side of the panel.

mjmokhtar, 2024-05-30:
okay i will make another issue