mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #636: HUB75E Not showing proper output when interfaced with ESP32S3

apurva-peppermint, 2024-05-29:
Hi, I am currently using ESP32-S3-WROOM-2 devkitC to drive a 64x64 (2 chained panels) LED Matrix panel. I am using Arduino IDE 2.1.1 and the master branch of this library. I have gone through the issues related to the ESP32S3 and have checked all the possible solutions as provided in these issues and still I am not getting proper output on the Display. For example, the simpleTestShapes code causes the code to stop in between and causes the esp32s3 core to reset and the bitmap code simply doesnt run. I am attaching a video below for reference. Also, I have followed the default connections as provided by the library and have connected GPIO 21 of the ESP32S3 to the Address Line E of the HUB75E Panel.

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/132329071/b2be7a05-22f2-4312-86c2-4fb3218096cb

mrcodetastic, 2024-05-30:
Power related I think.

apurva-peppermint, 2024-05-30:
I am using a stable dc variable power supply unit to power the Displays. Also I have tried the code with esp32 and it works. It only has this issue on esp32s3 (wroom 1 and wroom 2 both)

apurva-peppermint, 2024-05-30:
I am also trying to use esp-idf instead of arduino and I am facing some problems with it. I am currently using the 3.0.9 version of the ESP32-HUB75-MatrixPanel-DMA library at https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/tree/3d7c2dfa4f3305ed80a5a86401e7ef9ed68e8038. I got this one from https://github.com/acvigue/SmartMatrix-IDF/tree/main/components this git repo. I am attaching some screenshots for your reference. I am using the same CmakeLists.txt file which is there in the repo that I have mentioned.

[image]
[image]
[image]
[image]

apurva-peppermint, 2024-05-30:
Hi @mrcodetastic managed to add the library as a component and build + flash the code successfully in the esp32s3. But now I am getting a watchdog trigger. I am attaching the output below
[image]

mrcodetastic, 2024-05-30:
The esp-idf has a function to disable the watchdog.  I forget what it is,  but put it at the start of the app logic. 

The arduino framework does this automatically I think.

apurva-peppermint, 2024-05-31:
Hi @mrcodetastic tried disabling the watchdog still no output on display it stays blank. 
[image]

mrcodetastic, 2024-08-24:
Did you find a solution for this issue?