mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #642: P2 128x64 MOEN panel with FM6363C LED Driver Chips

Speedy-VI, 2024-06-14:
I have been trying to find P2 128x64 LED panels that I can control with an Adafruit Matrix Portal S3 (ESP32-S3). I have several P4 64x32 panels from Ali Express and a P2.5 64x32 panel from Adafruit that work fine with the Matrix Portal S3.  I have been writing CircuitPython code but can switch to Arduino IDE if necessary. I could also get a PI 4 if necessary. I learned the hard way that a lot of LED panels on Ali Express that are 64x64 or 128x64 only have 2 or 3 address lines and use LED driver chips that basically have shift registers controlling the row addresses. I thought if I could find panels that have address lines A through E, the Matrix Portal S3 could control them.

I found P2 128x64 panels with 5 address lines, but when I connect one to the Matrix Portal S3 I get nothing, not even a flash when I power up. These panels have FM6363C LED driver chips. I found your list of chips you have managed to support and chips you cannot support but the FM6363C is not listed. I also looked at PixelMatrix's list and hzeller on github.  I have not found anyone listing the FM6363C. Do you know if your library, or anyone's library can control a panel with FM6363C driver chips? Any insights are appreciated.

board707, 2024-06-14:
> Many panels have 2 or 3 address lines and basically have shift registers controlling the row addresses.

There is nothing wrong with panels with shift registers to control addressing; many libraries can work with such panels.

Unlike shift registers, FM6363 driver is a much more complex thing and you will most likely have problems with them. I don't know any libraries for ESP32 that work with this driver.
I have code for this driver, but it is for an STM32 controller

Speedy-VI, 2024-06-15:
Thank you for replying so quickly. I found the datasheet for the FM6363A and translated the text into English. The more I read, the less my hope was that I could control these panels with a Matrix Portal S3. The FM6363C is a complex chip, but since the panels have 5 address lines I thought they would work with the Matrix Portal S3 but they do not, at least with the code I have. I am pretty new to this and have had no problems with the 64x32 panels.

This is my second attempt at buying P2 128x64 panels on Ali Express. The first time I had no idea there were panels with only 2 or 3 address lines. I bought 2 that showed up with only HUB75 input connectors (despite the product photo showing HUB75 in and out). Those have ChipOne ICND2038S chips which I think are similar to the ICN2038S that I found on another github list (pixelmatrix) of chips that are not supported. Even if I could get them working, the missing HUB75 output connector is a problem because I want to daisy-chain 3 panels.

By digging around on GitHub I learned that a lot of panels with various driver chips can be controlled with a PI and libraries that have been developed to support them. I could buy a PI but if I want to use the Matrix Portal S3 I think I may be limited to 64x32 panels.  If I bought an Adafruit STM32F405 Express board  [https://www.adafruit.com/product/4382](url) could I use your code for the FM6363C with these panels? Thanks again for your help.

board707, 2024-06-15:
>  I found the datasheet for the FM6363A 

Is this a full datasheet? Could you attach it to the message?

> If I bought an Adafruit STM32F405 Express board [https://www.adafruit.com/product/4382](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/url) could I use your code for the FM6363C with these panels?

The code is experimental and tested with STM32F401 & F411 boards only. It is a part of my [DMD_STM32](https://github.com/board707/DMD_STM32) library.

Speedy-VI, 2024-06-15:
The data sheet I found is 13 pages long. I have attached a copy with Chinese to English translation of most of the text. The translation of page 1 text is on 2 pages I inserted after page 1. The rest of the translated text is crammed into the original pages. 

The diagram on page 11 shows a DLCK and a GCLK.  The CircuitPython matrix definition code I have been using defines display Width and Height, Bit Depth, 6 RGB pins, 5 address pins, CLK, LAT, and OE. There is no GLCK which may be part of the reason I get nothing. I also attached a pic of the pinout printed on the panel. Please let me know if you think there is any chance of controlling these panels with the Matrix Portal S3.

I have started reading your DMD_STM32 library docs which say RP2040 boards are supported. I have an Adafruit Feather RP2040 Scorpio board. Could I control these panels with that? Many thanks.

[2108241130_Shenzhen-Fuman-Elec-FM6363C-B_C2887418.pdf](https://github.com/user-attachments/files/15848719/2108241130_Shenzhen-Fuman-Elec-FM6363C-B_C2887418.pdf)

[image]

board707, 2024-06-16:
This is not a complete datasheet; it does not contain the most important thing - configuration registers. In order for drivers of this type to work correctly, before sending data they need to send several 16-bit numbers that contain all the necessary settings. such as matrix size, number of scans, brightness by channel, and so on. Without this, the matrix will not show anything.

board707, 2024-06-16:
For further information, read the discussion #324 and this link

https://github.com/hzeller/rpi-rgb-led-matrix/issues/466

Speedy-VI, 2024-06-17:
Thank you.  Apologies about the incomplete FM6363C spec sheet. That was all I could find and I did not know that it was missing crucial information. I spent time today reading through the threads you recommended. I have to say I am impressed with the work you, hzeller, and others have done to figure out how to make Gen-2 and Gen-3 LED panels work. Since you posted that you got an image on an FM6363 panel, doesn't that mean you already have the configuration registers figured out? I noticed in those threads that there were references to FM6363 and FM6363B, but not FM6363C, so maybe the FM6363C is a later iteration of the FM6363 and does not use the same parameters as the FM6363 you were able to crack.

My walk away is that the FM6363C and other Gen-3 chips require specific signals and timing to work, and that I probably cannot provide what they need with the Matrix Portal S3. daveythacher said in the Issue #324 thread that there are 3 generations of LED driver chips, and that Gen-3 chips are now commonly found in higher density indoor panels with high refresh rates. He also said he thought that Gen-2 and Gen-3 panels would probably not be widely adopted by hobbyists due to lack of economical support and insufficient documentation, and that some chips do things that are proprietary and in some cases protected by NDA's. Even if someone gets out their logic analyzer and figures out what they are doing, replicating their function could raise legal issues.

I am a beginner with these panels and what you guys are doing is over my head. I may be able to control these FM6363C panels with an RP2040 but I have my doubts. I will probably send them back. I did look at the chips on the P4 and P2.5 64x32 panels I have that work with the Matrix Portal S3. The P4 panels use the RUL2038S. I don't know if that is the same as an ChipOne ICN2038S or ICND2038S which is what the first set of P2 128x64 panels I bought have. Since the P4 64x32 panels with the RUL2038S work fine with the Matrix Portal, I guess it is not the same as the ICND2038S. The P2.5 64x32 panel I have uses Chipone ICN2037BP and also works fine with the Matrix Portal S3. Do you think it is possible that there are P2 128x64 panels that use the RUL2038S or ChipOne ICN2037BP chips or are those chips limited to smaller panels like 64x32?

Many thanks for your insights.

board707, 2024-06-17:
> The P2.5 64x32 panel I have uses Chipone ICN2037BP and also works fine with the Matrix Portal S3. Do you think it is possible that there are P2 128x64 panels that use the RUL2038S or ChipOne ICN2037BP chips or are those chips limited to smaller panels like 64x32?

Unfortunately, panel size and resolution cannot be relied upon to determine the driver. The panel I encountered with the FM6363 driver was 64x32. And the fact, that you bought P2.5 64x32 panel with Chipone ICN2037BP does not guarantee that the next time you buy it there will not be a completely different chip.
I have listed a some known drivers [in the table](https://github.com/board707/DMD_STM32/wiki/Led_drivers) in my library's Wiki. The table contains links to examples of using DMD panels with these drivers with different libraries.

Speedy-VI, 2024-06-23:
> Unfortunately, panel size and resolution cannot be relied upon to determine the driver. The panel I encountered with the FM6363 driver was 64x32. And the fact, that you bought P2.5 64x32 panel with [...]

Thanks for the clarification. I bought the P2,5 64x32 panel with Chipone ICN2037BP chips from Adafruit when I bought the Matrix Portal S3. I paid their premium price because I wanted to start with a panel known to work with their Matrix Portal, and I wanted to be able to get support on their forum if I had problems. If I bought more of them they would probably be from the same batch, but this was just a test panel. I don't think I will find P2 128x64 panels with 5 address lines and LED driver chips that will work with the Matrix Portal S3. They all seem to use these complex "Gen-3" LED driver chips. I think I either need to get a Raspberry Pi and find panels that use LED driver chips that have support or give up and go back to P4 64x32 panels.

Adafruit does have Coreman P2 64x64 panels that will work with the Matrix Portal S3 but they are $50 a pop and I need 6 of them. I may send Coreman a message on Ali Express asking if they have the same P2 64x64 panels that Adafruit sells. Adafruit's site has a Coreman spec sheet for the panel and a clear photo of the back. I found 2 P2 Coreman P2 64x64 panels on Coreman’s Ali Express site that are almost identical. I can see that they have ICN2037BP chips. The only difference is the Coreman panels have 11 metal can components (I think caps). The Adafruit panels have places for 11 but only 4 are installed. With shipping, the Coreman panels on Ali Express are about half the price of the Adafruit panels.
[image]
[image]

ocram72nl, 2024-12-31:
Hi, I have also bought P2-1515-128x64-32S-S2 panels with the FM6363C driver chips which obviously do not work with the current implementation of this library.
From the posts above I conclude that at least @board707 managed to get such a panel working using STM32 based board: [DMD_STM32](https://github.com/board707/DMD_STM32).
User @CamelCaseName has created a library for ArduinoNano/Uno for similar (S-PWM) chips: [[HUB75Enano](https://github.com/CamelCaseName/HUB75Enano)](https://github.com/CamelCaseName/HUB75Enano/tree/main); this makes me believe it should be possible on ESP32 too.

The challenge I am facing is that I want to be able to use WiFi (and MQTT) to be able to show (scrolling) text, but the WiFi handling could be challenging if DCLK signal needs to be available all the time. Not sure if this can be done, and if I need specific I/O port on ESP32 for output of clock signals (there is no TIM module in ESP32, but ESP32 does have PWM peripherals (LEDC and MCPWM) that could be used instead).

In the end I am just curious if it makes sense to spend more time on this, or should I focus on different approach (like using STM32 for driving the panels, and ESP32 for WiFi connectivity?). Are others still working on this?

Thanks and all the best for 2025!

bluesubway, 2025-04-05:
I have detailed control and programming guidelines here. I hope it can be of help to you.

[FM6363控制及编程指南v0.1(2).pdf](https://github.com/user-attachments/files/19616639/FM6363.v0.1.2.pdf)

bluesubway, 2025-04-05:
I have the programming and control manuals here. I hope they can be of help to you.
@njh @Speedy-VI @board707 @ocram72nl @mrcodetastic 

[FM6363控制及编程指南v0.1.pdf](https://github.com/user-attachments/files/19616710/FM6363.v0.1.pdf)

board707, 2025-04-06:
Thank you very much

board707, 2025-04-06:
Google translation

[FM6363_Programming_Guide_v0.1_en.pdf](https://github.com/user-attachments/files/19623191/FM6363_Programming_Guide_v0.1_en.pdf)

mrcodetastic, 2025-04-08:
> Hi, I have also bought P2-1515-128x64-32S-S2 panels with the FM6363C driver chips which obviously do not work with the current implementation of this library. 

These panels will never work with this library as they are essentially a framebuffer device and this library will only be for shift register based LED Matrix Panels.

With that said, a while back I was donated a bunch of MBI5153 based PWM panels and got a working implementation for both the S2 and S3. From looking at the datasheet that @bluesubway kindly provided and @board707 translated, it would not be too difficult for somebody to look at my MBI5153 implementation and re-code for the FM6363.

https://github.com/mrcodetastic/ESP32-PWM-MatrixPanel-DMA/tree/main/MBI5153

> [FM6363_Programming_Guide_v0.1_en.pdf](https://github.com/user-attachments/files/19623191/FM6363_Programming_Guide_v0.1_en.pdf)

Excellent work.

sityware, 2025-04-30:
> I have the programming and control manuals here. I hope they can be of help to you. [@njh](https://github.com/njh) [@Speedy-VI](https://github.com/Speedy-VI) [@board707](https://github.com/board707) [...]

Hi,
Do you happen to have the same guide for FM6373 ?

Thanks,

lthuillie002, 2025-06-10:
> > I have the programming and control manuals here. I hope they can be of help to you. [@njh](https://github.com/njh) [@Speedy-VI](https://github.com/Speedy-VI) [@board707](https://github.com/board707) [@ocram72nl](https://github.com/ocram72nl) [@mrcodetastic](https://github.com/mrcodetastic)
> > [FM6363控制及编程指南v0.1.pdf](https://github.com/user-attachments/files/19616710/FM6363.v0.1.pdf)
> 
> Hi, [@bluesubway](https://github.com/bluesubway) - do you happen to have the same guide for FM6373 ?
> 
> Thanks,

Hi,
@sityware I found the datasheet for the FM6373C driver if that helps. I'm interested if someone can find the complementary guide to programming the registers and signals timing of this driver.

[FM6373C_EN.pdf](https://github.com/user-attachments/files/20671450/FM6373C_EN.pdf)

JarekC-DIY, 2025-07-16:
Hi,
@bluesubway - do you happen to have the same guide for FM6565C ?
I have 256x128 P1.25 panel with this chipset.

Thanks, JarekC