mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #643: "GHOSTING" only when printing white text

adamNZRI, 2024-06-15:
Hello all,

First of all, I apologize if my English is confusing, because I use Google Translate.

I've used this library for "P5 RGB 64x32 Pixel HUB75 Scan 1/16S Panel" and it works fine. Thank you very much for this library.

Then I looked in some discussions on this library, whether this library works on Panel P10 RGB 32x16 1/8S and I found some people have tried it and it works. I found one here : https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/455

Then I tried it myself using this library for a P10 RGB 32x16 1/8S Panel. As a result I found some kind of "GHOSTING" on my panel. But it only shows if I print the text on white. Here are the photos :

I marked "GHOSTING" in the red box.
[image]
[image]

"GHOSTING" does not appear in red, green and blue text.
[image]
[image]
[image]
[image]
[image]
[image]

The following is the code I use :

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 23
#define B_PIN 19
#define C_PIN 5
#define D_PIN -1
#define E_PIN -1
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16

#define PANEL_RES_X 32
#define PANEL_RES_Y 16
#define PANEL_CHAIN 1 

MatrixPanel_I2S_DMA *dma_display = nullptr;

uint16_t myBLACK = dma_display->color565(0, 0, 0);
uint16_t myWHITE = dma_display->color565(255, 255, 255);
uint16_t myRED = dma_display->color565(255, 0, 0);
uint16_t myGREEN = dma_display->color565(0, 255, 0);
uint16_t myBLUE = dma_display->color565(0, 0, 255);

void setup() {
  // put your setup code here, to run once:

  delay(2000);
  Serial.begin(115200);

  HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
  delay(100);

  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   //--> module width.
    PANEL_RES_Y,   //--> module height.
    PANEL_CHAIN,   //--> Chain length.
    _pins          //--> pin mapping.
  );
  delay(100);

  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;
  delay(100);

  mxconfig.latch_blanking = 4;
  delay(100);

  mxconfig.clkphase = false;
  delay(100);

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(10);
  delay(100);

  dma_display->clearScreen();
  delay(100);
  
  dma_display->fillScreen(myWHITE);
  delay(3000);
  dma_display->fillScreen(myRED);
  delay(3000);
  dma_display->fillScreen(myGREEN);
  delay(3000);
  dma_display->fillScreen(myBLUE);
  delay(3000);
  
  dma_display->clearScreen();
  delay(1000);

  dma_display->setTextSize(1);    
  dma_display->setTextWrap(false);
  delay(100);
}

void loop() {
  // put your main code here, to run repeatedly:

  dma_display->clearScreen();
  dma_display->setCursor(0, 0);
  dma_display->setTextColor(myWHITE);
  dma_display->println("1234");  
  dma_display->setCursor(0, 9);
  dma_display->setTextColor(myWHITE);
  dma_display->print("1234");
  delay(5000);
  
  dma_display->clearScreen();
  dma_display->setCursor(8, 0);
  dma_display->setTextColor(myWHITE);
  dma_display->println("1234");
  dma_display->setCursor(8, 9);
  dma_display->setTextColor(myWHITE);
  dma_display->print("1234");
  delay(5000);
  
  dma_display->clearScreen();
  dma_display->setCursor(0, 0);
  dma_display->setTextColor(myRED);
  dma_display->println("1234");  
  dma_display->setCursor(0, 9);
  dma_display->setTextColor(myRED);
  dma_display->print("1234");
  delay(2500);
  
  dma_display->clearScreen();
  dma_display->setCursor(8, 0);
  dma_display->setTextColor(myRED);
  dma_display->println("1234");
  dma_display->setCursor(8, 9);
  dma_display->setTextColor(myRED);
  dma_display->print("1234");
  delay(2500);
  
  dma_display->clearScreen();
  dma_display->setCursor(0, 0);
  dma_display->setTextColor(myGREEN);
  dma_display->println("1234");  
  dma_display->setCursor(0, 9);
  dma_display->setTextColor(myGREEN);
  dma_display->print("1234");
  delay(2500);
  
  dma_display->clearScreen();
  dma_display->setCursor(8, 0);
  dma_display->setTextColor(myGREEN);
  dma_display->println("1234");
  dma_display->setCursor(8, 9);
  dma_display->setTextColor(myGREEN);
  dma_display->print("1234");
  delay(2500);
  
  dma_display->clearScreen();
  dma_display->setCursor(0, 0);
  dma_display->setTextColor(myBLUE);
  dma_display->println("1234");  
  dma_display->setCursor(0, 9);
  dma_display->setTextColor(myBLUE);
  dma_display->print("1234");
  delay(2500);
  
  dma_display->clearScreen();
  dma_display->setCursor(8, 0);
  dma_display->setTextColor(myBLUE);
  dma_display->println("1234");
  dma_display->setCursor(8, 9);
  dma_display->setTextColor(myBLUE);
  dma_display->print("1234");
  delay(2500);
}

Software and libraries used :
- Arduino IDE Version 1.8.19
- Arduino core for the ESP32 Version 2.0.16
- ESP32 HUB75 LED MATRIX PANEL DMA Display Version 3.0.10

The ESP32 that I use is the ESP32 DEVKIT V1 module.

Information written on the P10 RGB 32x16 panel that I use:
- P10-2727-8S-32X16-A
- IC/Chip on panel :
 - RUC7258D
 - DP5125D
 - 74HC245KA

Hopefully someone can help me.

I apologize if there are incorrect terms or words that I used.
Thank You.

board707, 2024-06-15:
> I tried it myself using this library for a P10 RGB 32x16 1/8S Panel. As a result I found some kind of "GHOSTING" on my panel.

I can confirm it, I also observed this on 32x16 boards.
See video
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/50185434/401ad801-c8b5-4b0c-81b5-4d4bba56e3d0

adamNZRI, 2024-06-16:
@board707 I've seen your video and I see the ghosting.

According to your observations, this ghosting occurs because there is an error on the 32x16 panel board or maybe there is a setting in the program code that I missed ?

board707, 2024-06-16:
Typically, such interference is caused by an incorrect sequence of switching signals ABCDE, LAT and OE

adamNZRI, 2024-06-16:
@board707 Can you recommend what I should do to overcome the ghosting ?

For wiring, I can ensure that the cable I use is in good condition, because I have checked it repeatedly.

For PINs D and E, my 32x16 panel board doesn't have them. Only ABC is available.

The details are like this:
DR1 | DG1
DB1 | GND
DR2 | DG2
DB2 | GND
 A    |   B
 C    | GND
CLK | STB
OE   | GND

board707, 2024-06-16:
> Can you recommend what I should do to overcome the ghosting ?

There's nothing you can do other than check the wires again. A fuzzy connection can also give this.

If everything is fine with the wires, then the problem should be looked for in the library code, I think.

adamNZRI, 2024-06-17:
@board707 Thank you very much for your answer. Your answer has been very helpful, because until now you are the only one willing to answer my question.

> There's nothing you can do other than check the wires again. A fuzzy connection can also give this.

I'll check the cables again or maybe I'll try using another ESP32 board.

6BQ5-2, 2024-06-18:
In June 2021, I made a few projects using 32x16 panels by joining two or four of them to form 64x16 and 64x32 displays. Recently I wanted to make some fine tunings and found all of them could not be compiled. When this one eventually compiled OK, the upper 8 rows were completely scrambled. When I put my fingers on the CLK pin and the ground, it looked much better.
[image]
I put a 30pF capacitor between the CLK and Ground on the ESP32 PCB. The upper lines became almost normal, just some ghosts were still there. I then changed the capacitor to 68pF. Immediately everything worked normally. No more ghosts!
[image]
So I recommend you try to solder a 68pF cap on the CLK pin and the ground. The idea comes from the datasheet of TM1638, which is something similar to MAX7912.
[image]
This application circuit shows that three 100pF caps are added to the STB, CLK, DIO lines with pull-up resistors. But I think a 33-68pF cap on the CLK line and ground will solve most of the problems. This trick can also be applied to MAX7912 displays. Hope this will solve your problem.

I must thank mrcodetastic and others that the recent version of ESP32-HUB75-MatrixPanel-DMA has been improved a lot to eliminate the ghost problems compared the version of June 2021 that I used.

adamNZRI, 2024-06-19:
@6BQ5-2 I've followed your recommendation, but it doesn't solve the problem, ghosting still appears on my panel. Maybe the panel you use is different from the one I use (maybe different chips or something else). But whatever, thank you very much for the advice, maybe it can help someone else.

In my case, perhaps the P10 32x16 RGB panel I was using was not completely compatible with this library.
The reason I say that is because I have tried my panel with an Arduino Uno using the "Adafruit RGB Matrix Panel" library, the result is that no "ghosting" appears.
Then I tried my panel with ESP32 using the "PxMatrix" library, the result was also no "ghosting" appeared.

6BQ5-2, 2024-06-20:
My panels are P5 32x16. I also started with "Adafruit RGB Matrix Panel" library using Arduino Mega2560. There was no ghosting but I could observe white flash across the panel every few seconds. I tried it again 2 days ago and the same thing still happening. 

When I switched to ESP32 and started using “ESP32-HUB75-MatrixPanel-DMA” in 2021, there was very obvious “Tail” on each pixel. I believe that was caused by the old version of library in 2021.
With the recent version of “ESP32-HUB75-MatrixPanel-DMA”, the “Tail” problem seems being solved, although I still can spot some very slight and dim “Ghost” spots. The result is more than satisfactory. That is why I suspect that your problem is a hardware problem rather than a software one.
You may have already read that but just in case. Please refer to the “Power, Power and Power!” section of the home page (https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA). By adding a large cap on the Panel will eliminate noise from the 5V power supply. 

Also try to lower the brightness to avoid overloading the drivers.

That is all that I can think of for now.

adamNZRI, 2024-06-20:
@6BQ5-2 

> With the recent version of “ESP32-HUB75-MatrixPanel-DMA”, the “Tail” problem seems being solved, although I still can spot some very slight and dim “Ghost” spots. The result is more than satisfactory. That is why I suspect that your problem is a hardware problem rather than a software one.

I don't know for sure whether this problem comes from the hardware or software side. But as I explained before, that if my P10 panel uses the PxMatrix library + ESP32, no "ghosting" appears. I have tried displaying text in the position where "ghosting" appears if using this library, but in the same position no "ghosting" appears if using the PxMatrix library. Sorry if my explanation is a little confusing, I hope you understand what I mean.

> Also try to lower the brightness to avoid overloading the drivers.

I have tried this method. But in my case, this only works if the brightness is lowered from the "dma_display->color565()" side instead of the "dma_display->setBrightness8()" side.

For example :
dma_display->setCursor(8, 9);
dma_display->setTextColor(dma_display->color565(255, 255, 255));
dma_display->print("1234");

If I display white text (because in my case, "ghosting" only appears when I display white text), whatever brightness value I set (except 0) in "setBrightness8()", if "color565(255,255,255) ", then "ghosting" appears. But if I reduce the white color by setting the RGB value to "color565(64,64,64)", the result is that the "ghosting" still appears, but the light from the "ghosting" is very dim and can only be seen when viewed from a very close distance (I assume "ghosting" is gone). But in this way, the displayed text is very dim (loses too much brightness).

> You may have already read that but just in case. Please refer to the “Power, Power and Power!” section of the home page (https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA). By adding a large cap on the Panel will eliminate noise from the 5V power supply.

I've read the “Power, Power and Power!” section, but I haven't tried it. Because when I used this library for the "P5 RGB 64x32" panel, I didn't add capacitors and there was no problem. But this time I will try adding a capacitor to my P10 panel, hopefully it will work.

adamNZRI, 2024-06-21:
@6BQ5-2 I've tried adding capacitors as suggested in the “Power, Power and Power!” section, and the result is that "ghosting" still appears on my P10 panel.

> I can't set the "clkphase" in the sketch, so I have to do it in the library. If anybody knows how, please help me out!

As far as I know it can be changed from the sketch, here is the command:
mxconfig.clkphase = true;
I have set the clkphase section with "true" and "false", and the result "ghosting" still appears.

6BQ5-2, 2024-06-21:
> As far as I know it can be changed from the sketch, here is the command: mxconfig.clkphase = true; I have set the clkphase section with "true" and "false", and the result "ghosting" still appears.

I tried that but it complaint that mxconfig was not declared, so took the fast way to edit the library.

Since you have tried everything and it still not working for you, what left is either bear with it or use something else, or wait until further updates are available.

adamNZRI, 2024-06-22:
@6BQ5-2 

> Since you have tried everything and it still not working for you, what left is either bear with it or use something else, or wait until further updates are available.

Yes, what I am currently doing is using the PxMatrix library to complete my project. After my project is finished, I will try to use this library again to find a solution to overcome "ghosting" on my P10 panel. I really like this library, therefore I still want to find a solution to overcome "ghosting".

mrcodetastic, 2024-07-23:
Try using a lower clock speed when using this library.

LucasFeh, 2024-09-25:
First of all, I apologize for my English to. 

I am also using the same library, and I’m facing the same ghosting issues on the panel. However, I found a temporary solution. For some reason, each panel shows ghosting at different power points. I used three identical panels, but the first panel doesn’t show ghosting at 4V, while the second works normally at 5V. However, if I use the same power supply for both panels, the problem occurs. Therefore, the temporary solution has been to power each panel with a different power source. As I mentioned, this solution is temporary, so if anyone has a solution for my problem, please give me a "LIGHT" XD

Dom-coder932, 2026-08-28:
Gooooooddd!

    Where did you buy these panels? Do you have a link to the exact P10-2727-8S-32X16-A panel you used?

board707, 2026-08-28:
> Do you have a link to the exact P10-2727-8S-32X16-A panel you used?

There are nothing specific in this label:  P10-2727-8S-32X16-A    
I think the each second P10 32x16 panel in the market has such specs