mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #644: Is there support for FM6124C HUB75E displays?

TheSouthernFrontier, 2024-06-19:
I have a HUB75E board that uses a FM6124C chip.
Does this library have support for this chip, or is there an alternative library that can support the FM6124C with ESP32?

mrcodetastic, 2024-08-24:
This library does not contain code specific to FM6124C, so probably not supported.

closed 2024-08-24 by mrcodetastic

cris-maillo, 2024-09-29:
@TheSouthernFrontier my board has FM6124C chip and I've managed to get it working (although I'm having an issue with some of the pixels being mapped incorrectly but I think it must be one of the pins being in the wrong place)

board707, 2024-09-29:
There is nothing specific in FM6124 driver, it should be supported by the library without any modification of the current code.