mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #645: Issue with 64x64 panel based on DP5125D and DP32020A

mzashh, 2024-06-20:
i got a new HUB75 Panel today and it does not seem to work properly with the library. The panel initializes fine and seems to display stuff but everything is mapped incorrectly and overlapping. the panel seemed to work more or less the same with different driver types, but it flickered with some the best one i found which worked is MBI5124 and I am using the same one in the below pictures.

Here is the SimpleTestShapes demo (64x64)

[image]

and here is the Plasma demo (64x64)

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/58217787/3a7ad8ec-2ff1-41e6-bb8e-baef87be240b

So i there any way I can drive this panel properly?

davemaster, 2024-06-20:
Can You share a picture of your connections? E_pin present? Thanks in advance

mzashh, 2024-06-20:
> Can You share a picture of your connections? E_pin present? Thanks in advance

like my other 64x64 hub75 panel (which works fine cuz different driver) the E and D pins are either labelled as GND or NC and yes i have connected those and defined them in my code. i did a continuity test on them and D which is labelled as NC is not grounded while as E is grounded so either the input is ABC or ABCD, no E pin

board707, 2024-06-20:
I think, the problem is in the fact that DP32020A is a shift register miltiplexor - do you know how to work with it?
By the way - what is the scan factor of the panel?

board707, 2024-06-20:
> since the E pin is pretty much shorted to ground and D is no continuity to ground it is either a 1/16 or a 1/8 scan panel, pretty sure about 1/8 scan. 

As panel uses a shift register multiplexer, the number of ABC lines means nothing about scans. Please show a clear photo pf panel rear side 

> i found this library ... it has support for DP5125D 

The DP5125 is a standard driver, you don't need a special library for it. The current library is definitely supports this driver.

mzashh, 2024-06-20:
> Please show a clear photo pf panel rear side.

[image]

board707, 2024-06-21:
According to number of chips, the panel is 1/32scans

mzashh, 2024-06-21:
> According to number of chips, the panel is 1/32scans

1/32 scan but no D and E pin man this is one weird panel, so any ideas on how to drive it?

board707, 2024-06-21:
> 1/32 scan but no D and E pin man this is one weird panel

Again, DP32020 is a shift register, it don't need more than ABC to control 32 scans.
Read [this](http://kuku.eu.org/?projects/hub75e/index), the part titled "HX6158" - it is an explanation how the SR multiplexors works.

board707, 2024-06-21:
> this is one weird panel, so any ideas on how to drive it?

There is nothing weird about this panel. The 5125 driver is very common; some settings are needed only for the 32020 multiplexer. As far as I remember, @mrcodetastic  had code for panels with such multiplexers.
Now I need to get busy, a little later in the day I will find links in the "Issues" on such panels

board707, 2024-06-21:
The code to work with shift register multiplexors discussed in #164, #429 and some other issues

AzonInc, 2024-12-02:
Hi, I just received such a panel but I dont't really know which configuration I need to drive this. For now I got it to display something but there is always something missing or it doesn't power on at all. It uses the DP5125D chip.

> Do you have an experience to use this library with a "standard" 1/32 panel? If so, I will show what to change in the code to work with your panel.

@board707 Do you know something how to make them work? Or are there any examples how to use them?

Thanks in advance :)

board707, 2024-12-03:
> I just received such a panel 

Your panel is different from the one of @mzashh .
According to chip types your matrix is ordinal two scan panel, that should works with standard library settings.

AzonInc, 2024-12-03:
Thanks for the info, I just found the issue with my panel.
They labelled the D pin with GND for no reason. 
It can't work that way, now it finally works as expected :)

board707, 2025-05-19:
> Is there any exact solution to drive the panel with DP5125D and DP32020A chips?

The driver DP5125D is the standard one and not a problem.
As about a line switcher DP32020 - it is of a "shift register type" which needs an additional setup.

In case that we use panels based on that chips, we have to  set an option:
```
mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;
```

 before creating MatrixPanel_I2S_DMA object

nadir-dogan, 2025-05-19:
@board707 thanks you for your quick reply.
I had many tries after got your reply but unfortunately there is not a good result.
Please see my whole code and the pictures.
I'm rooky on this rgb panels, I just started a few days ago.

#include <Arduino.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define R1_PIN 18
#define G1_PIN 14
#define B1_PIN 16
#define R2_PIN 17
#define G2_PIN 13
#define B2_PIN 21
#define A_PIN 34
#define B_PIN 8
#define C_PIN 36
#define D_PIN 4
#define LAT_PIN 2
#define OE_PIN 40
#define CLK_PIN 38
#define E_PIN 10

/********** CONFIG *************/
#define PANEL_WIDTH 64
#define PANEL_HEIGHT 32
#define PANELS_NUMBER 1

MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup() {
Serial.begin(115200);

HUB75_I2S_CFG mxconfig;
mxconfig.mx_width = PANEL_WIDTH;
mxconfig.mx_height = PANEL_HEIGHT;
mxconfig.chain_length = PANELS_NUMBER;
mxconfig.gpio.r1 = R1_PIN;
mxconfig.gpio.g1 = G1_PIN;
mxconfig.gpio.b1 = B1_PIN;
mxconfig.gpio.r2 = R2_PIN;
mxconfig.gpio.g2 = G2_PIN;
mxconfig.gpio.b2 = B2_PIN;
mxconfig.gpio.a = A_PIN;
mxconfig.gpio.b = B_PIN;
mxconfig.gpio.c = C_PIN;
mxconfig.gpio.d = D_PIN;
mxconfig.gpio.e = E_PIN;
mxconfig.gpio.lat = LAT_PIN;
mxconfig.gpio.oe = OE_PIN;
mxconfig.gpio.clk = CLK_PIN;

//mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

dma_display = new MatrixPanel_I2S_DMA(mxconfig);
dma_display->begin();
dma_display->setBrightness8(70);
dma_display->setCursor(0, 0);
dma_display->print("HELLO");

// draw a blue circle
dma_display->drawCircle(10, 10, 10, dma_display->color444(0, 0, 15));
}

void loop() {}

included line like this     mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;
[image]

excluded line like this     //mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;
[image]

board707, 2025-05-19:
> //mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

Why this line is commented out?

nadir-dogan, 2025-05-19:
I had tried with and without the line, you may see two picture.

nadir-dogan, 2025-05-19:
#include <Arduino.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define R1_PIN 18
#define G1_PIN 14
#define B1_PIN 16
#define R2_PIN 17
#define G2_PIN 13
#define B2_PIN 21
#define A_PIN 34
#define B_PIN 8
#define C_PIN 36
#define D_PIN 4
#define LAT_PIN 2
#define OE_PIN 40
#define CLK_PIN 38
#define E_PIN 10

/********** CONFIG *************/
#define PANEL_WIDTH 64
#define PANEL_HEIGHT 32
#define PANELS_NUMBER 1

MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup() {
Serial.begin(115200);

HUB75_I2S_CFG mxconfig;
mxconfig.mx_width = PANEL_WIDTH;
mxconfig.mx_height = PANEL_HEIGHT;
mxconfig.chain_length = PANELS_NUMBER;
mxconfig.gpio.r1 = R1_PIN;
mxconfig.gpio.g1 = G1_PIN;
mxconfig.gpio.b1 = B1_PIN;
mxconfig.gpio.r2 = R2_PIN;
mxconfig.gpio.g2 = G2_PIN;
mxconfig.gpio.b2 = B2_PIN;
mxconfig.gpio.a = A_PIN;
mxconfig.gpio.b = B_PIN;
mxconfig.gpio.c = C_PIN;
mxconfig.gpio.d = D_PIN;
mxconfig.gpio.e = E_PIN;
mxconfig.gpio.lat = LAT_PIN;
mxconfig.gpio.oe = OE_PIN;
mxconfig.gpio.clk = CLK_PIN;

mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

dma_display = new MatrixPanel_I2S_DMA(mxconfig);
dma_display->begin();
dma_display->setBrightness8(70);

dma_display->setCursor(0, 0);
dma_display->print("1234567890");

// draw a blue circle
//dma_display->drawCircle(10, 10, 10, dma_display->color444(0, 0, 15));
}

void loop() {}

[image]

board707, 2025-05-19:
Your panel is 64x32 s1/8 - so it is i.e. a "four scan panel".
To work with it you have to use a VirtualPanel class and manage a pixel mapping.
Please run a Pixel_Mapping_Test and show the video.

nadir-dogan, 2025-05-19:
Slightly modified...

/*************************************************************************
   Description:

   The underlying implementation of the ESP32-HUB75-MatrixPanel-I2S-DMA only
   supports output to HALF scan panels - which means outputting
   two lines at the same time, 16 or 32 rows apart if a 32px or 64px high panel
   respectively.
   This cannot be changed at the DMA layer as it would require a messy and complex
   rebuild of the library's internals.

   However, it is possible to connect QUARTER (i.e. FOUR lines updated in parallel)
   scan panels to this same library and
   'trick' the output to work correctly on these panels by way of adjusting the
   pixel co-ordinates that are 'sent' to the ESP32-HUB75-MatrixPanel-I2S-DMA
   library.

 **************************************************************************/

/* Use the Virtual Display class to re-map co-ordinates such that they draw
   correctly on a 32x16 1/4 or 64x32 1/8 Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

// Define custom class derived from VirtualMatrixPanel
class CustomPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

// custom getCoords() method for specific pixel mapping
inline VirtualCoords CustomPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

  //uint8_t pxbase = panelResX;  // pixel base
  uint8_t pxbase = 32;  // pixel base

  // mapper for panels with 32 pixs height  (64x32 or 32x32)
  if (panelResY == 32)
  {
    if ((coords.y & 8) == 0)
    {
      coords.x += ((coords.x / pxbase) + 1) * pxbase;    // 1st, 3rd 'block' of 8 rows of pixels
    }
    else
    {
      coords.x += (coords.x / pxbase) * pxbase;          // 2nd, 4th 'block' of 8 rows of pixels
    }
    coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
  }

  // mapper for panels with 16 pixs height  (32x16 1/4)
  else if (panelResY == 16)
  {
    if ((coords.y & 4) == 0)
    {
      // 1. Normal line, from left to right
      coords.x += ((coords.x / pxbase) + 1) * pxbase;     // 1st, 3rd 'block' of 4 rows of pixels
      //2. in case the line filled from right to left, use this (and comment 1st)
      //coords.x = ((coords.x / pxbase) + 1) * 2 * pxbase  - (coords.x % pxbase) - 1;
    }
    else
    {
      coords.x += (coords.x / pxbase) * pxbase;          // 2nd, 4th 'block' of 4 rows of pixels
    }
    coords.y = (coords.y >> 3) * 4 + (coords.y & 0b00000011);
  }
  return coords;
}

// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

// Use a single panel for tests
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// Chain settings, do not cnahge
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_BOTTOM_RIGHT_UP

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
CustomPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X * 2,            // DO NOT CHANGE THIS
    PANEL_RES_Y / 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new CustomPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,  VIRTUAL_MATRIX_CHAIN_TYPE);

}

void loop() {
//  for (int i = 0; i < 32; i++){
int i=0;
    for (int j = 0; j < 64; j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(0, 63, 0));
      delay(10);
       }
    for (int j = 0; j < 64; j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(0, 0, 0));
      delay(10);
    }
//  }

/*
  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(127, 63, 0));
      delay(10);
      Serial.print(j);Serial.print("-");Serial.println(i);
    }
  }
  */
  delay(2000);
  dma_display->clearScreen();
} // end loop

https://github.com/user-attachments/assets/7ebb5889-f846-434e-9dca-cc9aea78f4fd

nadir-dogan, 2025-05-20:
Dear @board707 ,
I think it looks like OK now. Thank you for your all help.

What I must understand about these settings? How is this panel?
I was think all panels with HUB75 is same but I see a lot of panel is different each other...

#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

// Define custom class derived from VirtualMatrixPanel
class CustomPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

// custom getCoords() method for specific pixel mapping
inline VirtualCoords CustomPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

  //uint8_t pxbase = panelResX;  // pixel base
  uint8_t pxbase = 32;  // pixel base

  // mapper for panels with 32 pixs height  (64x32 or 32x32)
  if (panelResY == 32)
  {
    if ((coords.y & 8) == 0)
    {
      coords.x += ((coords.x / pxbase) + 1) * pxbase;    // 1st, 3rd 'block' of 8 rows of pixels
    }
    else
    {
      coords.x += (coords.x / pxbase) * pxbase;          // 2nd, 4th 'block' of 8 rows of pixels
    }
    coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
  }

  // mapper for panels with 16 pixs height  (32x16 1/4)
  else if (panelResY == 16)
  {
    if ((coords.y & 4) == 0)
    {
      // 1. Normal line, from left to right
      coords.x += ((coords.x / pxbase) + 1) * pxbase;     // 1st, 3rd 'block' of 4 rows of pixels
      //2. in case the line filled from right to left, use this (and comment 1st)
      //coords.x = ((coords.x / pxbase) + 1) * 2 * pxbase  - (coords.x % pxbase) - 1;
    }
    else
    {
      coords.x += (coords.x / pxbase) * pxbase;          // 2nd, 4th 'block' of 4 rows of pixels
    }
    coords.y = (coords.y >> 3) * 4 + (coords.y & 0b00000011);
  }
  return coords;
}

// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

// Use a single panel for tests
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// Chain settings, do not cnahge
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_BOTTOM_RIGHT_UP

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
CustomPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X * 2,            // DO NOT CHANGE THIS
    PANEL_RES_Y / 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

  mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new CustomPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,  VIRTUAL_MATRIX_CHAIN_TYPE);

}

void loop() {
  for (int i = 0; i < 32; i++){

    for (int j = 0; j < 64; j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(0, 63, 0));
      delay(10);
       }
  }

/*
  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(127, 63, 0));
      delay(10);
      Serial.print(j);Serial.print("-");Serial.println(i);
    }
  }
  */
  delay(2000);
  dma_display->clearScreen();
} // end loop

https://github.com/user-attachments/assets/63d929fc-0c87-4975-ade8-30d0747ce7b5

board707, 2025-05-20:
My congratulations!

nadir-dogan, 2025-05-20:
Congratulations for you :)

board707, 2025-05-20:
> I was think all panels with HUB75 is same but I see a lot of panel is different each other...

You are right, there are a lot of panel driver chips - and many of them require a special protocol to work with them.
I am always happy if I can help

nadir-dogan, 2025-05-20:
Dear @board707 ,
I was waiting to see clear work but unexpected problem I saw :(
What can you say about this?

void loop() {
    for (int x = 0; x < 4; x++)
    {
      FourScanPanel->setCursor(0, x*8);
      FourScanPanel->print("1234567890");
      delay(2000);
       }
dma_display->clearScreen();
delay(2000);
} // end loop

https://github.com/user-attachments/assets/e2b6df0b-943e-4b76-8f23-65fbb4da419a

board707, 2025-05-20:
I can't say for sure... Looks like your power supply don't have enough power to light whole panel...

nadir-dogan, 2025-05-20:
My power supply is 5V 40A and I measure voltage is 5.06V and never changed.
I think about A B C D E pins. May be needs disconnect unused pins.

board707, 2025-05-20:
Sorry, I have not another ideas

awanish-singh-kl0335, 2026-01-28:
Hi....
I am speaking from INDIA (GMT+5:30).
Please tell me your time window in which you can help me so that i can reply to you in the same time window. I think this is better way to connect