mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #646: Display errors with ESP32 S3, but ESP32 works

mkalkbrenner, 2024-06-23:
I'm currently adding ESP32 S3 support to ZeDMD.
If I only change the the default pins according to https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/src/platforms/esp32s3/esp32s3-default-pins.hpp , disconnect the ESP32 and connect the ESP32 S3 to the same LED panels, it basically works, but the content is erroneous:
[image]

4 rows are displayed, then 4 rows are black.  And it seems that the content of the 4 black rows is shifted up or down and mixed into the 4 visible rows.

The picture is taken from a 256x64 panel setup. But same error happens with the 128x32 setup.
Any ideas?

I tried different versions of the arduino framework and even the latest version of https://github.com/tasmota/platform-espressif32
I also tried different versions and the current master of ESP32-HUB75-MatrixPanel-DMA, the result was always the same.

mkalkbrenner, 2024-06-24:
It seems to be an issue with the break out board for the S3. Pin 3 to HUB75 pin C has no connection.

mkalkbrenner, 2024-06-25:
yes, it is a hardware issue. The breakout board connects to IO pin 46 instead of 3.
Is it safe to do that mapping on the S3 or does that pin has any speed limitations, so it might not be suitable?
```cpp
#define C_PIN 46
```

toine512, 2024-06-29:
GPIO 46 has a special use during startup, better not use it if you have other options.
ESP32-S3 TRM section 8.1

mkalkbrenner, 2024-06-30:
@toine512 Thanks for your feedback!

closed 2024-07-04 by mrcodetastic