mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #647: HUB75 -> No Longer Works on ESP32 & S3/S2... Anymore!..

Knottis, 2024-06-24:
### Not working on ESP32-S3

**I don't think HUB75 is works anymore on ESP32 & S3/S2 ?..**

_(I got a New and Clean PC and installed only the latest of "these" drivers)._

---

### Can Developer test HUB75 with the latest version of:

**Arduino IDE - 2.3.2
ESP32 – 3.0.1**

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <GFX.h>
#include <Adafruit_GFX.h>

----
**On the following modules:**

ESP32S3 Dev Module
ESP32 Dev Module

davemaster, 2024-06-24:
Specifically, what  ESP32-S3 board do You have? Picture....

Knottis, 2024-06-24:
### I use this card::
----

[image]

----

[image]

----

[image]

toine512, 2024-06-29:
Works with ESP-IDF 5.2.1 on ESP32-S3, with a shit clone of YD-ESP32-S3 board.
(no GFX, no Arduino)

Knottis, 2024-07-03:
### No this won't work!.

**Have a new and clean PC with only the necessary libraries installed.**

**Arduino IDE 2.3.2**  =>  https://www.arduino.cc/en/software
**ESP 3.0.2**  => https://github.com/espressif/arduino-esp32

Using library ESP32 HUB75 LED MATRIX PANEL DMA Display at version **3.0.10**
Using library Adafruit GFX Library at version **1.11.9**  
Using library FastLED at version **3.7.0** 
Using library Adafruit BusIO at version **1.16.1** 
Using library Wire at version **2.0.0** 
Using library SPI at version **2.0.0** 

----

[image]

----

### PatternPlasma.ino   

_Test compiles with default values:_

**Compilation error:**

FQBN: esp32:esp32:esp32s3:FlashSize=16M,PartitionScheme=default
Using board 'esp32s3' from platform in folder: C:\Users\XYZ\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2
Using core 'esp32' from platform in folder: C:\Users\XYZ\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2

cmd /c if exist "C:\\Users\\XYZ\\AppData\\Local\\Temp\\.arduinoIDE-unsaved202463-17068-qs6et8.y8t7n\\2_PatternPlasma\\partitions.csv" COPY /y "C:\\Users\\XYZ\\AppData\\Local\\Temp\\.arduinoIDE-unsaved202463-17068-qs6et8.y8t7n\\2_PatternPlasma\\partitions.csv" "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\partitions.csv"
cmd /c if not exist "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\partitions.csv" if exist "C:\\Users\\XYZ\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.2\\variants\\esp32s3\\partitions.csv" COPY "C:\\Users\\XYZ\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.2\\variants\\esp32s3\\partitions.csv" "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\partitions.csv"
cmd /c if not exist "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\partitions.csv" COPY "C:\\Users\\XYZ\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.2\\tools\\partitions\\default.csv" "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\partitions.csv"
cmd /c IF EXIST "C:\\Users\\XYZ\\AppData\\Local\\Temp\\.arduinoIDE-unsaved202463-17068-qs6et8.y8t7n\\2_PatternPlasma\\bootloader.bin" ( COPY /y "C:\\Users\\XYZ\\AppData\\Local\\Temp\\.arduinoIDE-unsaved202463-17068-qs6et8.y8t7n\\2_PatternPlasma\\bootloader.bin" "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\2_PatternPlasma.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\XYZ\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.2\\variants\\esp32s3\\bootloader.bin" ( COPY "C:\\Users\\XYZ\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.2\\variants\\esp32s3\\bootloader.bin" "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\2_PatternPlasma.ino.bootloader.bin" ) ELSE ( "C:\\Users\\XYZ\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.6\\esptool.exe" --chip esp32s3 elf2image --flash_mode dio --flash_freq 80m --flash_size 16MB -o "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\2_PatternPlasma.ino.bootloader.bin" "C:\\Users\\XYZ\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-bd2b9390ef\\esp32s3\\bin\\bootloader_qio_80m.elf" ) )
esptool.py v4.6
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
cmd /c if exist "C:\\Users\\XYZ\\AppData\\Local\\Temp\\.arduinoIDE-unsaved202463-17068-qs6et8.y8t7n\\2_PatternPlasma\\build_opt.h" COPY /y "C:\\Users\\XYZ\\AppData\\Local\\Temp\\.arduinoIDE-unsaved202463-17068-qs6et8.y8t7n\\2_PatternPlasma\\build_opt.h" "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\build_opt.h"
cmd /c if not exist "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\build_opt.h" type nul > "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3\\build_opt.h"
cmd /c type nul > "C:\\Users\\XYZ\\AppData\\Local\\Temp\\arduino\\sketches\\82A3FD3D10B64790BC2DD14A8CA4D0A3/file_opts"
Detecting libraries used...
C:\Users\XYZ\AppData\Local\Arduino15\packages\esp32\tools\esp-xs3\2302/bin/xtensa-esp32s3-elf-g++ -c @C:\Users\XYZ\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32s3/flags/cpp_flags -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32S3_DEV" -DARDUINO_VARIANT="esp32s3" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=16M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 @C:\Users\XYZ\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32s3/flags/defines -IC:\Users\XYZ\AppData\Local\Temp\.arduinoIDE-unsaved202463-17068-qs6et8.y8t7n\2_PatternPlasma -iprefix C:\Users\XYZ\AppData\Local\Arduino15\packages [... cut ...]
Using library Adafruit BusIO at version 1.16.1 in folder: C:\Users\XYZ\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 2.0.0 in folder: C:\Users\XYZ\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\Wire 
Using library SPI at version 2.0.0 in folder: C:\Users\XYZ\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\SPI 
**_Using library FastLED at version 3.7.0 in folder: C:\Users\XYZ\Documents\Arduino\libraries\FastLED 
exit status 1_**

**Compilation error: exit status 1**

----

mrcodetastic, 2024-07-04:
Can you create / upload a sample project that causes this issue please?

Knottis, 2024-07-04:
### This is your "own program" example that comes with the library!..

_You cannot use the development PC for testing. You have to start testing on a clean PC that, for example, runs in **Virtualbox**, so you quickly see what works and what doesn't work!.._

---------

### I have a New and Clean PC with only the necessary libraries installed.

**Arduino IDE 2.3.2** => https://www.arduino.cc/en/software
**ESP 3.0.2** => https://github.com/espressif/arduino-esp32

Using library **ESP32 HUB75 LED MATRIX PANEL DMA Display at version 3.0.10**
Using library **Adafruit GFX Library at version 1.11.9**
Using library **FastLED at version 3.7.0**
Using library **Adafruit BusIO at version 1.16.1**
Using library **Wire at version 2.0.0**
Using library **SPI at version 2.0.0**

---------

```c

/*
 * Portions of this code are adapted from Aurora: https://github.com/pixelmatix/aurora
 * Copyright (c) 2014 Jason Coon
 *
 * Portions of this code are adapted from LedEffects Plasma by Robert Atkins: https://bitbucket.org/ratkins/ledeffects/src/26ed3c51912af6fac5f1304629c7b4ab7ac8ca4b/Plasma.cpp?at=default
 * Copyright (c) 2013 Robert Atkins
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy of
 * this software and associated documentation files (the "Software"), to deal in
 * the Software without restriction, including without limitation the rights to
 * use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
 * the Software, and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 */
 

// HUB75E pinout
// R1 | G1
// B1 | GND
// R2 | G2
// B2 | E
//  A | B
//  C | D
// CLK| LAT
// OE | GND

/*  Default library pin configuration for the reference
  you can redefine only ones you need later on object creation

#define R1 25
#define G1 26
#define BL1 27
#define R2 14
#define G2 12
#define BL2 13
#define CH_A 23
#define CH_B 19
#define CH_C 5
[... 136 lines cut ...]
    ++cycles;
    ++fps;

    if (cycles >= 1024) {
        time_counter = 0;
        cycles = 0;
        currentPalette = palettes[random(0,sizeof(palettes)/sizeof(palettes[0]))];
    }

    // print FPS rate every 5 seconds
    // Note: this is NOT a matrix refresh rate, it's the number of data frames being drawn to the DMA buffer per second
    if (fps_timer + 5000 < millis()){
      Serial.printf_P(PSTR("Effect fps: %d\n"), fps/5);
      fps_timer = millis();
      fps = 0;
    }
} // end loop

``` 

---

### 😬
Have double checked several times. Has also reconnected several times. 
_But this has nothing to do with getting errors when you compile the example code!.._

board707, 2024-07-04:
> But this has nothing to do with getting errors when you compile the example code!..

and all your errors has nothing to do with this library..
I think  the cause of the errors is a new Expessif 3.x.x code. 
Downgrade it to latest 2.x.x. version and all errors will gone, I think

Knottis, 2024-07-04:
### 😬 _"I will never downgrade any software or drivers!..."_
🛠⚙📝😉So here the _Development Team_ has something to deal with to get sorted out/work on, this summer!..
---
🔍**Arduino** will "_automatically_" update all of the library!..

---

**Arduino IDE 2.3.2
ESP 3.0.2**

board707, 2024-07-04:
> "I will never downgrade any software or drivers!..."

I have different rules - I never update software without a reason :)

harissutanrafiq, 2024-07-10:
> > "I will never downgrade any software or drivers!..."
> 
> I have different rules - I never update software without a reason :)

i never update software when everything its work :D