mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #651: 64x32 P4 Panel with ICs ICN2037BP and 74HC1385 "half blank" last row, ghost pixel with ESP32 D1 MINI

RestlessRabbits, 2024-07-05:
Hi,
I am trying to get the Tetris Clock to work as an example and tried different driver settings without success or any change in behavior.

The pintout/connections:
| HUB75 | GPIO |
| ------------- | ------------- |
| R1 | 25 |
| G1 | 26  |
| B1 | 27 |
| R2  | 14 |
| G2  | 12 |
| B2  | 13 |
| A  |  23 |
| B  |  19 |
| C  |  5 |
| D  |  17 |
| E   |  18 |
| LAT  | 4 |
| OE  | 15 |
| CLK  | 16 |

I tried the drivers
-  HUB75_I2S_CFG::FM6126A
-  HUB75_I2S_CFG::SHIFTREG
-  HUB75_I2S_CFG::ICN2038S
without any changes in behavior.
Here is an image of the issues compared to a Waveshare 64x32 P2.5 Panel:
[image]
Here is the back of the panel I struggle to get to work:
[image]

The Panels EAN is 4060137084737.
They also are listed with "ListenVision" and QITA as a producer.
For driving it lists "1/32-Sweep-Constantcurrentdrive" - which I just thought to be 1/32 scan, but maybe I'm mistaken here.
Does anybody have experience or hints to get this display to work?

Thank you very much in advance!

board707, 2024-07-05:
Your panel is built on drivers ICN2037 & 74HC138. These are the most commonplace drivers that do not require any special settings to work. The only thing that makes your panel unusual is that with a height of 32, it also has a scan of 32. Such panels have already come across here. Unfortunately, no one has yet figured out how to work with such panels. I dream that such a panel will fall into my hands.
You can easily run a single panel by setup it as a 64x64 matrix. But so far no one has succeeded in connecting two or more such panels so that they work correctly

RestlessRabbits, 2024-07-05:
Thank you very much! It works - I have to make adjustmends to the rest of the code because of the "wrong" assumtions in size, but in general, I finally get something displayed on the bottom half as well.
Since I have two of these and one is planned to be used. Where are you located? Maybe I can send you at least that one panel to figure a few things out (sadly only one spare, so for chaining tests, you would need to get your hands on another)

board707, 2024-07-05:
Thank you, but I hope to get such panel soon.  Or perhaps someone else has already figured out such panels and will tell you how to work with them

closed 2024-07-07 by RestlessRabbits