mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #652: LED panel with ICN2038S have graphical distortions

PodmogilnyiEvgenii, 2024-07-06:
Hello! 
I have led matrix 128x64 with ICN2038S. It works well with the receiving card Colorlight 5A-75B.
https://www.youtube.com/shorts/qVjnT78mICI

When I used the library I got graphic distortions on the panel.
//my program configuration panel
`HUB75_I2S_CFG::i2s_pins _pins = { R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN };
HUB75_I2S_CFG mxconfig(
    128,                     // width
    64,                      // height
    1,                       // chain length
    _pins,                   // pin mapping
    HUB75_I2S_CFG::ICN2038S  // driver chip
);

mxconfig.latch_blanking = 4;
mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;
mxconfig.clkphase = false;`

ICN2038S:
[image]

Same program with ICN2037:
[image]

What can be done to solve this problem?

mrcodetastic, 2024-07-23:
Reduce the clock rate perhaps?

mrcodetastic, 2024-08-24:
Did you find the solution to the issue?