mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #653: Is there support for 128x64 Panel with ICND1065 chip

KEBAP5000, 2024-07-08:
Hello!

Thanks for your great library!
Today i got new panels with 128x64px (HUB75E) and chip ICND1065. The problem is, that the whole display stays black. Is this chip not supported?

[image]

Thanks for your answer!

board707, 2024-07-08:
According to my databse, ICDN1065 is a s-PWM type driver. It means that the panel hardly will works with the library.

KEBAP5000, 2024-07-09:
Thanks for the quick reply!

Do you happen to know whether it only doesn't work with this library, or whether it generally can't work with the ESP32?

And if it could work with another library, which one would that be? :)

closed 2024-07-09 by KEBAP5000

board707, 2024-07-09:
I don't know the library (for any board), that support this driver

KEBAP5000, 2024-07-09:
Thanks!

One last question: Do you happen to know which 128x64 (P2.5) Panel (or which chips) would work with the ESP32? And do you know a reseller for those?

board707, 2024-07-09:
I don't know the specific panels. As for chip models, I have a table of drivers that people have tried to run with this and other libraries. Of course, it is not complete and may be inaccurate, but in principle this data can be useful.
[LED Drivers database](https://github.com/board707/DMD_STM32/wiki/Led_drivers)

ibhenkz, 2024-09-17:
Hi board707, I have same issue with chip ICND1065 pairing chip 74HC245. There is no response in panel 128x64 (fully black).
are you plan to make a config for enable those chips, because I'am really happy using ESP32-HUB75-MatrixPanel-DMA for many projects. Thanks for Advance.

board707, 2024-09-17:
> are you plan to make a config for enable those chips, because I'am really happy using ESP32-HUB75-MatrixPanel-DMA for many projects.

Hi
First of all, the ESP32-HUB75-MatrixPanel-DMA is not my project, so you should ask the author - @mrcodetastic.

As for me, I am interesting to develop code for S-PWM drivers, but it is possible only for panels that I have available. 
I don't have panels with the 1065 driver, so I won't be working on them in the near future.

ibhenkz, 2024-09-18:
> Hi First of all, the ESP32-HUB75-MatrixPanel-DMA is not my project, so you should ask the author - @mrcodetastic.
>
> As for me, I am interesting to develop code for S-PWM drivers, but it is possible [...]

Hi board707,
I am sorry and Thanks for information..