mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #654: Ghosting  and other problems with a FM6124 driver

B-Zimmer, 2024-07-10:
Hello, I know this driver isn't listed, but I still thought I'd give it a try and see if I can get this working properly. I have a 64x32 panel.

[image]

I have read through the "READ THIS FIRST" section, I've tried changing the latch blanking and i2sspeed variables. This did seem to help, but overall ghosting was still present. I am in the process of shortening jumper wires to give it another attempt. I also have vertical output alignment issues but am in the process of fixing this. 

Furthermore, I am also seemingly unable to change brightness, but that may be a problem for another day. I can give more info or upload more photos if needed. Could this be a driver issue? A panel issue? Or something else outside my understanding?

Thank you!
Best of luck.

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/161541910/6023f98e-6d78-47fe-a24d-0b121e140542

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/assets/161541910/3d49ef65-870c-46c0-8526-55371f880753

board707, 2024-07-10:
Hi
Please show a clear photo of the panel's rear side.
Please provide a names of ALL chips on the panel - you have to find at least three types of them.

B-Zimmer, 2024-07-10:
Yup can do. I have fixed the main issue (it was the cable length, thank you for including that in the READ ME FIRST page). However, I am still getting a split between the top and bottom halves (visible in videos). 

the three chip names are:

SM5166PS
FM6124D
74HC245KA

Thank you for the help so far

[image]
[image]

board707, 2024-07-10:
> SM5166PS
> FM6124D

These are standard drivers, that doesn't need any specific settings to run. 
Sorry, but I have no ideas regarding your issue.

B-Zimmer, 2024-07-12:
No worries, thank you either way.

mrcodetastic, 2024-07-23:
Try PX matrix library and see if it works with that.