mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #664: Cyrillic Font Rendering Issues on 64x32 LED Matrix Display

Ezopython, 2024-08-12:
Describe the bug
I am experiencing issues with rendering Cyrillic characters on a 64x32 LED matrix display. The text appears distorted, with characters overlapping and misaligned.

To Reproduce
Steps to reproduce the behavior:

Use the provided Cyrillic font data with the Adafruit GFX or similar library.
Render text on a 64x32 LED matrix display.
Observe the overlapping and misaligned characters.
Expected behavior
The Cyrillic characters should render clearly without overlapping, with correct alignment and spacing.

#include <Adafruit_GFX.h>
#include <RGBmatrixPanel.h>

const uint8_t MyCyrillicFontBitmaps[] PROGMEM = {
    // (Include the bitmap array here)
};

const GFXglyph MyCyrillicFontGlyphs[] PROGMEM = {
    // (Include the glyph array here)
};

const GFXfont MyCyrillicFont PROGMEM = {
    (uint8_t  *)MyCyrillicFontBitmaps,
    (GFXglyph *)MyCyrillicFontGlyphs,
    0x0410, 0x044F, 11   // Start and end character, font height
};

void setup() {
    // Initialize display here

    // Set font
    display.setFont(&MyCyrillicFont);

    // Render text
    display.setCursor(0, 0);
    display.print(F("Привет мир!"));
    display.show();
}

void loop() {
  // Nothing here
}
[image]

board707, 2024-08-12:
> const uint8_t MyCyrillicFontBitmaps[] PROGMEM = {
// (Include the bitmap array here)
};

> const GFXglyph MyCyrillicFontGlyphs[] PROGMEM = {
// (Include the glyph array here)
};

Your Cyrillic font bitmaps and glyphs are empty, what are you expecting to see?

Ezopython, 2024-08-12:
> > const uint8_t MyCyrillicFontBitmaps[] PROGMEM = {
> > // (Include the bitmap array here)
> > };
> 
> > const GFXglyph MyCyrillicFontGlyphs[] PROGMEM = {
> > // (Include the glyph array here)
> > };
> 
> Your Cyrillic font bitmaps and glyphs are empty, what are you expecting to see?
-- full version is here 
#pragma once
#include <stdint.h>
const uint8_t MyCyrillicFontBitmaps[] PROGMEM = {
    0x18, 0x18, 0x38, 0x3C, 0x3C, 0x3C, 0x66, 0x7E, 0x66, 0xC6, 0xC2, 
    0x7E, 0x60, 0x60, 0x60, 0x7E, 0x67, 0x63, 0x63, 0x63, 0x67, 
    0x7E, 0x7E, 0x6E, 0x62, 0x6E, 0x7E, 0x67, 0x63, 0x63, 0x63, 
    0x67, 0x7E, 0x3F, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 
    0x30, 0x30, 0x30, 0x3E, 0x36, 0x26, 0x26, 0x26, 0x66, 0x66, 
    0x66, 0x66, 0x66, 0xFF, 0xC3, 0x3F, 0x30, 0x30, 0x30, 0x3C, 
    0x30, 0x30, 0x30, 0x30, 0x30, 0x3F, 0x6E, 0xF6, 0xFC, 0x7C, 
    0x78, 0x78, 0x7C, 0xFC, 0xF6, 0x6E, 0x92, 0x3E, 0x3F, 0x23, 
    0x07, 0x1E, 0x07, 0x03, 0x03, 0x03, 0x77, 0x3E, 0x00, 0x63, 
    0x67, 0x67, 0x6F, 0x6F, 0x7F, 0x7B, 0x7B, 0x73, 0x73, 0x63, 
    0x1C, 0x63, 0x67, 0x67, 0x6F, 0x6F, 0x7F, 0x7B, 0x7B, 0x73, 
    0x73, 0x63, 0x63, 0x67, 0x66, 0x6C, 0x7C, 0x78, 0x7C, 0x6C, 
    0x66, 0x66, 0x63, 0x63, 0x1F, 0x1B, 0x1B, 0x33, 0x33, 0x33, 
    0x63, 0x63, 0x63, 0xE3, 0xC3, 0x82, 0xC6, 0xCE, 0xEE, 0xFA, 
    0xBA, 0x82, 0x82, 0x82, 0x82, 0x82, 0x82, 0x63, 0x63, 0x63, 
    0x63, 0x63, 0x7F, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x3E, 
    0x77, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x77, 0x3E, 
    0x7F, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 
    0x63, 0x7E, 0x67, 0x63, 0x63, 0x67, 0x7E, 0x60, 0x60, 0x60, 
    0x60, 0x60, 0x1F, 0x38, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 
    0x30, 0x38, 0x1F, 0x7F, 0x08, 0x08, 0x08, 0x08, 0x08, 0x08, 
    0x08, 0x08, 0x08, 0x08, 0x63, 0x63, 0x63, 0x63, 0x63, 0x73, 
    0x3F, 0x03, 0x03, 0x03, 0x07, 0x3E, 0x30, 0x30, 0xFE, 0xFE, 
    0xB3, 0x33, 0x33, 0xB3, 0xFE, 0xFE, 0x30, 0x30, 0xC7, 0x66, 
    0x6E, 0x3C, 0x3C, 0x18, 0x3C, 0x3C, 0x6E, 0x66, 0xC7, 0xC6, 
    0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xFF, 
    0x03, 0x03, 0x63, 0x63, 0x63, 0x63, 0x63, 0x73, 0x3F, 0x03, 
    0x03, 0x03, 0x03, 0x03, 0x92, 0xB2, 0xB2, 0xB2, 0xB2, 0xB2, 
    0xB2, 0xB2, 0xB2, 0xB2, 0xB2, 0xFE, 0x92, 0xB2, 0xB2, 0xB2, 
    0xB2, 0xB2, 0xB2, 0xB2, 0xB2, 0xB2, 0xB2, 0xFF, 0x03, 0x03, 
    0xE0, 0x60, 0x60, 0x60, 0x7E, 0x6E, 0x66, 0x62, 0x66, 0x6E, 
    0x7E, 0xC2, 0xC3, 0xC3, 0xC3, 0xC3, 0xC3, 0xC3, 0xC3, 0xC3, 
    0xC3, 0xC3, 0xFF, 0x60, 0x60, 0x63, 0x63, 0x63, 0x63, 0x63, 
    0x67, 0x66, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 
    0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xFF, 0x03, 0x03, 0x7E, 0x60, 
    0x60, 0x60, 0x60, 0x60, 0xE0, 0x60, 0x60, 0x60, 0x60, 0x60, 
    0x3F, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 
    0x30, 0x3E, 0x36, 0x26, 0x26, 0x26, 0x66, 0x66, 0x66, 0x66, 
    0x66, 0xFF, 0xC3, 0x3F, 0x30, 0x30, 0x30, 0x3C, 0x30, 0x30, 
    0x30, 0x30, 0x30, 0x3F, 0x6E, 0xF6, 0xFC, 0x7C, 0x78, 0x78, 
    0x7C, 0xFC, 0xF6, 0x6E, 0x92, 0x3E, 0x3F, 0x23, 0x07, 0x1E, 
    0x07, 0x03, 0x03, 0x03, 0x77, 0x3E, 0x00, 0x63, 0x67, 0x67, 
    0x6F, 0x6F, 0x7F, 0x7B, 0x7B, 0x73, 0x73, 0x63, 0x1C, 0x63, 
    0x67, 0x67, 0x6F, 0x6F, 0x7F, 0x7B, 0x7B, 0x73, 0x73, 0x63, 
    0x63, 0x67, 0x66, 0x6C, 0x7C, 0x78, 0x7C, 0x6C, 0x66, 0x66, 
    0x63, 0x63, 0x1F, 0x1B, 0x1B, 0x33, 0x33, 0x33, 0x63, 0x63, 
    0x63, 0xE3, 0xC3, 0x82, 0xC6, 0xCE, 0xEE, 0xFA, 0xBA, 0x82, 
    0x82, 0x82, 0x82, 0x82, 0x82
};
const GFXglyph MyCyrillicFontGlyphs[] PROGMEM = {
    {     0,   8,  11,   9,    0,  -11 },   // 0x0410 'А'
    {    11,   7,  11,   8,    0,  -11 },   // 0x0411 'Б'
    {    22,   7,  11,   8,    0,  -11 },   // 0x0412 'В'
    {    33,   7,  11,   8,    0,  -11 },   // 0x0413 'Г'
    {    44,   8,  11,   9,    0,  -11 },   // 0x0414 'Д'
    {    55,   7,  11,   8,    0,  -11 },   // 0x0415 'Е'
    {    66,   9,  11,  10,    0,  -11 },   // 0x0416 'Ж'
[... 39 lines cut ...]
    {   516,   7,  11,   8,    0,  -11 },   // 0x043E 'о'
    {   527,   7,  11,   8,    0,  -11 },   // 0x043F 'п'
    {   538,   7,  11,   8,    0,  -11 },   // 0x0440 'р'
    {   549,   7,  11,   8,    0,  -11 },   // 0x0441 'с'
    {   560,   7,  11,   8,    0,  -11 },   // 0x0442 'т'
    {   571,   7,  11,   8,    0,  -11 },   // 0x0443 'у'
    {   582,   7,  11,   8,    0,  -11 },   // 0x0444 'ф'
    {   593,   7,  11,   8,    0,  -11 },   // 0x0445 'х'
    {   604,   7,  11,   8,    0,  -11 },   // 0x0446 'ц'
    {   615,   7,  11,   8,    0,  -11 },   // 0x0447 'ч'
    {   626,   9,  11,  10,    0,  -11 },   // 0x0448 'ш'
    {   639,   9,  11,  10,    0,  -11 },   // 0x0449 'щ'
    {   652,   8,  11,   9,    0,  -11 },   // 0x044A 'ъ'
    {   663,   8,  11,   9,    0,  -11 },   // 0x044B 'ы'
    {   674,   7,  11,   8,    0,  -11 },   // 0x044C 'ь'
    {   685,   7,  11,   8,    0,  -11 },   // 0x044D 'э'
    {   696,   9,  11,  10,    0,  -11 },   // 0x044E 'ю'
    {   709,   7,  11,   8,    0,  -11 }    // 0x044F 'я'
};

const GFXfont MyCyrillicFont PROGMEM = {
    (uint8_t  *)MyCyrillicFontBitmaps,
    (GFXglyph *)MyCyrillicFontGlyphs,
    0x0410, 0x044F, 11   // Start and end character, font height
};

board707, 2024-08-12:
> { 0, 8, 11, 9, 0, -11 }, // 0x0410 'А'

According to the comments, you use UTF16 cyrillic codes, while Arduino IDE uses UTF8 codes.
For example, the  'А' letter has code 0xD090 in UTF8 codetable. 

See more:
https://www.fileformat.info/info/unicode/char/0410/index.htm

Ezopython, 2024-08-12:
thans for avialable for me, i use vscode-> PlatformIO IDE  which is same with arduino IDE i guess. Here is my full bitmap and glyph data for all Cyrillic characters in UTF-8 encoding according to advise. after try below code i got nothing displayed
#include <stdint.h>

// Bitmap data for Cyrillic characters
const uint8_t MyCyrillicFontBitmaps[] PROGMEM = {
    0x18, 0x18, 0x38, 0x3C, 0x3C, 0x3C, 0x66, 0x7E, 0x66, 0xC6, 
    0xC2, 0x7E, 0x60, 0x60, 0x60, 0x7E, 0x67, 0x63, 0x63, 0x63,
    0x67, 0x7E, 0x7E, 0x6E, 0x62, 0x6E, 0x7E, 0x67, 0x63, 0x63,
    0x63, 0x67, 0x7E, 0x3F, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30,
    0x30, 0x30, 0x30, 0x3E, 0x36, 0x26, 0x26, 0x26, 0x66, 0x66,
    0x66, 0x66, 0x66, 0xFF, 0xC3, 0x3F, 0x30, 0x30, 0x30, 0x3C,
    0x30, 0x30, 0x30, 0x30, 0x30, 0x3F, 0xB6, 0xF6, 0xFC, 0x7C,
    0x78, 0x78, 0x7C, 0xFC, 0xF6, 0xB6, 0x92, 0x3E, 0x3F, 0x23,
    0x07, 0x1E, 0x07, 0x03, 0x03, 0x03, 0x77, 0x3E, 0x00, 0x63,
    0x67, 0x67, 0x6F, 0x6F, 0x7F, 0x7B, 0x7B, 0x73, 0x73, 0x63,
    0x1C, 0x63, 0x67, 0x67, 0x6F, 0x6F, 0x7F, 0x7B, 0x7B, 0x73,
    0x73, 0x63, 0x63, 0x67, 0x66, 0x6C, 0x7C, 0x78, 0x7C, 0x6C,
    0x66, 0x66, 0x63, 0x63, 0x1F, 0x1B, 0x1B, 0x33, 0x33, 0x33,
    0x63, 0x63, 0x63, 0xE3, 0xC3, 0x82, 0xC6, 0xCE, 0xEE, 0xFA,
    0xBA, 0x82, 0x82, 0x82, 0x82, 0x82, 0x82, 0x63, 0x63, 0x63,
    0x63, 0x63, 0x7F, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x3E,
    0x77, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x77, 0x3E,
    0x7F, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63,
    0x63, 0x7E, 0x67, 0x63, 0x63, 0x67, 0x7E, 0x60, 0x60, 0x60,
    0x60, 0x60, 0x1F, 0x38, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30,
    0x30, 0x38, 0x1F, 0x7F, 0x08, 0x08, 0x08, 0x08, 0x08, 0x08,
    0x08, 0x08, 0x08, 0x08, 0x63, 0x63, 0x63, 0x63, 0x63, 0x73,
    0x3F, 0x03, 0x03, 0x03, 0x07, 0x3E, 0x30, 0x30, 0xFE, 0xFE,
    0xB3, 0x33, 0x33, 0xB3, 0xFE, 0xFE, 0x30, 0x30, 0xC7, 0x66,
    0x6E, 0x3C, 0x3C, 0x18, 0x3C, 0x3C, 0x6E, 0x66, 0xC7, 0xC6,
    0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xFF,
    0x03, 0x03, 0x63, 0x63, 0x63, 0x63, 0x63, 0x73, 0x3F, 0x03,
    0x03, 0x03, 0x03, 0x03, 0x92, 0xB2, 0xB2, 0xB2, 0xB2, 0xB2,
    0xB2, 0xB2, 0xB2, 0xB2, 0xB2, 0xFE, 0x92, 0xB2, 0xB2, 0xB2,
    0xB2, 0xB2, 0xB2, 0xB2, 0xB2, 0xB2, 0xB2, 0xFF, 0x03, 0x03,
    0xE0, 0x60, 0x60, 0x60, 0x7E, 0x6E, 0x66, 0x62, 0x66, 0x6E,
    0x7E, 0xC2, 0xC3, 0xC3, 0xC3, 0xC3, 0xC3, 0xC3, 0xC3, 0xC3,
    0xC3, 0xC3, 0xFF, 0x60, 0x60, 0x63, 0x63, 0x63, 0x63, 0x63,
    0x67, 0x66, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6,
    0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xFF, 0x03, 0x03, 0x7E, 0x60,
    0x60, 0x60, 0x60, 0x60, 0xE0, 0x60, 0x60, 0x60, 0x60, 0x60,
    0x3F, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30,
    0x30, 0x3E, 0x36, 0x26, 0x26, 0x26, 0x66, 0x66, 0x66, 0x66,
    0x66, 0xFF, 0xC3, 0x3F, 0x30, 0x30, 0x30, 0x3C, 0x30, 0x30,
    0x30, 0x30, 0x30, 0x3F, 0xB6, 0xF6, 0xFC, 0x7C, 0x78, 0x78,
    0x7C, 0xFC, 0xF6, 0xB6, 0x92, 0x3E, 0x3F, 0x23, 0x07, 0x1E,
    0x07, 0x03, 0x03, 0x03, 0x77, 0x3E, 0x00, 0x63, 0x67, 0x67,
    0x6F, 0x6F, 0x7F, 0x7B, 0x7B, 0x73, 0x73, 0x63, 0x1C, 0x63,
    0x67, 0x67, 0x6F, 0x6F, 0x7F, 0x7B, 0x7B, 0x73, 0x73, 0x63,
    0x63, 0x67, 0x66, 0x6C, 0x7C, 0x78, 0x7C, 0x6C, 0x66, 0x66,
    0x63, 0x63, 0x1F, 0x1B, 0x1B, 0x33, 0x33, 0x33, 0x63, 0x63,
    0x63, 0xE3, 0xC3, 0x82, 0xC6, 0xCE, 0xEE, 0xFA, 0xBA, 0x82,
    0x82, 0x82, 0x82, 0x82, 0x82, 0x63, 0x63, 0x63, 0x63, 0x63,
    0x7F, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x3E, 0x77, 0x63,
    0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x77, 0x3E, 0x7F, 0x63,
    0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x63, 0x7E,
    0x67, 0x63, 0x63, 0x67, 0x7E, 0x60, 0x60, 0x60, 0x60, 0x60,
    0x1F, 0x38, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x38,
    0x1F, 0x7F, 0x08, 0x08, 0x08, 0x08, 0x08, 0x08, 0x08, 0x08,
    0x08, 0x08, 0x63, 0x63, 0x63, 0x63, 0x63, 0x73, 0x3F, 0x03,
    0x03, 0x03, 0x07, 0x3E, 0x30, 0x30, 0xFE, 0xFE, 0xB3, 0x33,
    0x33, 0xB3, 0xFE, 0xFE, 0x30, 0x30, 0xC7, 0x66, 0x6E, 0x3C,
    0x3C, 0x18, 0x3C, 0x3C, 0x6E, 0x66, 0xC7, 0xC6, 0xC6, 0xC6,
    0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xFF, 0x03, 0x03,
    0x63, 0x63, 0x63, 0x63, 0x63, 0x73, 0x3F, 0x03, 0x03, 0x03,
    0x07, 0x3E, 0x30, 0x30, 0xFE, 0xFE, 0xB3, 0x33, 0x33, 0xB3,
    0xFE, 0xFE, 0x30, 0x30, 0xC7, 0x66, 0x6E, 0x3C, 0x3C, 0x18,
    0x3C, 0x3C, 0x6E, 0x66, 0xC7
};

[... 48 lines cut ...]
    { 494, 6, 8, 7, 0, -8 },   // 0xD0BE 'о'
    { 502, 6, 8, 7, 0, -8 },   // 0xD0BF 'п'
    { 510, 6, 11, 7, 0, -11 }, // 0xD180 'р'
    { 521, 6, 8, 7, 0, -8 },   // 0xD181 'с'
    { 529, 9, 8, 10, 0, -8 },  // 0xD182 'т'
    { 537, 6, 12, 7, 0, -12 }, // 0xD183 'у'
    { 549, 9, 14, 10, 0, -14 },// 0xD184 'ф'
    { 563, 6, 8, 7, 0, -8 },   // 0xD185 'х'
    { 571, 7, 10, 8, 0, -10 }, // 0xD186 'ц'
    { 581, 6, 8, 7, 0, -8 },   // 0xD187 'ч'
    { 589, 9, 8, 10, 0, -8 },  // 0xD188 'ш'
    { 597, 9, 10, 10, 0, -10 },// 0xD189 'щ'
    { 607, 8, 8, 9, 0, -8 },   // 0xD18A 'ъ'
    { 615, 7, 8, 8, 0, -8 },   // 0xD18B 'ы'
    { 623, 6, 8, 7, 0, -8 },   // 0xD18C 'ь'
    { 631, 6, 8, 7, 0, -8 },   // 0xD18D 'э'
    { 639, 7, 8, 8, 0, -8 }    // 0xD18E 'ю'
};

// Define the font using UTF-8 encoded glyphs
const GFXfont MyCyrillicFont PROGMEM = {
    (uint8_t  *)MyCyrillicFontBitmaps,
    (GFXglyph *)MyCyrillicFontGlyphs,
    0xD090, 0xD18F, 11   // Start and end of UTF-8 encoded Cyrillic range
};

board707, 2024-08-12:
it might be a problem with a panel itself - did you test it with standard latin font and text?

Ezopython, 2024-08-12:
> it might be a problem with a panel itself - did you test it with standard latin font and text?

yes its fine for latin font 
[image]

board707, 2024-08-12:
It seems that the `print()` method does not support working with multi-byte characters. In such a situation, I can advise using a font in a single-byte encoding of the cyrillic character set - for example Win-1251 or Koi8. And the text from the arduino should first be recoded from UTF8 to font encoding.
You can see an example in my DMD-STM32 library

Ezopython, 2024-08-13:
> DMD-STM32

can i use this library directly with my ESP32

board707, 2024-08-13:
No, but you could to use this code as an example and port it to your sketch

Ezopython, 2024-08-31:
does this library support Cyrillic Font in future with ESP32

board707, 2024-08-31:
What is the library you mean?