mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #666: Dark red background with FM6126A 64x64 panel

Bluescreen2001, 2024-08-23:
Hi 

I have three types of 64x64 panels here.
1. ICN2037/ICN2012 which works perfectly without initialisation
2. FM6126A/TC7262 which works but all the black pixels show up as dark red. If you look closely at the LEDs you see that red and green LEDs are glowing slightly. On one singe one 16x32 area, green glows a bit brighter. Can this be an initialization issue? What can I try
3. FM6126A/TC7258GN which shows no signs of life. If you power off/onn it rapidly you sometimes get colorful vertical lines. It's the same driver than 2. Why don't it work?

Thanks
Thorsten

board707, 2024-08-23:
Did you set the FM6126A driver setting in the library config?

Bluescreen2001, 2024-08-23:
Yes, I set the driver in config. Without it, the panel didn't do anything, so that setting seems to work, but shows the mentioned issue.

Bluescreen2001, 2024-08-23:
Hi,

I just coukd resolve the problem with the dark red background of panel 2. I have no idea about the meaning of the config registers, so I changed some bits and checked the result.
When I changed the bits from REG1[16] = {0,0,0,0,0, 1,1,1,1,1,1, 0,0,0,0,0} to REG1[16] = {0,0,1,1,0, 1,1,1,1,1,1, 0,0,0,0,0} the backgroud is absolute black now.
Does anyone have an idea what all the config bits are about?
There is one small issue left with panel 2. I have one dark green vertical line at position x=33. But I think this is caused by a defective LED. The topmost LED in that column only has a working red LED, green and blue is not working.

Now I have to try to get panel 3 running.
BTW: I want to create a big panel with 16 64x64 modules powered by an FPGA and use this library for learning how to use these panels. But fear to get panels which does not work with this librarie's init routine. Any idea where to get modules with a defined chipset and not such random lottery game?

board707, 2024-08-23:
While searching, look for the panels with drivers, listed as "standard" in that table:
https://github.com/board707/DMD_STM32/wiki/Led_drivers#led-drivers

Bluescreen2001, 2024-08-23:
Yes I already tried this, but it seems no chinese supplier can guarantee a specific chipset. It seems to be always random delivery. :-(

board707, 2024-08-23:
Are there no local dealers in your area where you can look at panels before you buy?

Bluescreen2001, 2024-08-23:
There are, but they are double or three times more expensive. And for 20 panels that would be too much for me.

planevina, 2024-09-25:
我也有这个问题，感谢看到这个解决方案

planevina, 2024-09-25:
> 你好
> 
> 我刚刚解决了面板 2 的深红色背景的问题。我不知道 config registers 的含义，所以我更改了一些位并检查了结果。当我将位从 REG1[16] = {0,0,0,0， 1,1,1,1,1,1， 0,0,0,0,0} 更改为 REG1[16] = {0,0,1,1,1,1,1,1,1， 0,0,0,0,0} 时，背景现在是绝对黑色的。有谁知道所有配置位是关于什么的？面板 2 留下一个小问题。我在位置 x=33 处有一条深绿色的垂直线。但我认为这是由 LED 故障引起的。该列中最顶部的 LED 只有一个正常工作的红色 LED，绿色和蓝色不工作。
> 
> 现在我必须尝试让面板 3 运行。顺便说一句：我想创建一个包含 16 个 64x64 模块的大面板，这些模块由 FPGA 提供支持，并使用此库来学习如何使用这些面板。但是害怕得到与这个库的 init 例程不兼容的面板。知道在哪里可以获得具有定义芯片组的模块，而不是这种随机的彩票游戏吗？
I also have this problem, thank you for seeing this solution. If you have completely resolved it, please make sure to let me know. Thank you

planevina, 2024-09-25:
> 你好
>
> 我刚刚解决了面板 2 的深红色背景的问题。我不知道 config registers 的含义，所以我更改了一些位并检查了结果。当我将位从 REG1[16] = {0,0,0,0， 1,1,1,1,1,1， 0,0,0,0,0} 更改为 REG1[16] = {0,0,1,1,1,1,1,1,1， 0,0,0,0,0} 时，背景现在是绝对黑色的。有谁知道所有配置位是关于什么的？面板 2 [...]

感谢你的方案，我的相同问题也解决了

mrcodetastic, 2024-09-25:
> However, there is another issue with my FM6126A panel: the x-axis is offset to the left by 1, which means that drawPixcel (1,0) is equivalent to drawPixcel (0,0) I don't know why

clkphase setting?

planevina, 2024-09-25:
> > However, there is another issue with my FM6126A panel: the x-axis is offset to the left by 1, which means that drawPixcel (1,0) is equivalent to drawPixcel (0,0) I don't know why
> 
> clkphase setting?

734/10000
实时翻译
划译
Thank you for your reply. I couldn't find a place to set CLK in the library, maybe I didn't notice. This is my initialization code, the default settings I am using
HUB75_I2S_CFG mxconfig;
mxconfig.driver = HUB75_I2S_CFG::FM6126A;
mxconfig.gpio.r1 = R1_PIN;
mxconfig.gpio.g1 = G1_PIN;
mxconfig.gpio.b1 = B1_PIN;
mxconfig.gpio.r2 = R2_PIN;
mxconfig.gpio.g2 = G2_PIN;
mxconfig.gpio.b2 = B2_PIN;
mxconfig.gpio.a = A_PIN;
mxconfig.gpio.b = B_PIN;
mxconfig.gpio.c = C_PIN;
mxconfig.gpio.d = D_PIN;
mxconfig.gpio.e = E_PIN;
mxconfig.gpio.lat = LAT_PIN;
mxconfig.gpio.oe = OE_PIN;
mxconfig.gpio.clk = CLK_PIN;
mxconfig.mx_height = PANEL_RES_Y;
mxconfig.mx_width = PANEL_RES_X;
mxconfig.chain_length = PANEL_CHAIN_LENGTH;
dma_display = new MatrixPanel_I2S_DMA(mxconfig);

planevina, 2024-09-25:
ok， I found it .  mxconfig.clkphase   . I will try it later.

planevina, 2024-09-25:
> > However, there is another issue with my FM6126A panel: the x-axis is offset to the left by 1, which means that drawPixcel (1,0) is equivalent to drawPixcel (0,0) I don't know why
> 
> clkphase setting?

 mxconfig.clkphase = false ; 

Now it's working perfect. Thank you