mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #668: Anyone knows if driver chip ICN2046 works with this library?

Bluescreen2001, 2024-08-24:


mrcodetastic, 2024-08-24:
There's no specific support of this LED driver chip, or at least, mention of the ICN2046 in the code base.

Feel free to buy us a panel if you want me to look at support. Looks like it might be possible, however the caveat will getting REG1 and REG2 values correct.

https://olympianled.com/wp-content/uploads/2021/05/ICND2046_datasheet_EN_2019_V1.2.pdf

Bluescreen2001, 2024-08-24:
Yes, I found that datasheet, too. Hopefully icnd2046 and icn2046 is the same or icn was a typo of the supplier. BTW, this is the first datasheet that exactly describes the config registers. Problem, for my project I need 20 panels, which is a bit risky. I can order one for you, too, if shipping and customs to your country is affordable.

mrcodetastic, 2024-08-24:
Can't you just order one panel to start off with? I'm in the UK.

mrcodetastic, 2024-08-24:
Also, you're unlikely to be able to have 20 panels chained working using a single ESP32 because of the limited memory.

If they're 64x32 pixel panels you'll be hard pressed to get 3bit colour across all 20. 64x64? No chance.

Bluescreen2001, 2024-08-24:
I can do that, but because these chinese suppliers have every week other panels that would be unrelieable. But I asked him if he can deliver these panels a few weeks later after Iborder two test panels.

Bluescreen2001, 2024-08-24:
I know, I cannot use 20 panels on the esp. I will take your library as example and learnung platform. And when I have it running I will try to realize it with an FPGA.

BTW. It will become a screen for retro computer games with 320x256 pixel resolution. Hopefully I can include a ZX Spectrum or Atari ST into the FPGA.