mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #670: White background

beiyer838, 2024-08-25:
I am using a 128x64 with ICN2038S.  I am not able to clear the screen completely. 
All the pixels seem to dimly lit.  It disappears if I disconnect and reconnect my ESP32 from the HUB75E.

    HUB75_I2S_CFG::i2s_pins _pins={R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK};
    HUB75_I2S_CFG mxconfig(
                            128,   // width
                            64,   // height
                             1,   // chain length
                         _pins,   // pin mapping
      HUB75_I2S_CFG::ICN2038S      // driver chip
    );
    
mxconfig.latch_blanking = 4;
mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_8M;
mxconfig.clkphase = 0;

//dma_display->clearScreen();
//dma_display->fillScreenRGB888(0,0,0);
dma_display->fillScreen(0);
dma_display->setTextSize(4);
dma_display->setTextColor(dma_display->color565(0,255,0));
dma_display->setCursor(1, 0); //0
dma_display->print("HELLO");

White background :
[image]

Disappeared after re-connecting ESP32 board to hub75 :
[image]

mrcodetastic, 2024-08-26:
Looks like an electrical grounding issue. Is the ESP's and the panel grounds connected?

beiyer838, 2024-08-26:
Yes ESP and panel share the same power source. 

[image]

beiyer838, 2024-08-26:
mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;
The panel look like this with clock speed at 20M :

[image]

beiyer838, 2024-08-27:
The panel has 2 pins for GND connection.  Disconnecting either one of them from ESP32 solves the problem.
Thanks mrcodetastic for your guidance.  

[image]