mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #681: QIANGLI 1/8 "5125 + 32019" vs "5125 + 5158" panels

rigorka, 2024-09-26:
The Led panels supplier that I have to work with presently only have two types of panels available for the form factor that I need: "5125 + 32019" vs "5125 + 5158". 

It seems both 32019 and 5158 are the shift registers that this library does not support "out of the box" presently.
I can get one of those panel types for testing easily, and I am reasonably fluent in C++, so could possibly try to implement the support for one of those shift regs. However the info on those chips seem to be a bit scarce.

I would greatly appreciate an advice from the developers of this lib re:
* is this an exercise worth undertaking?
* if so, which of the shift regs available (32019 vs 5158) might be a better option to choose?

board707, 2024-09-26:
According to datasheet, the DP32019 driver is not a shift register, it is a normal "138 type", compatible with a library code.

board707, 2024-09-26:
@rigorka 
What are the panel scan and dimensions in pixels?

rigorka, 2024-09-26:
@board707 Thank you very much for the wiki, it is really very useful! The panels are 32x32px, 1/8 scan. Looks like I shall try 32019 version then.

board707, 2024-09-26:
Keep in mind that the 32x32 1/8 panel will most likely have a non-standard pixel pattern, you will have to decipher it yourself.

rigorka, 2024-09-26:
Yep, that's what I'm good at - had to implement pixel mapping for a number of strange panels to date, just never worked with these chips before. Once again, thank you for the clarifications!

closed 2024-09-26 by rigorka

board707, 2024-09-26:
If there are any problems, I could help

rigorka, 2024-09-26:
Oops. My supplier just told me they're out of 32019 version so I will have to implement 5158, which is in fact a shiftreg. So might ask you for some advice after I receive the test panels early next week and do my homework on them.

board707, 2024-09-26:
This is actually a good occasion, because I have long wanted to try adding a shift register driver support to this library :)