mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #683: Need help with single 64x32 matrix with 1/8 scan rate

brianmackessy, 2024-10-03:
I'm having a lot of issues with 64x32 matrix with a 1/8 scan rate, in this code I am trying to draw a simple white border around the matrix. I am driving the matrix with a Adafruit Matrix Portal S3.

I tried to follow the instructions here to no avail:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/578#issue-2106096259

[image]

```
#include <Arduino.h>
#include <Adafruit_GFX.h>
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

// Define panel resolution
#define PANEL_RES_X 64
#define PANEL_RES_Y 32

// Declare pointers for DMA display and virtual display
MatrixPanel_I2S_DMA *dma_display = nullptr;
VirtualMatrixPanel *virtualDisp = nullptr;

void setup() {
  Serial.begin(115200);

  // Initialize HUB75 configuration
  HUB75_I2S_CFG mxconfig;

  // Set panel dimensions
  mxconfig.mx_height = PANEL_RES_Y;
  mxconfig.mx_width = PANEL_RES_X;
  mxconfig.chain_length = 1;  // Single panel

  // Set custom pin mapping
  mxconfig.gpio.r1  = 42;
  mxconfig.gpio.g1  = 41;
  mxconfig.gpio.b1  = 40;
  mxconfig.gpio.r2  = 38;
  mxconfig.gpio.g2  = 39;
  mxconfig.gpio.b2  = 37;
  mxconfig.gpio.a   = 45;
  mxconfig.gpio.b   = 36;
  mxconfig.gpio.c   = 48;
  mxconfig.gpio.d   = 35;
  mxconfig.gpio.e   = -1;  // Disable E_PIN for 1/8 scan panels
  mxconfig.gpio.lat = 47;
  mxconfig.gpio.oe  = 14;
  mxconfig.gpio.clk = 2;

  // Other configuration options
  mxconfig.clkphase = false;
  mxconfig.driver   = HUB75_I2S_CFG::SHIFTREG;

  // Initialize DMA display
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // Check if DMA display began successfully
  if (!dma_display->begin()) {
    Serial.println("****** I2S memory allocation failed ***********");
    return;
  }
  dma_display->setBrightness8(192);

  // Initialize virtual display
  virtualDisp = new VirtualMatrixPanel((*dma_display), 1, 1, PANEL_RES_X, PANEL_RES_Y);

  // Set the physical panel scan rate for 1/8 scan
  virtualDisp->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);
}

void loop() {
  // Clear the screen
  virtualDisp->fillScreen(virtualDisp->color565(0, 0, 0));

  // Draw a white rectangle border around the entire matrix
  virtualDisp->drawRect(0, 0, virtualDisp->width(), virtualDisp->height(), virtualDisp->color565(255, 255, 255));

  // Update the display
  delay(1000);
}

```

brianmackessy, 2024-10-07:
@board707 
I upgraded to the more modern solution you suggested, but I am running into an issue only half of the screen is being filled with red. Also the matrix doesn't seem as bright as it should be. I might try to buy 2 outdoor 32x32 matrices. People seem to be having more luck with them. Anything you can tell me helps.

https://github.com/user-attachments/assets/417a2441-9628-41e2-bbe0-3e36e77d99a7

```
/*************************************************************************
   Description:

   The underlying implementation of the ESP32-HUB75-MatrixPanel-I2S-DMA only
   supports output to HALF scan panels - which means outputting
   two lines at the same time, 16 or 32 rows apart if a 32px or 64px high panel
   respectively.
   This cannot be changed at the DMA layer as it would require a messy and complex
   rebuild of the library's internals.

   However, it is possible to connect QUARTER (i.e. FOUR lines updated in parallel)
   scan panels to this same library and
   'trick' the output to work correctly on these panels by way of adjusting the
   pixel co-ordinates that are 'sent' to the ESP32-HUB75-MatrixPanel-I2S-DMA
   library.

 **************************************************************************/

/* Use a custom Virtual Display class to re-map co-ordinates such that they draw
   correctly on a Four Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

 if ((coords.y & 8) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);

  return coords;
}
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
[... 76 lines cut ...]
  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,    VIRTUAL_MATRIX_CHAIN_TYPE);

  // THE IMPORTANT BIT BELOW!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
// **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
 // FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
      delay(20);
    }
  }

} // end loop
```

board707, 2024-10-07:
Hi
According to the video, the code is definitely not compatible with your panel. Could you look up all the chips on the panel and list it's ID codes on the topic?