mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #687: how to run on esp-idf

hijoh, 2024-10-10:
thank you so much ,do you have a sample on esp-idf？

abrender, 2024-10-10:
Hi @hijoh - there are two example folders in [examples/esp-idf](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/esp-idf) showing how to use this library with `esp-idf` (there is a README.md in each directory).

If you run into any specific problems please let us know and we'll do our best to help you out :)

hijoh, 2024-10-11:
esp-idf5.1,succsessful,thank you!
[image]

hijoh, 2024-10-13:
> Hi @hijoh - there are two example folders in [examples/esp-idf](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/esp-idf) showing how to use this library with [...]

I successfully ran in IDF 5.1 and with GFX. But how can I display Chinese?

closed 2024-10-16 by hijoh