mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #694: 160x80 pannel, support for 1x40 scan?

maghostmil, 2024-11-02:
Hi!

I bought LED panel from https://pl.aliexpress.com/item/1005006340698500.html
It's 1/40 scan. 160x80. 74HC245KA + RUL5158C. 

I can't drive this panel by any of the HUB75 libraries.

I tried to run this panel by Huidu D16 - chinese FPGA controller for LED panels.
I managed to do it after updating the D16 firmware.

I started test mode (full red screen) and I grabbed the signals form HUB75 connector, saved by KingstVis analyzer software, linked there:
https://drive.google.com/file/d/1BUEdgxgLq7yn_YyzUYndk9jWa-j90thd/view?usp=sharing

Could someone take a look at this waveform and analyze how it's control?
Will there ever be an option in this library to control this kind of display?

board707, 2024-11-03:
Could you show a screenshot of D16 hardware settings page?

maghostmil, 2024-11-03:
[image]

board707, 2024-11-05:
According to the Internet, the HX8863 is a PWM type driver with SRAM, incompatible with current library code.

maghostmil, 2024-11-07:
@board707 Thanks!

closed 2024-11-07 by maghostmil