mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #695: [Bug Report] Display Ghosting and Color Inconsistency Issues with ESP32-HUB75-MatrixPanel-DMA Library on P3 RGB LED Panel (MBI5124)

neiloe8023, 2024-11-06:
# [Bug Report] Ghost Image Issues with ESP32-HUB75-MatrixPanel-DMA Library on MBI5124 Driven P3 Panel

## Hardware Setup
- Controller: ESP32-S3
- LED Panel: P3 Full-Color LED Module
  - Size: 144mm x 48mm
  - Resolution: 16(H) × 48(W)
  - Driver IC: MBI5124
  - Scan Mode: 1/16 dual scan
  - LED Type: 2121 LED

## Issue Description
When using the ESP32-HUB75-MatrixPanel-DMA library to drive the above panel, the following display issues occur:

1. Text Scrolling Effect Shows Ghost Images
   - Yellow text displays with blue and purple trailing effects
   - Ghost images are particularly noticeable during text movement

2. Bouncing Rectangle Animation
   - Slight ghosting when displaying cyan rectangle
   - Inconsistent color appearance in the middle of the rectangle

## Attempted Solutions
1. Adjusted driver configuration:
	cpp
    mxconfig.driver = HUB75_I2S_CFG::MBI5124;
    mxconfig.clkphase = false;
    mxconfig.double_buff = false;

2. Used basic color values (0-7) instead of RGB565
3. Implemented independent pixel control and buffer management
4. Adjusted refresh timing and delays

## Expected Behavior
- Text scrolling should display solid colors without ghosting
- Rectangle animation should show uniform color without inconsistencies

## Additional Notes
1. Custom low-level driver code works correctly without ghosting issues
2. Issues only occur when using the ESP32-HUB75-MatrixPanel-DMA library

## Related Code
`#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

// 定义面板配置
#define PANEL_RES_X 48      // 面板宽度为48像素
#define PANEL_RES_Y 32      // 32用于16扫双扫描
#define PANEL_CHAIN 1      // 串联数量为1

// 定义HUB75E引脚配置 - ESP32-S3
#define R1_PIN  48  // R1 数据引脚
#define G1_PIN  47  // G1 数据引脚
#define B1_PIN  21  // B1 数据引脚
#define R2_PIN  45  // R2 数据引脚 (下半部分)
#define G2_PIN  0   // G2 数据引脚 (下半部分)
#define B2_PIN  14  // B2 数据引脚 (下半部分)
#define A_PIN   12  // 行选择 A
#define B_PIN   11  // 行选择 B
#define C_PIN   10  // 行选择 C
#define D_PIN   9   // 行选择 D
#define E_PIN   -1  // 不使用E引脚
#define LAT_PIN 46  // 锁存信号
#define OE_PIN  3   // 输出使能
#define CLK_PIN 13  // 时钟信号

// 创建显示配置对象
HUB75_I2S_CFG mxconfig;

// 创建矩阵面板对象
MatrixPanel_I2S_DMA *dma_display = nullptr;

// 定义颜色常量
#define COLOR_BLACK   0x00
#define COLOR_RED     0x01
#define COLOR_GREEN   0x02
#define COLOR_BLUE    0x04
#define COLOR_YELLOW  0x03
#define COLOR_PURPLE  0x05
#define COLOR_CYAN    0x06
#define COLOR_WHITE   0x07

// 显示缓冲区
uint8_t frame_buffer[PANEL_RES_Y][PANEL_RES_X];

void setup() {
  Serial.begin(115200);

  // 基本配置
  mxconfig.mx_width = PANEL_RES_X;   
  mxconfig.mx_height = PANEL_RES_Y;  
  mxconfig.chain_length = PANEL_CHAIN;

  // GPIO配置
  mxconfig.gpio.r1 = R1_PIN;
  mxconfig.gpio.g1 = G1_PIN;
  mxconfig.gpio.b1 = B1_PIN;
  mxconfig.gpio.r2 = R2_PIN;
  mxconfig.gpio.g2 = G2_PIN;
  mxconfig.gpio.b2 = B2_PIN;
  mxconfig.gpio.a = A_PIN;
  mxconfig.gpio.b = B_PIN;
  mxconfig.gpio.c = C_PIN;
  mxconfig.gpio.d = D_PIN;
  mxconfig.gpio.e = E_PIN;
  mxconfig.gpio.lat = LAT_PIN;
  mxconfig.gpio.oe = OE_PIN;
  mxconfig.gpio.clk = CLK_PIN;

  // 简化MBI5124配置
  mxconfig.driver = HUB75_I2S_CFG::MBI5124;
  mxconfig.clkphase = false;
  mxconfig.double_buff = false;

  // 初始化显示
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  if (!dma_display->begin()) {
    Serial.println("***面板初始化失败!***");
    return;
  }

  dma_display->setBrightness8(90);
  dma_display->clearScreen();
  
  // 清空缓冲区
  memset(frame_buffer, 0, sizeof(frame_buffer));
  
  Serial.println("面板初始化成功!");
}

// 清屏函数
void led_matrix_clear() {
    memset(frame_buffer, 0, sizeof(frame_buffer));
    dma_display->clearScreen();
}

// 设置像素点
void led_matrix_set_pixel(int x, int y, uint8_t color) {
    if(x >= 0 && x < PANEL_RES_X && y >= 0 && y < PANEL_RES_Y) {
        frame_buffer[y][x] = color;
        
        // 根据颜色值设置RGB
        uint16_t rgb565 = 0;
        if(color & 0x01) rgb565 |= 0xF800;  // R
        if(color & 0x02) rgb565 |= 0x07E0;  // G
        if(color & 0x04) rgb565 |= 0x001F;  // B
        
        dma_display->drawPixel(x, y, rgb565);
    }
}

// 滚动文字效果
void scrollingText() {
    static int x = PANEL_RES_X;
    const char *text = "ESP32-S3";
    
    while(x > -100) {
        led_matrix_clear();
        
        // 使用color565
        dma_display->setTextColor(dma_display->color565(255, 255, 0));  // 黄色
        dma_display->setCursor(x, 13);
        dma_display->print(text);
        
        x--;
        delay(50);
    }
    x = PANEL_RES_X;
}

// 彩虹效果
void rainbowEffect() {
    led_matrix_clear();
    
    for(int i = 0; i < PANEL_RES_X; i++) {
        uint8_t red = 127 * (sin(i * 3.14159 / 24) + 1);
        uint8_t green = 127 * (sin(i * 3.14159 / 24 + 2.094) + 1);
        uint8_t blue = 127 * (sin(i * 3.14159 / 24 + 4.189) + 1);
        
        for(int y = 0; y < PANEL_RES_Y; y++) {
            // 使用基本颜色
            uint8_t color = 0;
            if(red > 127) color |= COLOR_RED;
            if(green > 127) color |= COLOR_GREEN;
            if(blue > 127) color |= COLOR_BLUE;
            
            led_matrix_set_pixel(i, y, color);
        }
    }
    delay(50);
}

// 弹跳矩形效果
void bounceRect() {
    static int x = 0;
    static int y = 0;
    static int dx = 1;
    static int dy = 1;
    const int rectWidth = 8;
    const int rectHeight = 8;
    
    for(int i = 0; i < 100; i++) {
        led_matrix_clear();
        
        // 使用基本颜色绘制矩形
        for(int ry = y; ry < y + rectHeight; ry++) {
            for(int rx = x; rx < x + rectWidth; rx++) {
                led_matrix_set_pixel(rx, ry, COLOR_CYAN);
            }
        }
        
        x += dx;
        y += dy;
        
        if(x <= 0 || x >= PANEL_RES_X - rectWidth) dx = -dx;
        if(y <= 0 || y >= PANEL_RES_Y - rectHeight) dy = -dy;
        
        delay(30);
    }
}

void loop() {
    scrollingText();
    delay(1000);
    rainbowEffect();
    delay(1000);
    bounceRect();
    delay(1000);
}`

[image]
[image]
[image]
[image]
[image]
[image]

neiloe8023, 2024-11-06:
Custom low-level driver code:
`#include <Arduino.h>

// **********************************
// HUB75 引脚定义 - ESP32-S3
#define PIN_R1    48  // R1 数据引脚
#define PIN_G1    47  // G1 数据引脚
#define PIN_B1    21  // B1 数据引脚
#define PIN_R2    45  // R2 数据引脚 (下半部分)
#define PIN_G2    0   // G2 数据引脚 (下半部分)
#define PIN_B2    14  // B2 数据引脚 (下半部分)

// 行选择引脚 - 16扫需要4位地址线
#define PIN_A     12  // 行选择 A
#define PIN_B     11  // 行选择 B
#define PIN_C     10  // 行选择 C
#define PIN_D     9   // 行选择 D
#define PIN_E     20  // 行选择 D

#define PIN_CLK   13  // 时钟信号
#define PIN_LAT   46  // 锁存信号
#define PIN_OE    3   // 输出使能

// 显示参数定义
#define LED_chain 2 // 级联数为2
#define PHYSICAL_HEIGHT 16    // 物理行数（显示屏高度）
#define MATRIX_WIDTH   48  // 显示屏宽度
#define MATRIX_HEIGHT  (PHYSICAL_HEIGHT * 2)  // 缓冲区行数
#define SCAN_ROWS     16  // 扫描行数

// 颜色定义
#define COLOR_BLACK   0x00
#define COLOR_RED     0x01
#define COLOR_GREEN   0x02
#define COLOR_BLUE    0x04
#define COLOR_YELLOW  0x03
#define COLOR_PURPLE  0x05
#define COLOR_CYAN    0x06
#define COLOR_WHITE   0x07

// 缓冲区
uint8_t frame_buffer[MATRIX_HEIGHT][MATRIX_WIDTH];

// 字体定义(5x7)
const uint8_t font_5x7[][5] PROGMEM = {
    {0x3E, 0x51, 0x49, 0x45, 0x3E}, // 0
    {0x00, 0x42, 0x7F, 0x40, 0x00}, // 1
    {0x42, 0x61, 0x51, 0x49, 0x46}, // 2
    {0x21, 0x41, 0x45, 0x4B, 0x31}, // 3
    {0x18, 0x14, 0x12, 0x7F, 0x10}, // 4
    {0x27, 0x45, 0x45, 0x45, 0x39}, // 5
    {0x3C, 0x4A, 0x49, 0x49, 0x30}, // 6
    {0x01, 0x71, 0x09, 0x05, 0x03}, // 7
    {0x36, 0x49, 0x49, 0x49, 0x36}, // 8
    {0x06, 0x49, 0x49, 0x29, 0x1E}  // 9
};

// 初始化函数
void led_matrix_init() {
    // 配置引脚
    pinMode(PIN_R1, OUTPUT);
    pinMode(PIN_G1, OUTPUT);
    pinMode(PIN_B1, OUTPUT);
    pinMode(PIN_R2, OUTPUT);
    pinMode(PIN_G2, OUTPUT);
    pinMode(PIN_B2, OUTPUT);
    pinMode(PIN_A, OUTPUT);
    pinMode(PIN_B, OUTPUT);
    pinMode(PIN_C, OUTPUT);
    pinMode(PIN_D, OUTPUT);
    pinMode(PIN_CLK, OUTPUT);
    pinMode(PIN_LAT, OUTPUT);
    pinMode(PIN_OE, OUTPUT);
    
    // 清空缓冲区
    memset(frame_buffer, 0, sizeof(frame_buffer));
}

// 清屏函数
void led_matrix_clear() {
    memset(frame_buffer, 0, sizeof(frame_buffer));
}

// 设置像素点
void led_matrix_set_pixel(int x, int y, uint8_t color) {
    if(x >= 0 && x < MATRIX_WIDTH && y >= 0 && y < MATRIX_HEIGHT) {
        frame_buffer[y][x] = color;
        // 调试输出
        if(y >= 8) {
            Serial.printf("Setting pixel at x=%d, y=%d, color=%d\n", x, y, color);
        }
    }
}

// 绘制数字
void led_matrix_draw_number(int x, int y, int num, uint8_t color) {
    if(num < 0 || num > 9) return;
    
    for(int row = 0; row < 7; row++) {
        for(int col = 0; col < 5; col++) {
            if(pgm_read_byte(&font_5x7[num][col]) & (1 << row)) {
                led_matrix_set_pixel(x + col, y + row, color);
            }
        }
    }
}

// 绘制字符串
void led_matrix_draw_string(int x, int y, const char* str, uint8_t color) {
    int pos_x = x;
    while(*str) {
        if(*str >= '0' && *str <= '9') {
            led_matrix_draw_number(pos_x, y, *str - '0', color);
            pos_x += 6; // 字符间距
        }
        str++;
    }
}

// 刷新显示
void led_matrix_refresh() {
    static uint8_t row = 0;
    
    // 关闭输出使能
    digitalWrite(PIN_OE, HIGH);
    
    // 设置行地址 (修改为正确的行地址映射)
    digitalWrite(PIN_A, row & 0x01);
    digitalWrite(PIN_B, (row >> 1) & 0x01);
    digitalWrite(PIN_C, (row >> 2) & 0x01);
    digitalWrite(PIN_D, (row >> 3) & 0x01);
    
    // 输出数据
    for(int col = 0; col < MATRIX_WIDTH; col++) {
        // 上半部分数据 (0-7行)
        digitalWrite(PIN_R1, frame_buffer[row][col] & 0x01);
        digitalWrite(PIN_G1, frame_buffer[row][col] & 0x02);
        digitalWrite(PIN_B1, frame_buffer[row][col] & 0x04);
        
        // 下半部分数据 (8-15行) - 修正索引
        digitalWrite(PIN_R2, frame_buffer[row + MATRIX_HEIGHT/2][col] & 0x01);
        digitalWrite(PIN_G2, frame_buffer[row + MATRIX_HEIGHT/2][col] & 0x02);
        digitalWrite(PIN_B2, frame_buffer[row + MATRIX_HEIGHT/2][col] & 0x04);
        
        // 时钟脉冲
        digitalWrite(PIN_CLK, HIGH);
        delayMicroseconds(1);
        digitalWrite(PIN_CLK, LOW);
    }
    
    // 锁存数据
    digitalWrite(PIN_LAT, HIGH);
    delayMicroseconds(1);
    digitalWrite(PIN_LAT, LOW);
    
    // 打开输出使能
    digitalWrite(PIN_OE, LOW);
    delayMicroseconds(300);
    
    // 更新行计数 (修改为正确的行数范围)
    row = (row + 1) % (MATRIX_HEIGHT/2);  // 改为扫描16行
}

// 示例动画效果
void demo_animation() {
    static int count = 0;
    led_matrix_clear();
    
    // 显示数字
    char buf[10];
    sprintf(buf, "%02d", count);
    led_matrix_draw_string(5, 16, buf, COLOR_YELLOW);
    led_matrix_draw_string(20, 20, buf, COLOR_GREEN);
    led_matrix_draw_string(36, 24, buf, COLOR_BLUE);

    
    count = (count + 1) % 100;
}

void setup() {
    led_matrix_init();
}

void loop() {
    demo_animation();
    
    // 刷新显示(约60Hz刷新率)
    for(int i = 0; i < 100; i++) {  // 增加刷新次数
        led_matrix_refresh();
        delayMicroseconds(100);     // 减少延时
    }
}`

VonHirsch, 2024-11-10:
Power Supply?

neiloe8023, 2024-11-10:
> Power Supply?

Thank you for your quick response! The power supply is 5V 3A. Is there any issue with that?

VonHirsch, 2024-11-10:
Power should be good unless it's one of those "phone charger" supplies.

You might have to shorten your wiring.  See here: https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/134

[image]

mrcodetastic, 2024-11-10:
This looks to be an electical issue and not a result of this library.

Do you have ground pin(s) connected between the HUB75 and the ESP32?

neiloe8023, 2024-11-13:
> Power should be good unless it's one of those "phone charger" supplies.
>
> You might have to shorten your wiring. See here: #134 [...]

I used an adjustable power supply, not a mobile phone charger. The wiring was also short, using Dupont wires, and the GND was correctly connected. I had already set latch_blanking according to #134, but it didn't work.https://streamable.com/8t27ij 这个链接的视频是我自己写的一些代码的效果，https://streamable.com/lqoz0p 这个链接的视频是使用“ESP32-HUB75-MatrixPanel-I2S-DMA”的效果。

neiloe8023, 2024-11-13:
> This looks to be an electrical issue and not a result of this library.
> 
> Do you have ground pin(s) connected between the HUB75 and the ESP32?

Yes, I connected GND to the ESP32S3

neiloe8023, 2024-11-13:
> Power should be good unless it's one of those "phone charger" supplies.
>
> You might have to shorten your wiring. See here: #134 [...]

https://github.com/user-attachments/assets/3e09242e-03c2-47de-b6ff-18f5c10cbb8e

New progress, after I adjusted the power supply to 6v, the ghosting effect is much better, but there are still some problems, please watch the video

neiloe8023, 2024-11-13:
> ew progress, after I adjusted the power supply to 6v, the ghosting effect is much better, but there are still some problems, please watch the video

New progress, after I adjusted the power supply to 6v, the ghosting effect is much better, but there are still some problems, please watch the video

Please check out my latest resume

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/695#issuecomment-2472907979