mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #696: Support for creating moving text

HoangNhatPhuc, 2024-11-14:
I want to create a moving text effect in an area to display more content. However, I can't find a way to do it. Can you assist me with this?

panki27, 2024-12-08:
Everything you need is already in the library.

- Move the cursor
- Draw text
- Clear screen
- Repeat

```c
int cursor_x = 64;
void loop() {
  display->setCursor(cursor_x, 10);
  display->print("Hello World");
  cursor_x -= 1;
  if (cursor_x <= -20) {
    cursor_x = 64;
  }
  delay(200);
  display->clearScreen();
}
```

selam2, 2025-02-10:
Hello I tried this code and got such a result help me please 

https://github.com/user-attachments/assets/b0ec2386-1ac4-41e1-a614-f485f24755dd

closed 2025-02-10 by mrcodetastic