mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #698: Need help with 1/8 scan outdoor RGB LED matrix

sagarsapkota1, 2024-11-16:
I have found a demo code for ESP32 and that works very well for indoor [P5 - 64x32 display(1/16 scan)](https://www.aliexpress.com/item/1005003236353714.html), but the same code does not work for outdoor[ P5- 64x32- 1/8Scan display](https://www.aliexpress.com/item/1005007574799891.html).  I have attached the entire code with all the libraries files.

If anyone has worked with outdoor 1/8s display, please help. 
[SimpleTestShapes.zip](https://github.com/user-attachments/files/17784476/SimpleTestShapes.zip)

[image]

nur476, 2024-11-16:
Hello, I am facing with the same issue. 

I have P5 OUTDOOR Panel 64x32 1/8 - Chips on it SM16208 and RUL5158. Supplier sent this also:
[image]

I am wondering that these panel can be used with this library? 

I tested Four_Scan_Panel example without any change and it shows this:
[image]

and test with 1_Simple_Test_Shapes without any change shows:

[image]

I think with some guidance it is possible to make it work. Am I right?

 
Any help will be appreciated.

board707, 2024-11-16:
@sagarsapkota1 
Please find all types of chips on the rear side of the panel and list it types here.

sagarsapkota1, 2024-11-17:
I could see SM5166PF and DP5125D. The LED package is 1921. The PCB has **P5-1921-64*32-8S-S2** .

board707, 2024-11-17:
The good news that your panel uses a standard type LED driver and multiplexor. So the only thing that you need to start using the panel is to develop a pixel mapping.
Read the #622 discussion and see an examples, mentioned in it.

sagarsapkota1, 2024-11-19:
I am just a beginner using ESP32 and RGB matrix displays, I am totally confused with lots of things. I am one of those users who just uses the library and ask chatgpt to code :) . I have no idea about library, cpp file and .h file. Can you please download the zip file I have attached and tell me which file should I play with? Is it .h file or .cpp file ? I am using chatgpt to understand the codes. Today I made changes in ESP32-HUB75-MatrixPanel-I2S-DMA.h file. In this part,  clk_speed _i2sspeed = HZ_10M,
    uint8_t _latblk  = 1, // Anything > 1 seems to cause artefacts on ICS panels
    bool _clockphase = true,
    uint8_t _min_refresh_rate = 85 . I changed to HZ_20M and the display has slightly improved. The fonts are now appearing in their actual size and I can read something. Please tell me which file should I work on and what are the next steps ? There are several files and I am confused where to try the codes. The library file and the example code for 1/8scan does not work. And there is one more thing I have noticed. In this section of the ESP32-HUB75-MatrixPanel-leddrivers.cpp file, void MatrixPanel_I2S_DMA::shiftDriver(const HUB75_I2S_CFG& _cfg){
    switch (_cfg.driver){
    case HUB75_I2S_CFG::ICN2038S:
    case HUB75_I2S_CFG::FM6124:
    case HUB75_I2S_CFG::FM6126A:
       
I noticed that there are drivers like FM6124,FM6126A, etc. and in my case the driver is SM5166PF. So, do you mean that for this specific driver SM5166PF I have to do pixel mapping ? My expectation is that I have to make some changes in one of the .cpp or .h files and that change will work for all the *.ino codes. I want to know which section of the cpp or .h file is that which file I should work on ? The improved display image has been attached. 
[image]

board707, 2024-11-19:
> in my case the driver is SM5166PF.

It is not a driver, it is a line switcher (multiplexor) chip. The driver on your panel is DP5125. This driver doesn't requires any specific settings to work.

> My expectation is that I have to make some changes in one of the .cpp or .h files and that change will work for all the *.ino codes.

yes, it is true. The file that you need to edit is src/ESP32-VirtualMatrixPanel-I2S-DMA.h. The way of the changes is described in #622. I am not able to explain in more detail what you should fix in the file, since you do not know C++ and I do not know English well.

TopoAS, 2026-08-29:
I have the same class of panel (outdoor P5 64x32, 1/8 scan, SMD1921) and got
it working. The generic `FOUR_SCAN_32PX_HIGH` does not drive it: this panel
has **no D address line** (A/B/C only), and its wiring has a period of 8 in x
rather than the big left/right halves fold that mapping assumes — so half the
rows stay unreachable and fine detail comes out scrambled.

I reverse-engineered the real mapping by probing the DMA buffer pixel by pixel
and submitted it as a new scan type, `FOUR_SCAN_5PX_64X32HIGH`, here: https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/pull/959

If you want to try it before it is merged, add the enum value and the mapping
branch from that PR to `src/ESP32-HUB75-VirtualMatrixPanel_T.hpp`, then use:

```cpp
using MyScanTypeMapping = ScanTypeMapping<FOUR_SCAN_5PX_64X32HIGH>;
VirtualMatrixPanel_T<CHAIN_NONE, MyScanTypeMapping>* virtualDisp = nullptr;
```

Remember the DMA config must be `PANEL_RES_X*2` by `PANEL_RES_Y/2` (128x16).

[image]

[image]

[image]

My panel is marked `HRLOP-P5-1921/V2.0` / `P5out-728T-2311-1A-240` /
`P5-smd1921-8S-2`. If yours differs, the probing method in the PR description
should still let you derive your own variant.

board707, 2026-08-30:
Hi @TopoAS 

> I have the same class of panel (outdoor P5 64x32, 1/8 scan, SMD1921) 

Sorry, but this say almost nothing about the panel. To make your description definitive, please inform us about a chips on the panel.

> this panel
has no D address line (A/B/C only)

... as any other 1/8 scan panel...

> I reverse-engineered the real mapping by probing the DMA buffer pixel by pixel
and submitted it as a new scan type, FOUR_SCAN_5PX_64X32HIGH

If you read the library example Pixel_Mapping_test, you will see that the library already has a way to manage many types of a "quarter scan panels" such as yours. 
I don't want to underestimate your efforts, but I think if you try this example, you could to resolve your issue with the panel easier.