mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #699: Need help with P5 Outdoor Panel 64x32 1/8 - Chips on it SM16208 and RUL5158

nur476, 2024-11-16:
Hi,

I have a board with specs on title, working on it but some lines never lights up. You can see here:

[image]

I tried this code: [https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/500#issuecomment-1722257816](url) with various X Y sizes, but some lines on Y never responded.  Btw I am double checked A,B,C cabling, it seems as a multiplexing problem but I don't know why.

What is that I am missing?

Thanks in advance.

nur476, 2024-11-16:
Something strange. I am using this code on ESP32-S2:

`#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"
/* Use the Virtual Display class to re-map co-ordinates such that they draw
 * correctly on a 32x16 1/8 Scan panel (or chain of such panels).
 */
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"
 // default pins:
#define R1_PIN 45
#define G1_PIN 42
#define B1_PIN 41
#define R2_PIN 40
#define G2_PIN 39
#define B2_PIN 38
#define A_PIN  37
#define B_PIN  36
#define C_PIN  35
#define D_PIN  -1
#define E_PIN  -1 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
#define LAT_PIN 26
#define OE_PIN  21
#define CLK_PIN 33
  // Panel configuration
  #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
  #define PANEL_RES_Y 12 // Number of pixels tall of each INDIVIDUAL panel module.
  #define NUM_ROWS 1// Number of rows of chained INDIVIDUAL PANELS
  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
  // ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS
  // Change this to your needs, for details on VirtualPanel pls read the PDF!
  #define SERPENT false
  #define TOPDOWN false
  // placeholder for the matrix object
  MatrixPanel_I2S_DMA *dma_display = nullptr;
  /******************************************************************************
   * Setup!
   ******************************************************************************/
  void setup()
  {
    delay(250);
    Serial.begin(115200);
    Serial.println(""); Serial.println(""); Serial.println("");
    Serial.println("*****************************************************");
    Serial.println("*         1/4 Scan Panel                            *");
    Serial.println("*****************************************************");
  HUB75_I2S_CFG::i2s_pins _pins =
  { R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN };
    HUB75_I2S_CFG mxconfig(
                PANEL_RES_X,              // DO NOT CHANGE THIS
                PANEL_RES_Y,              // DO NOT CHANGE THIS
                NUM_ROWS*NUM_COLS           // DO NOT CHANGE THIS
                ,_pins            // Uncomment to enable custom pins
    );
    //mxconfig.driver = HUB75_I2S_CFG::ICN2038S;
    mxconfig.latch_blanking=1;
    mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
    // OK, now we can create our matrix object
    //mxconfig.i2sspeed =HUB75_I2S_CFG::HZ_10M;
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
    // let's adjust default brightness to about 75%
    dma_display->setBrightness8(10);    // range is 0-255, 0 - 0%, 255 - 100%
    // Allocate memory and start DMA display
    if( not dma_display->begin())
      Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
    dma_display->clearScreen();
    delay(500);
   }
  void loop()
  {
      uint16_t color = dma_display->color565(50, 50,50); //blue
      for (int y = 0; y < PANEL_RES_Y; y++)
     // for (int i = 0; i <128; i=i+2)
      {
         for (int x = 0; x < PANEL_RES_X; x++)
       //for (int f = 0; f < 64; f=f+1)
        {
            dma_display->drawPixel(x,y, color);
            Serial.print("drawPixel: ");
            Serial.print("(");
            Serial.print(x, DEC);
            Serial.print(",");
            Serial.print(y, DEC);
            Serial.println(")");
            delay(50);
          }
      }
      delay(5000);
      dma_display->clearScreen();
  } // end loop````

When I used 

PANEL_RES_X 64
PANEL_RES_Y 12

This is when loop completed:

[image]

But if I change

PANEL_RES_X 64
PANEL_RES_Y 12 to 16

Less leds up:

[image]

How can this be possible? With more drawing, it draws led in fact and with same x,y coordinates, it draws diffrrent places.

board707, 2024-11-16:
> it draws led in fact and with same x,y coordinates, it draws diffrrent places.

Because your matrix uses a shift register (595 type) multiplexor - RU5158

Your panel is a difficult case - it requires changing the code in two places at once - one for the quarter-scan panel and the second for a new type of multiplexer not supported in the library.

nur476, 2024-11-17:
> Because your matrix uses a shift register (595 type) multiplexor - RU5158
>
> Your panel is a difficult case - it requires changing the code in two places at once - one for the quarter-scan panel and [...]

Thank you for your reply. I tried much things but I couldn't  make it work with this library but I want to. 

I used library recommended as alternative at readme  named PxPixel and made it works very well with these settings:

MuxPattern: SHIFTREG_ABC - not ABC_BIN_DE
ScanPattern: LINE
BlockPattern: DBCA

And something strange: B pin connected to GND due to use of SHIFTREG_ABC

I know these are different libraries but need some clearances about: these pattern info may help to use this library? Can you give me some starting point to tune this library for this board?

Thanks in advance.

board707, 2024-11-17:
> these pattern info may help to use this library?

This information doesn't give much, since we already knew the mux pattern (SHIFTREG_ABC is the same as 595 type), and the block pattern type only makes sense in its own library.
The main problem with your matrix is ​​that the SHIFTREG_ABC MuxPattern is not implemented in this library, and to start setting up your panel, you first need to add it to code.

nur476, 2024-12-16:
Closing this issue. Thank you.

closed 2024-12-16 by nur476