mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #700: Dual Scan (1/16 Scan with Dual Rows) Support Not Working

neiloe8023, 2024-11-16:
Description:
Hello, I am using a HUB75 LED display with the following specifications:

Resolution: 16(H) × 48(W)
Driver IC: MBI5124
Scan Mode: 1/16 Dual Scan
The library works well for single-scan displays, but for dual-scan displays, only the upper half of the screen shows content, while the lower half remains blank. I encountered this issue when attempting to use the library for my dual-scan panel.

What I Tried:
I consulted with AI, which suggested using in the struct. However, this member does not appear to exist in the library, as it caused the following compilation error:double_row_addressingHUB75_I2S_CFG
perl
复制代码
'struct HUB75_I2S_CFG' has no member named 'double_row_addressing'
Verified my hardware connections and ensured I am using the default pin configuration for ESP32-S3.
Observed Behavior:
The upper half of the screen displays content correctly.
The lower half remains blank.
Previous Issue Reference:
I previously submitted [Issue #695](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/695), which was related to ghosting effects. This current issue is unrelated but affects my display with the same setup.

Hardware and Software Details:
Controller: ESP32-S3
Pin Configuration: Default library pins
LED Display Specs:
Resolution: 16(H) × 48(W)
Driver IC: MBI5124
Scan Mode: 1/16 Dual Scan
Library Version: 3.0.11 
IDE: Arduino IDE
Expected Behavior:
Dual-scan support for HUB75 displays should allow both the upper and lower halves of the screen to display content correctly.

Additional Notes:
If dual-scan support is not currently implemented, I would appreciate any guidance on manually modifying the library or other workarounds. Is there any plan to include dual-scan support in future updates?

Thank you for your time and assistance!

neiloe8023, 2024-11-16:
My Code
```
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

// 定义屏幕的宽度和高度
#define PANEL_RES_X 48 // 面板的宽度（像素）
#define PANEL_RES_Y 16 // 面板的高度（像素）
#define PANEL_CHAIN 1  // 链接的面板数量

// 创建 MatrixPanel 对象
MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup() {
  // 初始化面板
  dma_display = new MatrixPanel_I2S_DMA();
  HUB75_I2S_CFG mxconfig(PANEL_RES_X, PANEL_RES_Y, PANEL_CHAIN);

  // 默认设置，适配常见 P3 灯板
  if (!dma_display->begin(mxconfig)) {
    Serial.println("MatrixPanel-I2S-DMA 初始化失败!");
    return;
  }

  // 清屏
  dma_display->fillScreen(dma_display->color444(0, 0, 0));
}

void loop() {
  // 显示 "Hello"
  dma_display->fillScreen(dma_display->color444(0, 0, 0)); // 清屏
  dma_display->setTextColor(dma_display->color444(15, 0, 15)); // 设置紫色字体
  dma_display->setCursor(4, 10); // 设置文字起始位置
  dma_display->print("Hello");  // 显示文字
  delay(1000);
}

```

Show
[image]

neiloe8023, 2024-11-16:
Update:
I discovered a workaround that allows the dual-scan display to function correctly. By modifying the code as follows, the display now works as expected:
```
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

// 定义屏幕的宽度和高度
#define PANEL_RES_X 48 // 面板的宽度（像素）
#define PANEL_RES_Y 32 // 面板的高度（像素）
#define PANEL_CHAIN 1  // 链接的面板数量

// 创建 MatrixPanel 对象
MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup() {
    // 初始化面板
    dma_display = new MatrixPanel_I2S_DMA();
    HUB75_I2S_CFG mxconfig(PANEL_RES_X, PANEL_RES_Y, PANEL_CHAIN);

    mxconfig.latch_blanking = 2;
    mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;

    // 默认设置，适配常见 P3 灯板
    if (!dma_display->begin(mxconfig)) {
        Serial.println("MatrixPanel-I2S-DMA 初始化失败!");
        return;
    }

    // 设置亮度
    dma_display->setBrightness8(80);
    // 清屏
    dma_display->fillScreen(dma_display->color444(0, 0, 0));
}

void loop() {
    // 显示 "Hello"
    dma_display->fillScreen(dma_display->color444(0, 0, 0)); // 清屏
    dma_display->setTextColor(dma_display->color444(15, 0, 15)); // 设置紫色字体
    dma_display->setCursor(0, 19); // 设置文字起始位置
    dma_display->print("Hello");  // 显示文字
    delay(1000);
}
```
### Key Changes:

1. **Panel Height (`PANEL_RES_Y`)**:
   - Changed from  to . Although my panel's physical resolution is , setting the height to  resolves the issue.`16``32``16×48``32`
2. **Additional Configurations**:
   - `mxconfig.latch_blanking = 2`
   - `mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M`

### Observations:

After these changes, the dual-scan display works properly, with both the upper and lower halves displaying content correctly. However, I don't fully understand why changing the height to  resolves the issue for a  display. Could this be related to how the library handles dual-scan addressing internally?`32``16×48`

neiloe8023, 2024-11-16:
Additional Observations:
After implementing the above workaround, I noticed an issue with the Y-coordinate alignment:

The effective Y-coordinate is shifted to . This means that when I try to draw something at , it actually appears at on the display.08y = 0y = 8
Similarly, other Y-coordinates are misaligned, and I need to manually offset the values to correctly position elements on the screen.
This behavior seems unusual and could indicate an internal handling issue in the library for dual-scan displays.

Updated Questions:
Why does setting to (for a physical panel) resolve the display issue?PANEL_RES_Y3216×48
Could the Y-coordinate misalignment ( mapping to ) be related to how the library addresses rows in dual-scan mode?y=0y=8
I hope this additional information helps in understanding and resolving the root cause of these behaviors. Let me know if more tests or logs are needed!

board707, 2024-11-16:
> I don't fully understand why changing the height to resolves the issue for a display.

Hi
I've encountered panels like this before. Technically, your panel is a half of the 48x32 matrix. 

The next problem you will face is that your connection method does not support chaining multiple panels.

neiloe8023, 2024-11-17:
> > I don't fully understand why changing the height to resolves the issue for a display.
> 
> Hi I've encountered panels like this before. Technically, your panel is a half of the 48x32 16scan matrix.
> 
> The next problem you will face is that your connection method does not support chaining multiple panels.

Thank you, have you resolved these issues? If it's resolved, can you share it with me? Thank you very much!

board707, 2024-11-17:
> have you resolved these issues?

Unfortunately no, I don't have any such panels for testing.

neiloe8023, 2024-11-19:
> > have you resolved these issues?
> 
> Unfortunately no, I don't have any such panels for testing.

Okay, thank you for your reply